ARMmbed/mbed-os #7659: LoRaWAN connect() wrongly reports BUSY after incomplete connection attempt

EnsilicaEdd, 2018-07-31:
### Description
I've been testing corner cases in my project: trying to connect to LoRaWAN over-and-over and/or using the wrong credentials.

What I've found is that LoRaWANInterface::connect() returns LORAWAN_STATUS_BUSY if I call it after a previous, incomplete connect().  Incomplete in this case means that I either deliberately called disconnect() before receiving a CONNECTED event, or that I waited for a JOIN_FAILURE event.

I believe the issue is that CONN_IN_PROGRESS_FLAG is not being cleared from _ctrl_flags (in LoRaWANStack.cpp) in these two cases.

I'm not especially up-to-date, but I can't see any recent changes in the areas I believe need tweaking.  I'll make a Pull Request with my suggested fix shortly.

### Issue request type
[ ] Question  
[ ] Enhancement  
[X] Bug

kivaisan, 2018-08-01:
Thanks for the report. I implemented a test case for this scenario and I can confirm the behaviour. The fix PR looks good.

0xc0170, 2018-08-01:
@EnsilicaEdd 👍  For reporting and fixing

kivaisan, 2018-08-02:
@EnsilicaEdd PR has been merged, so could you please verify and close this issue?

EnsilicaEdd, 2018-08-02:
Both cases tested successfully.  Thanks all.

closed 2018-08-02 by EnsilicaEdd