ARMmbed/mbed-os #7672: Target - nRF52840 - I2C driver does not work with mbed-Cloud-Client example

surajdagar, 2018-08-01:
### Description

elm3, 2018-08-01:
Adding @naveenkaje

surajdagar, 2018-08-01:
I am trying to write before read so it can be related to below issue.
https://github.com/ARMmbed/mbed-os/issues/7425

dlfryar, 2018-08-01:
**Reproducer, build, and debugging information:**

**I2C data payload dump on read and write**
 
[mbed-os/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/i2c_api.c](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/i2c_api.c)

```
    printf("i2c_read: data (");
    for (int i = 0; i < length; i++) {
        printf("%0x", data[i]);
    }
    printf(")\r\n");
 
 
    printf("i2c_write: data (");
    for (int i = 0; i < length; i++) {
        printf("%0x", data[i]);
    }
    printf(")\r\n");
 ```

**Cloud Client Example Repo - reads fail**
https://github.com/ARMmbed/mbed-cloud-client-example
 
**Build Cloud Client**
mbed compile -t GCC_ARM -m NRF52840_DK --profile profiles/debug_size.json
cp ./BUILD/NRF52840_DK/GCC_ARM/mbed-cloud-client-example.hex /Volumes/DAPLINK/ && sync

**Build example-gnss**
mbed compile -t GCC_ARM -m NRF52840_DK --profile debug
cp ./BUILD/NRF52840_DK/GCC_ARM/example-gnss.hex /Volumes/DAPLINK/ && sync
 
**DAPLINK**
[MSD_COMMANDS](https://github.com/ARMmbed/DAPLink/blob/master/docs/MSD_COMMANDS.md)
Erase entire flash going between gnosis example and Cloud Client due to SOTP/nvstore region
touch /Volumes/DAPLINK/erase.act
 
**u-blox MAX-88Q GPS mbed-os driver used by example and imported to Cloud Client**
https://os.mbed.com/teams/ublox/code/gnss/
 
u-blox MAX-88Q GPS mbed-os example - works as expected single thread
mbed import https://os.mbed.com/teams/ublox/code/example-gnss
 
Changed needed to GNSS driver to work around enable signal when using a [MAX8Q module](https://v3.airspy.us/product/upu-ublox-m8q3a-sma/)
Add I2C driver call
[GnssI2C gnss(I2C_SDA0, I2C_SCL0);](https://os.mbed.com/teams/ublox/code/example-gnss/file/746ae478fdf7/main.cpp/)
 
Comment out _powerOn() call in [gnss.cpp/GnssI2C::init](https://os.mbed.com/teams/ublox/code/gnss/file/56eda66d585b/gnss.cpp/)
 
Disable EN signal and set to NC
[gnss.cpp](https://os.mbed.com/teams/ublox/code/gnss/file/56eda66d585b/gnss.cpp/)/_gnssEnable = new DigitalInOut(NC, PIN_OUTPUT, PullNone, 1);

naveenkaje, 2018-08-02:
@dlfryar @surajdagar 
Thanks for the detailed analysis and steps.

I see that we read 0 in the cloud client example but it seems to have valid data on the gnss-example. see attached trace logs..
[image]
[image]

I tried our proposed fix for https://github.com/ARMmbed/mbed-os/issues/7425, but that did not help.

I also tried experimenting by adding a dummy read in GNSS Init(). That did not change the results either. 

I will continue to work on this and update the thread.

dlfryar, 2018-08-03:
cc: @jrc92024

dlfryar, 2018-08-06:
@teetak01 or @marcuschangarm do you have any suggestions why [mbed-cloud-client-example](https://github.com/ARMmbed/mbed-cloud-client-example/tree/1.4.0) adding a separate thread to poll the I2C GNSS module every 5 seconds seems to have a ~4 minute lock on the I2C bus? Log and sources [here](https://gist.github.com/dlfryar/7f281f7b53eba7b538d9c26428e0957d).

It seems you can get in a state where the I2C reads work for ~6 I/Os in the sleep interval then it stops working for ~4 minutes then you get another ~6 I/Os complete.

**I2C concurrency works with the same hardware and two threads in isolated example code:**
[example-gnss-with-spif](https://github.com/dlfryar/example-gnss-with-spif) in a loop with I2C polling an I2C module in a separate thread while another thread doing continuous reads on a littlefs SPI part on the the same or really any interval has no problems.

|Sec since last | Seconds | Minutes|Min since last|
|---------------|---------|--------|--------------|
| |348.27|5.80|
|5.07|353.34|5.89|0.08|
|5.08|358.41|5.97|0.08|
|5.07|363.49|6.06|0.08|
|5.07|368.56|6.14|0.08|
|5.07|373.63|6.23|0.08|
|5.07|378.70|6.31|0.08|
|91.13|469.83|7.83|1.52|
|5.07|474.90|7.91|0.08|
|5.07|479.97|8.00|0.08|
|5.07|485.04|8.08|0.08|
|5.07|490.11|8.17|0.08|
|5.07|495.18|8.25|0.08|
|5.07|500.25|8.34|0.08|
|5.07|505.32|8.42|0.08|
|247.82|753.15|12.55|4.13|
|5.07|758.22|12.64|0.08|
|5.07|763.29|12.72|0.08|
|5.07|768.36|12.81|0.08|
|5.07|773.43|12.89|0.08|
|5.07|778.50|12.97|0.08|
|5.07|783.57|13.06|0.08|
|5.07|788.63|13.14|0.08|
|404.45|1,193.08|19.88|6.74|
|5.07|1,198.15|19.97|0.08|
|5.07|1,203.21|20.05|0.08|
|5.07|1,208.29|20.14|0.08|
|5.07|1,213.36|20.22|0.08|
|5.07|1,218.43|20.31|0.08|
|5.07|1,223.49|20.39|0.08|
|5.07|1,228.56|20.48|0.08|
|242.71|1,471.27|24.52|4.05|
|5.07|1,476.34|24.61|0.08|
|5.07|1,481.41|24.69|0.08|
|5.07|1,486.48|24.77|0.08|
|5.07|1,491.55|24.86|0.08|
|5.07|1,496.63|24.94|0.08|
|5.07|1,501.70|25.03|0.08|
|5.07|1,506.77|25.11|0.08|

yogpan01, 2018-08-07:
Cloud Client doesnt interact with I2C interfaces directly so if some internal Mbed OS component is doing that on behalf of Client . its hard to tell. Is this out-of-box configuration or there some modifications done on example configuration and setup to make this work on NRF52 board? @marcuschangarm Do you think if this is related to some file system configuration which is causing this issue?

naveenkaje, 2018-08-07:
I continued to debug this issue. The main_application thread of the mbed cloud client has no effect on the location thread. I stubbed it out to effectively do a sleep for ever. I also saw there was some randomness to the success of the reads. screenshots and logs attached.

The observation is that if the read of the response to FD is successful, the following write (FF) results in valid data.

[Log.txt](https://github.com/ARMmbed/mbed-os/files/2268106/Log.txt)
[image]

surajdagar, 2018-08-07:
I also saw there was some randomness to the success of the reads. Just a random thought, Is it make sense to play with main_application and GNSS thread priorities?

naveenkaje, 2018-08-07:
I tried bumping the priority of the GNSS thread. That didn't lead to anything. I will keep that in mind as well. Thank you!

naveenkaje, 2018-08-08:
I have verified this change on my set up too and I don't see the error as I did before. @surajdagar , could you please provide your observations?

surajdagar, 2018-08-08:
I get GPS data couple of times, but no communication on the bus after that. Can you please share the changes you did? And are these changes on Mbed-Cloud-Client code?

dlfryar, 2018-08-08:
Working [code](https://gist.github.com/dlfryar/1cb68b8e218f62c3afd6c8537d4567fa) example polling the I2C/GNSS module in a separate mbed-os thread.

Note this is a modified version of the [u-blox example](https://os.mbed.com/teams/ublox/code/example-gnss).

I also changed the [I2C frequency](https://github.com/dlfryar/example-gnss-with-spif/blob/master/gnss/gnss.cpp#L341) to 400kHz

closed 2018-08-14 by surajdagar