ARMmbed/mbed-os #7674: nRF SDK15 and USB Support

AGlass0fMilk, 2018-08-02:
### Description
Hi,
I'm interested in moving up to the nRF52840 for a future project. The current nRF SDK supported by mbed is 14.2. The update for this was completed fairly recently. It looks like mbed is planning renewed support for the USB (see feature branch here: https://github.com/ARMmbed/mbed-os/tree/feature-hal-spec-usb-device), which would be great to have on the nRF52840.

The nRF SDK14.2 documentation lists the USB driver (and support for the nRF52840 in general) as "experimental" and not "production quality". Is there any plan to start updating to nRF SDK15 for production quality support for the nRF52840 and USB library? Might be something I'd like to help with.

@marcuschangarm have you looked into this? I vaguely remember that quite a bit changed in the structure of some drivers from v14.2 to v15.

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug

AGlass0fMilk, 2018-08-18:
Any word @marcuschangarm? I've started to work a bit with SDK V15 and it seems easy to switch over, however it would be quite mundane...

I'd be willing to help :)

DL6AKU, 2018-09-26:
Also interested in SDK 15

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTDEV-1591*

AGlass0fMilk, 2018-11-12:
I have started my own implementation of USBPhyHw for the nRF52840 target. You can check out my code here:
https://github.com/AGlass0fMilk/mbed-os/tree/nrf52840-usb-hal

AGlass0fMilk, 2018-11-12:
The current implementation is not functioning yet. I am having issues with hard faults when connecting USB. I am also not sure if my implementation completely follows the HAL spec.

If anyone could help it would be appreciated. I would love to see USB support for the nRF52840 out of the box for Mbed.

AGlass0fMilk, 2018-11-12:
I added code from SDK 15.2 to fix the hang when enabling USB. The issue was related to errata 171 and 187 in the nRF52840 production IC.

I suspect the hard fault on reset when USB is already plugged in is caused by some USB interrupt happening before the rest of the USB stack has been initialized. There are a few other issues that are keeping it from working.

0xc0170, 2018-11-12:
> I suspect the hard fault on reset when USB is already plugged in is caused by some USB interrupt happening before the rest of the USB stack has been initialized. There are a few other issues that are keeping it from working.

Do you keep track of issues you are having ?

cc @c1728p9 (might be interested in this)

AGlass0fMilk, 2018-11-12:
> Do you keep track of issues you are having ?
>
> cc @c1728p9 (might be interested in this)

I have been in my head so far :) I just started this over the weekend so I haven't had much time to formally write up issues. I'll document them here shortly.

marcuschangarm, 2018-11-12:
@TacoGrandeTX

AGlass0fMilk, 2018-11-13:
I have been basing the `USBPhyHw` implementation for the nRF52840 off of t[his example from the Nordic SDK](http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk5.v14.2.0/examples_usb.html). It shows how to use the raw USBD HAL/Driver without Nordic's "class"-based USB profile system.

I'm in the middle of refreshing myself on [USB in a NutShell](https://www.beyondlogic.org/usbnutshell/usb1.shtml) and have pretty much only dealt with chip-level issues at this point so there's lots of room for improvement...

Here are the issues with the implementation that I have noticed:

### First Issue
**Status: Resolved**

**Description:** When enabling the USBD peripheral by calling `nrf_drv_usbd_enable` the MCU would hang on the following loop:
https://github.com/ARMmbed/mbed-os/blob/0404701b5f5b9d2aff76432991c74a36bd80742c/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_14_2/drivers_nrf/usbd/nrf_drv_usbd.c#L1706-L1709

The `READY` event that was supposed to occur to break this loop would never happen. This was due to nRF52840 erratas [171 (USB might not power up)](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.nrf52840.Rev1.errata%2Fdita%2Ferrata%2FnRF52840%2FRev1%2Flatest%2Fanomaly_840_171.html&anchor=anomaly_840_171) and [187 (USB cannot be enabled)](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.nrf52840.Rev1.errata%2Fdita%2Ferrata%2FnRF52840%2FRev1%2Flatest%2Fanomaly_840_187.html&anchor=anomaly_840_187). 

**Solution:** The workarounds for these issues are included in nRF SDK 15.2 (the most recent release) so I added them in to the `nrf_drv_usbd.c` driver included with mbed.

### Second Issue
**Status: Unresolved**

**Description:** When USB is already plugged into the target and a reset occurs (I saw this while resetting via debugger), Mbed will `die`. This occurs very early in the startup process (before the stdout UART has time to be configured) so I think it is related to a USB or Power interrupt. I'm guessing the interrupt occurs before the USB stack has been fully initialized (or perhaps before the static `instance` has been populated) and probably calls a function on some `NULL` pointer or something. I still have to look into this one a bit more.

**Possible Solutions:**  Currently, the `USBPhwHw_Nordic` implementation uses the same logic as[ this example from the Nordic SDK](http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk5.v14.2.0/examples_usb.html). This means the `usbd` peripheral is enabled and started or stopped and disabled when USB-related power events happen. There are no flags to make sure the USB stack is ready to accept events from the `USBPhyHw` at this point, which I think could help solve this problem.

### Third Issue
**Status: Unresolved**

**Description:** USB events aren't synchronized and cause failures in the Nordic driver during the USB setup stage. 

I kind of just slapped `nrf_drv_usbd` calls inside their counterparts in the `USBPhyHw` HAL API and hoped it would somewhat work. After spending a few hours debugging the first issue up there, I finally got to the point where USB-related events were being sent to the upper layers of the stack. Where I'm at now is that the `usbd` peripheral is successfully enabled, receives `setup` events from the host and sends them to the upper level of the stack. I believe when I was debugging, the place where the failure occurred was here:
https://github.com/ARMmbed/mbed-os/blob/0404701b5f5b9d2aff76432991c74a36bd80742c/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_14_2/drivers_nrf/usbd/nrf_drv_usbd.c#L1954-L1962

So some synchronization needs to be introduced ([should've listened to the HAL...](https://os.mbed.com/docs/v5.8/reference/usb-device.html#notes)) to prevent this check from failing.

TacoGrandeTX, 2018-11-15:
@AGlass0fMilk, @DL6AKU - We'll soon be releasing SDK15.0 peripheral driver support as a feature branch.  This first release won't have USB support but that is on our backlog.

AGlass0fMilk, 2018-11-15:
@TacoGrandeTX I will see what I can do in terms of getting a reliable USB HAL implementation working then. I don't think much changed in terms of functionality between SDK14.2 and SDK15 so my code should at least give you guys a place to start.

Hopefully I can take USB off your backlog :smiley:

TacoGrandeTX, 2018-11-15:
@AGlass0fMilk Thank you so much - that will be great!  We also look forward to your comments on the feature release.

AGlass0fMilk, 2018-11-29:
I have been working on the nRF52840 USBDevice HAL in my spare time for a bit now. 

I ended up finding a good deal on a Total Phase Beagle 12 USB hardware analyzer so I picked that up and it's been a great help in debugging. It's very difficult to debug USB without knowing what's really happening on the bus... so that took a few days to come in.

The stage I'm at is this:

The USBPhyHw implementation passes setup packets properly to the upper layers of the stack. Enumeration goes fine, at least for getting the device descriptor and then it hangs at the next control transfer.

I have an STM32F407 Discovery board on hand so I compiled my code for that target (which already has USB Device support working). This let me see what the full transaction was supposed to look like.

My laptop's USB host asks for the first 9 bytes of the configuration descriptor. The USB device responds, letting the host know the full configuration descriptor size is 75 bytes. Then the host requests the full 75-byte configuration descriptor. The device sends the first 64 bytes (the max packet size) and then the host asks for the rest of the descriptor with another IN packet.

The trouble is this:
Mbed's USB implementation already splits transactions up into smaller transfers that do not exceed the max packet size (64 bytes). I found that Nordic's underlying call to `nrf_drv_usbd_ep_transfer` also seems to do this and expects the full packet length (75 bytes) provided as an input. When the host asks again for the last 11 bytes of the descriptor, the Nordic HAL driver is confused since it thinks it has already sent all data for this setup transaction. EP0 then stalls...

It seems I must implement my own `nrf_drv_usbd_feeder_t` to implement transfers that are facilitated by mbed. Not sure if I will have to do the same for OUT transfers at this point.

AGlass0fMilk, 2018-11-29:
Here's a screenshot of my USB analyzer's output showing enumeration with the STM32 (left, functioning correctly) and the nRF52840 (right, issue with configuration descriptor transfer):
[image]

AGlass0fMilk, 2018-11-29:
After looking through both the Mbed stack and the nRF USBD driver, it seems they aren't easily compatible.

The Nordic driver uses the concept of "feeders" where every time a DMA transfer is finished, a feeder function is called to set up the next DMA transaction. During this call, the feeder returns true or false, representing if this is the last transfer or not. This allows the DMA to be released.

Mbed's USB stack does not expose the total size of the transaction to the USBPhyHW layer, nor let it know if this is the last transfer.

I'm playing with a few ideas on how to hook these two systems up. I'm not really liking any of them.

Let me know what you think:

Idea 1.) This idea involves changing the USBPhyHw interface to add a `bool` flag argument to `ep0_write` and `endpoint_write` to tell the hardware layer this is the last transfer in a series of transfers, and that it may release the DMA afterwards. This would obviously make my code a lot simpler but I don't like the idea of changing an upper layer due to the needs of a lower layer. Though it kind of does make sense to let the hardware know when it may release DMA resources... let me know if this would be possible to add

Idea 2.) Collect the full packet in `ep0_write` and `endpoint_write`. This would involve somewhat "tricking" the upper layer into giving the USBPhyHw the full-length packet by simulating `IN tokens` from the host. Kind of hacky, I don't really like this idea. Not even sure if it's possible. Also would have to allocate more RAM, which isn't very great.

Idea 3.) Modify the Nordic driver to allow the DMA to be released externally (essentially set a flag telling the DMA code that the endpoint no longer needs it). I think the USBPhyHw would still need to know when the last packet was being sent to do this, so it may not be possible without idea 1. The caveat: If the device wants to transfer exactly what the host requested and it's a multiple of the endpoint size, there's no way to know what the last packet is based on chunk size. That's why zero-length packets exist.

Any ideas? I think I like idea 1 out of all three. It seems to be the most straightforward for me and doesn't really impact the other supported implementations. Other targets that don't need this information can ignore the flag. Upper layers of the stack know when they are sending the last packet as well, so it doesn't add much complexity there either.

Let me know your thoughts @TacoGrandeTX

AGlass0fMilk, 2018-12-06:
The Nordic hardware needs more synchronization to the control transfer stages than I first thought. So my PhyHw implementation had to keep track of control transfer sizes and trigger the appropriate synchronization events in the hardware (telling the HW when control transfer Data OUT stages were allowed, when the status stage was allowed, etc).

After using my nifty Beagle 12 USB analyzer, I worked out the bugs enough to get a basic USB serial example working!
[image]

I just got this working, so by no means is it tested or production ready. But if anyone would like to start using USB on the nRF52840 this is a starting point. Please report any bugs you encounter back to this issue.

I'm sure there are better ways to connect the Mbed HAL layer with the Nordic driver, so please give me feedback on my implementation ([see it here](https://github.com/AGlass0fMilk/mbed-os/blob/nrf52840-usb-hal/usb/device/targets/TARGET_NORDIC/TARGET_MCU_NRF52840/USBPhy_Nordic.cpp)).

Gonna go test some of the other built-in device types :grin:

AGlass0fMilk, 2018-12-06:
@TacoGrandeTX should I submit a pull request to the feature branch at this time?

cmonr, 2018-12-06:
Impressive. @c1728p9, how does this analysis compare with the changes/updates/fixes you made recently?

c1728p9, 2018-12-06:
Hi @AGlass0fMilk awesome work so far! I would definitely recommend submitting a PR. The implementation will need to pass the mbed-os USB tests before it can be accepted though. You can run the tests locally with the command:
 ```mbed test -t GCC_ARM -m NRF52840_DK -n mbed-os-tests-usb_device-*```

AGlass0fMilk, 2018-12-07:
@c1728p9 What kind of setup is required to run the USB device tests on Ubuntu? Are there any drivers needed? None of the basic USB device tests passed.

c1728p9, 2018-12-07:
Hi @AGlass0fMilk if you are getting a permission error you may need to add the account running the test to the `plugdev` and `dialout` groups, or alternatively run the command with `sudo`. Aside from that there shouldn't be any special setup required on Ubuntu. Just connect both USB ports and run the mbed test. As a stable reference you could try running the tests on any of the other supported targets on the feature branch.

AGlass0fMilk, 2018-12-09:
My user is in both `dialout` and `plugdev` groups. No permissions issues I can see.

Unfortunately the STM32-based target I have on hand right now is the DISCO-F407VG, which I flashed with JLINK software (converts STLINK to JLINK, super great). It isn't an "officially" supported mbed board either, but the target is there. This means it does not have a target id on-board and mbedls doesn't find it... it is surprisingly hard to figure out how to test a new board that does not have an mbed ID.

Anyway, are there any extra flags required to run a driver script on the host side? From just poking around I've found scripts that seem to be custom python USB drivers for running these tests.

To me it seems like something times out so the tests fail. I also don't see any traffic on the bus beyond basic enumeration and configuration...

[image]

AGlass0fMilk, 2018-12-09:
I tried the nRF52840 tests again with the `-vv` flag. It appears the host test is running but it fails to find the device by serial number. See output in this gist:

https://gist.github.com/AGlass0fMilk/c382d00ba81f3a539ec41683d6bfb7a2

AGlass0fMilk, 2018-12-10:
I have an STM32F401 Nucleo board that is supported by mbed, I don't have it on hand at the moment though. I'll hopefully be able to test with that later.

I think I sorted out the permissions issues by adding a udev rule (in `/etc/udev/rules.d/99-mbed-test-usb.rule`) as below:
`SUBSYSTEM=="usb", GROUP="dialout", MODE="0660"`
It could be more specific but this worked for me, please note that it affects permissions for ALL USB devices plugged into your machine. For anyone referencing this in the future, you also need to reload the udev rules: `sudo udevadm control --reload` and make sure your user is in the `dialout` group (check the output of `groups`). If you need to add yourself to that group, `sudo adduser <your_username> dialout` and then logout/login for it to take effect.

I also changed the way `find_device` finds the test device through pyusb. I changed it to look for devices with the vendor id `0x0D28`... not sure if this was a different issue or also related to permissions. Pyusb had trouble reading string descriptors because the Mbed test device had no "langid".

Anyway, my implementation appears to fail the first test when configured->deconfigured->reconfigured. So I will start tracking down bugs now.

AGlass0fMilk, 2018-12-11:
So the device is failing to reconfigure. I have narrowed it down to a call into the Nordic driver that returns `NRF_ERROR_BUSY`. You can see the device reports the failed setting of the configuration in the screenshot below:
[image]

It seems to caused by [`USBTester::setup_iterface`](https://github.com/ARMmbed/mbed-os/blob/8517deedcf827dd71b0bda36d19aa42ceb8688cd/TESTS/usb_device/basic/USBTester.cpp#L223-L232)
[image]

The consecutive read after having just added the endpoint causes the Nordic driver to return that it's busy. If I remove the immediate read the configuration gets set successfully.

Why is the immediate read necessary?

c1728p9, 2018-12-19:
Hi @AGlass0fMilk some devices nak out packets until a read is started on the endpoint. On these devices `read_start` must be called to start receiving data.  Often an interface is ready to receive data as soon as its endpoints are added so you'll often see a `read_start` after adding an out endpoint. For example the serial port USB class, USBCDC, does that [here](https://github.com/ARMmbed/mbed-os/blob/65d71bbcc5d029e25d5e6e45bb066552ba45008c/usb/device/USBSerial/USBCDC.cpp#L297).

AGlass0fMilk, 2018-12-23:
Slowly working through test failures when I get the time... the USB test suite seems pretty thorough!

Currently passing the first two tests. Working on getting it to pass the third test (usb control sizes test).

AGlass0fMilk, 2018-12-27:
@c1728p9 I'm hitting an assert in the nordic driver during the `endpoint test data correctness`. See below assert in nordic driver (line 1921)

https://github.com/AGlass0fMilk/mbed-os/blob/ceb2bc064ee74e1345fc090b426293705419f423/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_14_2/drivers_nrf/usbd/nrf_drv_usbd.c#L1918-L1921

After some debugging, it seems the endpoint tester class is set up to create ISO endpoints with a max packet size of 0 bytes. This doesn't seem like it follows the USB specification. If this is not a mistake, what is the intended behavior of the USB hardware implementation in this case? (see snippet below)

https://github.com/ARMmbed/mbed-os/blob/0e5dd39264e391be045e87ad505a86fd84bcd91b/TESTS/usb_device/basic/USBEndpointTester.cpp#L61-L85

cmonr, 2018-12-27:
@c1728p9 @fkjagodzinski Would y'all happen to know an answer to the above question?

AGlass0fMilk, 2018-12-28:
A few questions:

### **1.)** Is there a place where general Mbed USB stack development conversation should go? Like a forum post for feature development or something?

### **2.)** Immediate Read During Configuration

@c1728p9 concerning the immediate read a few comments ago: I was looking into how to solve this compatibility issue, thinking I may be able to use the `configure` method of `USBPhyHw`. I read through the code flow a bit more and the `read_start` call happens during the user/application `set_configuration` callback, before the USB stack has had a chance to call `USBPhyHw::configure()`.

So when the USBPhyHw receives an endpoint read request on a just-added endpoint, it has not yet had `configure` called on it telling it the new configuration is valid and that it should be ready to respond to endpoint events appropriately.

This probably would complicate a lot of the existing device classes if they aren't allowed to queue read/write requests during configuration. However, I don't think it follows logically to execute requests on endpoints before the hardware driver has been fully `configure`d.

Thoughts on this? I'm about ready to scrap the nordic transfer logic and write my own that is more compatible with the mbed structure.

### **3.)** Flow Diagram Request

An execution flow diagram would be _extremely_ handy during the porting process. There are so many nested function calls it becomes hard to follow exactly when and where USBPhyHw, USBDevice, and user implementation code is called.

I'm going to start working on one on draw.io here: https://drive.google.com/file/d/1-4Hu2yBPtXBtGy_E2vj5e6xscTX4AmO_/view?usp=sharing

Send me an edit request if you want to add to it.

AGlass0fMilk, 2018-12-28:
> Hi @AGlass0fMilk, this is not a mistake. In fact, the USB spec prohibits non-zero `wMaxPacketSize` values for iso endpoints in the default interface altsetting. Regarding all other altsettings, `0` is a placeholder value until the tests are updated.
> 
> A quote from USB 2.0 spec, paragraph 5.6.3:
> 
> > All device default interface settings must not include any isochronous endpoints with non-zero data payload sizes (specified via wMaxPacketSize in the endpoint descriptor). Alternate interface settings may specify non-zero data payload sizes for isochronous endpoints.

My mistake, I'll have to read through the USB spec a bit more thoroughly.

c1728p9, 2019-01-02:
Thanks for the work so for on the NRF52 USB stack @AGlass0fMilk. As for your questions:

> 1.) Is there a place where general Mbed USB stack development conversation should go? Like a forum post for feature development or something?

Github is probably the best place for this. Mention me with @c1728p9 to ensure I see the messages. 

> 2.) Immediate Read During Configuration

Does the NRF52 require an explicit call to enter the configured state before reads and writes are allowed? If so, one possible solution is to save off the arguments and set a read pending flag if `read_start` is called before the device is in the configured state. Then once `USBPhyHw::configure()` is called go through all the added endpoints with the pending flag and start the reads for real.

Is the NRF52 returning `NRF_ERROR_BUSY` on `read_start` due to being outside the configured state or beacuse it is in the middle of a control transfer? If it is because it is in the middle of a control transfer then the above solution I mentioned won't work. In additon to adding endpoints and calling `read_start` due to a `Set Configuration` request, endpoints are also added and started in the `Set Interface` request. This takes place while the device is in the configured state and a `USBPhy` callback isn't called after the request completes.

> 3.) Flow Diagram Request

The current USB techonology page documentation can be found in https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/reference/technology/USB.md. The `Control request state machine` section has a flow chart showing showing how control requests are handled. Is this the kind of execution flow diagram you are looking for? If there any any information that can be added to this page to make USB easier to understand I would be happy to see it added.

AGlass0fMilk, 2019-01-03:
Oops, accidentally submitted that last comment too early and closed the issue... I blame arcane GitHub keyboard shortcuts...

cmonr, 2019-01-03:
@AGlass0fMilk Been there, but with a PR instead...

AGlass0fMilk, 2019-01-11:
I have started to meticulously go through the USB stack execution flow. I think there are some corner cases or race conditions causing my implementation to fail some of the test cases in the `usb basic` test.

I will keep an updated table of tests that are passing/failing as I continue development. As of right now (1/10/2019) the testing status of my implementation is as shown below:

| test case                                  | status                  |
|--------------------------------------------|-------------------------|
| usb control basic test                     | :heavy_check_mark:  PASSING               |
| usb control stall test                     | :watch: :x:  INTERMITTENT FAILURE |
| usb control sizes test                     | :heavy_check_mark: PASSING               |
| usb control stress test                    | :heavy_check_mark:  PASSING               |
| usb device reset test                      | :heavy_check_mark:  PASSING               |
| usb soft reconnection test                 | :x:  FAILING               |
| usb repeated construction destruction test | :heavy_check_mark:  PASSING               |
| endpoint test data correctness             | :x:  FAILING               |
| endpoint test halt                         | N/A                     |
| endpoint test parallel transfers           | N/A                     |
| endpoint test parallel transfers ctrl      | N/A                     |
| endpoint test abort                        | N/A                     |
| endpoint test data toggle reset            | N/A                     |

I think the intermittent failure of the `usb control stall test` is caused by a race condition of some sort. 

My guess is that when a setup packet comes in, my `USBPhy` implementation sets up the hardware to enter either the data stage or the status stage based on the setup packet content. Depending on the latency of the USB host/bus, there may be enough time for the Mbed USB stack to `stall` the control endpoint (as expected) before the host tries the next transaction. If not, the hardware has already transitioned to the data/status stage and will automatically `ACK` transactions from the host... and then the host fails the test.

I have to spend a bit more time researching the other test cases. 

As I said, I'm going through the USB stack execution flow and charting it out. As I go along, I'm trying to identify places where there should be a critical section/synchronization to prevent race conditions. @c1728p9 @cmonr Any advice on this? I'm not 100% clear on how to properly identify where critical sections should go. I've read it's wherever "shared state" variables are accessed, but that seems pretty vague. Please point me to any good resources on synchronization/critical sections in embedded/interrupt-driven systems if you can.

The flowcharts will also help me ensure I am interacting with the Nordic hardware as it expects. You can [check out what I've mapped out so far here](https://drive.google.com/file/d/1MUVlNCkD-gm08xFkWmP5YhTIHQuVvt7z/view?usp=sharing).

USB is a lot to fit all in your head at once :confounded:

c1728p9, 2019-01-14:
> My guess is that when a setup packet comes in, my USBPhy implementation sets up the hardware to enter either the data stage or the status stage based on the setup packet content.

After the setup packet has been received the device should be NAKing, giving the software as much time as it needs to respond with either an ACK or a STALL. One way you can make this error more reproducible is to intentionally introduce a delay before sending the stall. USB hardware should be able to handle a +10ms delay between receiving a setup packet and responding with a stall. If with this delay the failures always occur, then there may be code that is setting ACK too early that needs to be removed.

> I'm trying to identify places where there should be a critical section/synchronization to prevent race conditions.

USBDevice serializes all access to the USBPhy with a critical section so you shouldn't need to provide any protection at the USBPhy layer.

> Please point me to any good resources on synchronization/critical sections in embedded/interrupt-driven systems if you can.

There is a general architecture page on synchronization at https://os.mbed.com/docs/mbed-os/v5.11/reference/thread-safety.html. 

> The flowcharts will also help me ensure I am interacting with the Nordic hardware as it expects.

I'm having a hard time seeing things when I zoom in on this image. Could you upload it as a .svg image or the raw draw.io xml?

AGlass0fMilk, 2019-01-14:
@c1728p9 Thanks for the tips, I'll look into that soon.

I shared the raw draw.io diagram but I guess it makes it a really tiny picture if you share it as read only.

[Here's a read/write link](https://drive.google.com/file/d/1MUVlNCkD-gm08xFkWmP5YhTIHQuVvt7z/view?usp=sharing). Hopefully that makes it more legible.

AGlass0fMilk, 2019-01-15:
@c1728p9 The reason I included critical sections in my USB implementation in the first place is because I started with the USBPhy_Template implementation. During the example `_usb_isr` and `process` implementations there are calls to disabled/enable USB interrupts in the NVIC:

https://github.com/ARMmbed/mbed-os/blob/8b6fffb8a869c5451adfe24a168386a3b9426806/usb/device/targets/TARGET_Template/USBPhy_template.cpp#L300-L309

Does your comment above mean this isn't necessary anymore in the USBPhyHw implementation?

AGlass0fMilk, 2019-01-15:
@fkjagodzinski concerning the ISO endpoint 0 size issue you pointed out, I have posted a parallel ticket on Nordic's Devzone website: https://devzone.nordicsemi.com/f/nordic-q-a/42572/usb-iso-endpoint-max-packet-size-of-0

c1728p9, 2019-01-16:
> Does your comment above mean this isn't necessary anymore in the USBPhyHw implementation?

The NVIC calls in `_usb_isr` and `process` are disable further USB interrupts while processing the current one. This is still good to have so I wouldn't recommend removing it.

When I say that USBDevice serializes all access to the USBPhy I mean that all calls it makes are protected by a lock. For example, [this call to phy->process()](https://github.com/ARMmbed/mbed-os/blob/8b6fffb8a869c5451adfe24a168386a3b9426806/usb/device/USBDevice/USBDevice.cpp#L1557) is protected by the `lock()` and `unlock()`. In USBDevice every call to USBPhy either calls `lock()` and `unlock()` explicitly or asserts that a lock is already held with a call to `assert_locked()`. In the case of `_usb_isr` this is a function getting called by an interrupt rather than by USBDevice, so you'll want to disable the USB interrupt and call `start_process` so USBDevice can continue processing it in a serialized manner.

> Can confirm if I take out the disable/enable of USB interrupts it breaks my code.

I recommend you keep the disable/enable of USB interrupts for the reason mentioned above. That said, I'm a bit surprised that not having this causes problems. The function `start_process()` first calls [lock()](https://github.com/ARMmbed/mbed-os/blob/8b6fffb8a869c5451adfe24a168386a3b9426806/usb/device/USBDevice/USBDevice.cpp#L1564) ,which uses a critical section to prevent all interrupts, and then calls `process()`. Since `process()` is called from a critical section, USB interrupt shouldn't be able to interfere.

c1728p9, 2019-01-16:
> You should be able to click "open in draw.io". If not, you should be able to save it and import it yourself.

The open in draw.io link didn't work for me, but I was able to save it and import it manually. I can see everything in detail now.

Looking up close at this diagram I see several different interrupts can occur such as `USBPWR
Detected` and `USBPWRRDY Event`. This may explain why removing the NVIC calls in `_usb_isr` and `process` breaks the code. Are you calling into USBPhyEvents directly from these interrupts?

AGlass0fMilk, 2019-01-19:
I removed the `size == 0` assert and now I'm hitting a different error in the Nordic stack. I don't think SDK14.2 supported the ISO endpoints very well. :man_shrugging: 

The culprit appears to occur in the setup process for the `USBEndpointTester` used for all the `endpoint test ...` test cases. At some point, the host requests the device to setup a new interface. During removal of the old endpoints, a call is made to `nrf_drv_usbd_ep_disable` with endpoint 8 (EPOUT8), which for the nRF52840 is actually the ISO OUT endpoint.

Inside the Nordic driver, eventually a call is made to `nrf_usbd_epout_clear`... this is where the ASSERT happens. The Nordic driver checks to see if the endpoint index is within the bounds of the EPOUT register array... which it isn't because the hardware handles the ISO OUT and ISO IN endpoints slightly differently.

I'm not sure how Nordic intended the user to remove ISO endpoints, I haven't seen any ISO-specific remove API call. Next step is to check if SDK15.2 has a patch that addresses this issue and try to merge it in. If not, then I guess I have another bug to report to Nordic :bug:

AGlass0fMilk, 2019-01-22:
It looks like what's happening is during interface reconfiguration, in the function call `USBEndpointTester::_setup_non_zero_endpoints`, a call is made to `read_start` on the ISO endpoint. When `_phy->endpoint_read` is called, the Nordic driver returns that it was busy and the read start failed. This prevents the `info->pending` counter from being incremented, but does not cause a direct test failure.

So when the next transaction comes on the bus, the driver gets stuck... I'm still working on investigating this further.

c1728p9, 2019-01-23:
Thanks for the updates @AGlass0fMilk!

AGlass0fMilk, 2019-01-23:
Yeah, I'm kinda just using this issue thread to jot down my ideas/thoughts during debugging. So some stuff may not be fully thought through.

I'm going to see why the Nordic driver returns that it is busy. My first idea for how to mitigate this is to just always keep _all_ endpoints enabled and mask interrupts that the upper layers of the stack haven't enabled. This way there should be no delay in when the driver is ready to start transfers on any endpoint after being added. If the device is connected over USB then the potential slight increase in power consumption should be negligible anyway...

loverdeg-ep, 2019-01-23:
@AGlass0fMilk Please let know if there is any individual efforts within this that you could use assistance on.
We have a resource here who will need this working at some point, and they may have time to spare.

AGlass0fMilk, 2019-01-23:
@loverdeg-ep Any help would be great! I'm doing this as a side project and I don't get as much time to work on it as I'd like. 

For reference, I am using a [Rigado BMD340 development kit ](https://www.digikey.com/product-detail/en/rigado-inc/BMD-340-EVAL/1604-1027-ND/8676236 )as a target and a [TotalPhase Beagle USB12](https://www.totalphase.com/products/beagle-usb12/) protocol analyzer to trace issues on the bus.

I am hoping my implementation is getting very close... control transfers seem to be pretty solid. If you guys want to reach me via email we can collaborate in a bit more detail.

george.beckstein@gmail.com

fkjagodzinski, 2019-01-28:
> The good news is that now `usb soft reconnection test` has started passing. @fkjagodzinski was anything changed in the test code related to that?

Hi @AGlass0fMilk, the test case you mentioned hasn't been updated recently.

AGlass0fMilk, 2019-01-31:
After wrestling with strange issues that are hard to debug, I'm starting to think the Nordic driver is not directly compatible with the Mbed USB HAL specification. The Nordic driver operates on entire transfer objects and expects to get the entire payload all at once. As a result, Nordic paces itself on the DMA events (which is nice for speed considerations).

Mbed handles breaking up the packets into multiples of `max_packet_size` and hands off these chunks to the USB HAL. Mbed is expecting to only get notified of a completed transfer (USBPhyEvents::in/out()) after a data transfer has been ACK'd by either side.

My theory is that the Nordic driver is notifying the upper layers of the stack that the transfer has been completed when in reality only the DMA transfer has completed. Most of the time this is okay because it is closely coupled with when a transfer _is_ completed. But if something happens that messes up this timing, bad things happen (eternally waiting IN/OUT endpoints, too many IN/OUT events, etc).

Mbed requires less intelligence in its USB HAL layer. So I think it would be easier to implement a custom USB HAL rather than continue trying to make the Nordic driver fit... At least I learned a bunch about USB and how the Nordic driver works.

I'm working out a program flow chart on draw.io again. Comment back if anyone wants to see/help with it.

loverdeg-ep, 2019-02-01:
Good work. 
So summary of your efforts thus far is:

* NRF SDK15's USB driver for NRF52 looks incompatible with Mbed USB HAL specification
* Quickest path forward might be developing a basic USB driver in place of Nordic's (I have maintenance fears but I suppose it would be better than nothing)

Our schedule hasn't worked out such that we've been able to apply our resource to this yet.

AGlass0fMilk, 2019-02-02:
Well, I guess it would be possible to rework the Nordic driver some. 

I think I've narrowed down the problem to this: the Nordic driver is currently notifying the upper layer that a transfer is "finished" when just the DMA transfer has finished. In reality, the host has not yet received/ACK'd the packet.

In most cases this is okay, but sometimes (like with 0-length or short transfers) the DMA transfer is done almost immediately and can cause the upper layer to take premature actions and become desynchronized.

It should be possible to modify the Nordic driver to notify the upper layers when a transfer is actually finished (when EPDATA or EP0DATADONE tasks complete).

Right now I'm getting intermittent failures during the regular endpoint tests. Usually what happens is I hit an assert in USBDevice.cpp at line 947 or 965. This assert makes sure there was a pending transfer when the USB HAL sends an `in` or `out` event.

AGlass0fMilk, 2019-02-02:
@c1728p9 I've also gotten intermittent `[Errno 32] Pipe error`'s and when I googled it I found libusb/libusb#241 :laughing:

AGlass0fMilk, 2019-02-02:
I worked on the Nordic-based implementation a bit more. I found a major bug that would cause the `control basic test` to fail. I never implemented `USBPhyHw::unconfigure` because I thought I didn't need it. Turns out the Mbed USB stack doesn't call `endpoint_remove` when going to the unconfigured/configuration 0 state. So the Nordic driver would never abort the last queued transfers on an endpoint and return `NRF_DRV_BUSY` when the next configuration change attempted to queue transfers.

I have updated the original comment test table... 4 tests are failing still. Still looking into them.

AGlass0fMilk, 2019-02-02:
I realized I was compiling with an older version of `USBEndpointTester.cpp` and running a newer version of the host test python script. There was a change in the format of a vendor-specific request and it was causing the `endpoint test data toggle reset` test to stall on the control endpoint. That test is now consistently passing.

This was also happening during the `endpoint test halt` test. This test is still failing for another reason though...

AGlass0fMilk, 2019-02-04:
My understanding of the Nordic driver has changed a bit after reviewing the datasheet more closely. The driver notifies the user code when a DMA transfer is finished for OUT transfers, which is correct because then the data is ready to be processed. For IN transfers, the user code is notified when the host ACK's the packet.

I have a few things on my todo list to hopefully iron out the rest of the bugs:
1.) Add asserts to the USBPhyHw implementation to catch the Nordic driver returning anything but `NRF_DRV_SUCCESS`
2.) Rework the control transfer code a bit. Right now I think it not only triggers the hardware-handled `status` stage but also triggers a zero-length transfer that may be causing issues.
3.) Run through the tests a few times and capture bus traffic during failures. I can post screenshots and the data files up here so maybe someone can suggest what may be causing a failure.

AGlass0fMilk, 2019-02-22:
I added a "virtual status stage" that is triggered when the opposite control transfer is initiated by the Mbed stack (signifying the status stage). There were issues with how status stage `ep0_in` and `ep0_out` callbacks were being sent since the Nordic driver doesn't actually notify the user when the status stage ACK has completed.

I have run the automated tests a few times. The first two times were very promising -- ** all tests passed!**

But after running it a few more times there are intermittent failures in the endpoint data transfer code that have to be debugged.

The control transfer code seems to be really fast and rock solid. Perhaps the endpoint data transfer intermittent failures are related to control transfers still.

 I will look into this later on.

AGlass0fMilk, 2019-02-25:
So I looked at the Nordic documentation website today and noticed that as of 1/31/2019 they have added a few errata pertaining to essentially all IC revisions (including IC rev 1 and rev 2)

One in particular caught my eye:
`[199] USBD: USBD cannot receive tasks during DMA`

GAHHH! I hope this can explain some of the weird behavior I've been seeing, including random hangs (where an event doesn't happen to queue an `IN` transfer) and random data correctness failures where the Nordic rarely writes back incorrect data.

Luckily there is a workaround mentioned. I'll try adding it into the Nordic driver when triggering DMA transfers and see if the tests fail less frequently.

AGlass0fMilk, 2019-02-26:
All tests _can_ pass. The control transfer tests always pass, I have not seen a failure there with the updated code yet.

The intermittent failures I am seeing have the following characteristics:

## Intermittent Failure - EP Halt Test Incorrect Payload
My understanding of the `endpoint test halt` test is:

1. OUT xfer - write a random payload to device endpoint
2. IN xfer - read back payload
3. Make sure data written and data read are the same
4. ...continue to loop...
5. OUT xfer - write a random payload to device endpoint
6. HALT IN endpoint
7. IN xfer - endpoint should reply STALL
8. CLEAR HALT on IN endpoint
9. OUT xfer - write a **new** random payload to device endpoint
10. IN xfer - read back payload
11. Make sure the **new** random payload and the data read are the same

**My theory:**
The failure occurs because the Mbed stack immediately loads the payload received during the first OUT stage into EasyDMA and begins the transfer for the corresponding IN endpoint. Then the endpoint is halted, it replies STALL as expected. But when the HALT is cleared, the hardware is still prepared to automatically ACK the next IN transfer that begins on that endpoint. It will then respond with the old payload -- causing the test to fail.

See screenshot of captured USB traffic below:
[image]

Sometimes the test passes, probably when the new data transfer is fast enough to overwrite the old payload in the DMA buffer before the IN transfer comes in.

This may be a point of discrepancy between the Mbed API spec and the Nordic USBD hardware capabilities -- Nordic does not present a clear way of canceling a "loaded" auto-ACK once the DMA transfer has been initiated. I am trying some hacky methods (disabling/reenabling an endpoint during a stall) to accomplish this, but I'm not sure if there's a solution right now.

Any ideas?

See my[ cross-post on Nordic Devzone](https://devzone.nordicsemi.com/f/nordic-q-a/44117/nrf52840-usb-cancel-in-transfer).

## Intermittent Failure - EP Parallel Ctrl Timeout

Sometimes during parallel transfers endpoint 1 IN will time out. I've seen it with other endpoints but I haven't collected enough definitive data.

It may be related to a hardware issue... there was an errata that was "fixed" where EPDATA events would not be generated. Maybe it still happens occasionally.

My other theory is that somehow the DMA scheduling algorithm (which essentially just loops through pending endpoints starting in a certain order) never reaches the EP1 IN DMA transfer since it is so busy with EP0 and EP2 IN/OUT transfers.

See screenshot below:
[image]

## Real Testing
These issues seem to be corner cases that affect a small percentage of transfers/transfer patterns. One out of every 150,000 transfers (< 0.001%)

I've done a little real testing using currently implemented classes (USBCDC, USBMSD) and haven't seen any apparent issues. I haven't done too much testing admittedly.

It was cool being able to hook up a bunch of Mbed APIs and have a working nRF52840-DK "flash drive" in under an hour. (QSPIFBlockDevice + USBMSD + FATFileSystem = 8.2MB flash drive!)

I'm not sure where to go from here. I've spent a lot of time wrestling with the Nordic hardware, I think I need some outside input.

dlfryar, 2019-02-26:
@AGlass0fMilk are using production boards not the ones labeled preview?  We had to work through spurious issues like you are seeing with QSPI and eDMA some of which turned out to be errata related to the preview boards.

@c1728p9 solved one of these issues where the UART was overflowing by turning down the DMA.
https://github.com/ARMmbed/mbed-os/pull/8784

[38 more replies]

closed 2019-06-18 by AGlass0fMilk