ARMmbed/mbed-os #7679: UBLOX cellular api's for UDP and TCP do not work for Sara R4 (CATM1)

NeilMacMullen, 2018-08-02:
This comments are raised in relation to https://github.com/ARMmbed/mbed-os/pull/7619.  Since the 3G (U201) modem appears to work well with those changes, I suggest that PR is merged and these issues are addressed separately...

First, the R4 does not support the UPSND command.  Something similar to the following changes is required.  For expediency I've use a macro to detect the difference between boards but it would be much better if the modem type could be detected dynamically.

```

 --- a/features/cellular/framework/targets/UBLOX/AT/UBLOX_AT_CellularNetwork.cpp
 +++ b/features/cellular/framework/targets/UBLOX/AT/UBLOX_AT_CellularNetwork.cpp
 @@ -128,7 +128,7 @@ nsapi_error_t UBLOX_AT_CellularNetwork::open_data_channel()
      if (!_stack) {
          return err;
      }
 -
 +#ifndef SARA_R4
      _at.cmd_start("AT+UPSND=" PROFILE ",8");
      _at.cmd_stop();
      _at.resp_start("+UPSND:");
 @@ -136,7 +136,10 @@ nsapi_error_t UBLOX_AT_CellularNetwork::open_data_channel()
      _at.read_int();
      active = _at.read_int();
      _at.resp_stop();
 -
 +#else
 +    tr_warn("Skipping upsnd and assuming profile ok\n");
 +    return NSAPI_ERROR_OK;
 +#endif
      if (active == 0) {
          // If the caller hasn't entered an APN, try to find it
          if (_apn == NULL) {
 @@ -170,6 +173,12 @@ bool UBLOX_AT_CellularNetwork::activate_profile(const char* apn,
                                                  const char* username,
                                                  const char* password)
  {
 +
 +#ifdef SARA_R4
 +tr_warn("skipping profile activation");
 +return true;
 +#endif
 +
      bool activated = false;
      bool success = false;

 --- a/features/cellular/framework/targets/UBLOX/AT/UBLOX_AT_CellularStack.cpp
 +++ b/features/cellular/framework/targets/UBLOX/AT/UBLOX_AT_CellularStack.cpp
 @@ -521,12 +522,20 @@ void UBLOX_AT_CellularStack::clear_socket(CellularSocket * socket)
  const char * UBLOX_AT_CellularStack::get_ip_address()
  {
      _at.lock();
 +#ifndef SARA_R4
      _at.cmd_start("AT+UPSND=" PROFILE ",0");
 -    _at.cmd_stop();
 -
 -    _at.resp_start("+UPSND:");
 +	_at.cmd_stop();
 +	_at.resp_start("+UPSND:");
 +#else
 +    _at.cmd_start("AT+CGPADDR=" PROFILE );
 +	_at.cmd_stop();
 +	_at.resp_start("+CGPADDR:");
 +#endif
 +   
      if (_at.info_resp()) {
 +#ifndef SARA_R4
          _at.skip_param();
 +#endif
          _at.skip_param();
          int len = _at.read_string(_ip, NSAPI_IPv4_SIZE-1);
          if (len == -1) {

```

The R4 also (incorrectly) garbles linefeeds when returning data in response to AT+USORD and this confused ATHandler.  The following changes allow the 'resp' method in ATHandler to cope with this...

```

 --- a/features/cellular/framework/AT/ATHandler.cpp
 +++ b/features/cellular/framework/AT/ATHandler.cpp
 @@ -661,6 +661,21 @@ bool ATHandler::match(const char *str, size_t size)
      return false;
  }
  
 +
 +void ATHandler::matchcrlf()
 +{
 +    rewind_buffer();
 +   
 +    while (_recv_pos < _recv_len)
 +    {
 +        char ch =_recv_buff[_recv_pos];
 +        if (ch != '\r' && ch != '\n')
 +        break;
 +        _recv_pos ++;
 +    } 
 + 
 +}
 +
  bool ATHandler::match_urc()
  {
      rewind_buffer();

 @@ -775,10 +790,10 @@ void ATHandler::resp(const char *prefix, bool check_urc)
      _prefix_matched = false;
      _urc_matched = false;
      _error_found = false;
 -
 + 
      while (!get_last_error()) {
  
 -        match(CRLF, CRLF_LENGTH);
 +        matchcrlf();
  
          if (match(OK, OK_LENGTH)) {
              set_scope(RespType);
  

  
  --- a/features/cellular/framework/AT/ATHandler.h
  +++ b/features/cellular/framework/AT/ATHandler.h
  @@ -458,6 +458,8 @@ private:
       // If URC match sets the scope to information response and after urc's cb returns
       // finishes the information response scope(consumes to CRLF).
       bool match_urc();
  +	// match any number and combination of cr/lf
  +	void matchcrlf();
       // Checks if any of the error strings are matching the receiving buffer content.
       bool match_error();
       // Checks if current char in buffer matches ch and consumes it,
 

```
Finally, R4 does not appear to support AT+UDNSRN required for hostname to IP address lookup.  I'll address this further in a separate issue.

NeilMacMullen, 2018-08-02:
Adding @TeemuKultala , @MarceloSalazar

wajahat-ublox, 2018-08-03:
@NeilMacMullen 
The revision history of latest [AT command manual](https://www.u-blox.com/sites/default/files/SARA-R4-SARA-N4_ATCommands_%28UBX-17003787%29.pdf) of SARA-R4 indicates that UDNSRN is supported now.

NeilMacMullen, 2018-08-03:
@wajahat-ublox  Interesting.  We're using modules that were shipped only a short time ago I believe so I'm not sure this change is available yet.  Do you know how/if we can upgrade the firmware to support this feature? (Preferably over the air!)

mudassar-ublox, 2018-08-03:
Currently we have added changes for target SARA_U201 and we have plan to add target support in `CellularTarget.h` for SARA_R4 once PR [#7619](https://github.com/ARMmbed/mbed-os/pull/7619) is merged. 

The changes you are suggesting for SARA_R4 already handled. `[open_data_channel](https://github.com/u-blox/mbed-os/blob/a8abeccdacb019c852335b19b652102155f322bc/features/cellular/framework/targets/UBLOX/AT/UBLOX_AT_CellularNetwork.cpp#L95)` is called from `connect` function and in `connect` function we are doing same as you mentioned. Similarly for [`activate_profile`](https://github.com/u-blox/mbed-os/blob/a8abeccdacb019c852335b19b652102155f322bc/features/cellular/framework/targets/UBLOX/AT/UBLOX_AT_CellularNetwork.cpp#L157) function is called form `open_data_channel`. Regarding `get_ip_address` will look into it.

Regarding AT+USORD response I have not experienced such thing, will check again when we add SARA_R4 support and if problem exists then we will raise this issue with arm separately.

mudassar-ublox, 2018-08-17:
Above changes are irrelevant when target is `C030_R104M`. As `open_data_channel` function will not be called when target is `C030_R104M`.

https://github.com/ARMmbed/mbed-os/blob/f5126682308d31c60851943168c79949e44774d7/features/cellular/framework/targets/UBLOX/AT/UBLOX_AT_CellularNetwork.cpp#L95

screamerbg, 2018-09-10:
@mudassar-ublox PR https://github.com/ARMmbed/mbed-os/pull/7619 was merged. What's the next step?

screamerbg, 2018-09-12:
CC @asifrizwanubx @fahim-ublox

fahim-ublox, 2018-09-13:
Work has been started on Cellular APIs for Sara-R4, I will come back on it soon

Roger-Strassburg, 2018-09-14:
> Regarding AT+USORD response I have not experienced such thing, will check again when we add SARA_R4 support and if problem exists then we will raise this issue with arm separately.

I saw this problem on an earlier version of the R410M modem. What would happen is that on a binary read, if the modem saw a LF it would insert a CR in front of it. I worked around it at the time by ignoring CR if followed by a LF.

We're now using firmware version 05.05 on 02B. It was my understanding from ublox that this problem is fixed in this and/or an earlier version.

NeilMacMullen, 2018-09-14:
@Roger-Strassburg  - do you know if the modem supports over-the-air upgrade?  Unfortunately we have a number of devices which have been build with the older firmware.

Roger-Strassburg, 2018-09-14:
The release notes for FW version 05.02 / 02B has the following note:

Tools, updating of firmware:

In order to flash update the firmware, USB interface to modem and EasyFlash tool are required. Please contact u-blox support team. Note that the EasyFLash firmware upgrade takes approximately 4-5 minutes and should be completed without a power disruption, as a power loss during firmware upgrade may render the device unusable.
Prototype sample firmware will not be upgradable to ES (Engineering Sample) firmware via the FOTA/FOAT method. Prototype sample firmware will be upgradable to ES via EasyFlash tool as per above description.

We updated our modules from FW version 05.02 to 05.05 via the EasyFlash tool.

Note that only modules produced on or after calendar week 38, 2017 can be updated in this manner. You can see the production week at the top of the module label. It needs to be 17/38 or later.

Roger

The information contained in this message is confidential.
Data security advice: Please be aware, that data exchange via Internet or E-Mail is not 100% safe. Therefore, please handle sensitive or confidential (bank) data with care. Please take note, that Rosenberger is not asking you for bank data or providing you such data via E-Mail, in principle. In case of deviations or questions please contact us immediately.

**
Rosenberger Hochfrequenztechnik GmbH & Co. KG, Registergericht: Amtsgericht Traunstein, HRA 5566

persönlich haftender Gesellschafter:
Rosenberger Hochfrequenztechnik GmbH, Registergericht: Amtsgericht Traunstein, HRB 4405

Geschäftsführer: Dipl.-Kfm. Hans Rosenberger, Dipl.-Ing. Bernhard Rosenberger, Dipl.-Ing. Peter Rosenberger, Dr. rer. nat. Tosja Zywietz

Disclaimer http://www.rosenberger.com/Disclaimer/disclaimer.html

Please consider our data policy: https://www.rosenberger.com/data-policy

NeilMacMullen, 2018-09-14:
@Roger-Strassburg   Roger - thanks for this.  I believe our modules were made in 2018 so hopefully the FOTA route will be possible.

Roger-Strassburg, 2018-09-14:
@NeilMacMullen 
My understanding is that FOTA is available at FW version 05.05 and beyond. 05.05 came out in late January, so I assume it would've been on modules built shortly after that. You can see the week it was built on the top edge of the module label in yy/ww format, e.g. 18/01 for the first week of 2018.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-6056*

mudassar-ublox, 2019-05-23:
Cellular libraries was updated to SARA-R4. 
[ONBOARD_UBLOX.cpp](https://github.com/ARMmbed/mbed-os/blob/b2abfc3529f2ba0cbee2c62c37c97760537c34c3/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F437xG/TARGET_UBLOX_C030/ONBOARD_UBLOX.cpp#L28)

Following are cellular api's for Sara-R4.
[Sara-R4-Api](https://github.com/ARMmbed/mbed-os/tree/master/features/cellular/framework/targets/UBLOX/AT)

0xc0170, 2019-05-23:
@mudassar-ublox This issue was resolved and can be closed?

mudassar-ublox, 2019-05-24:
Yes you can close this issue.

closed 2019-05-24 by 0xc0170