ARMmbed/mbed-os #7680: SocketAddress should be extended to support address-less host names

NeilMacMullen, 2018-08-02:
The UBLOX SARA-R4 CATM1 modem includes an on-board IP stack but does not provide an AT command to perform a DNS lookup from a host name.  The cellular stack passes a SocketAddress to a socket-open implementation and the assumes that this contains an IP address which has already been retrieved.  

Here's some representative code from UBLOX_AT_CellularStack::socket_connect
```
   _at.cmd_start("AT+USOCO=");
    _at.write_int(socket->id);
   _at.write_string(addr.get_ip_address(), true);
    _at.write_int(addr.get_port());
```
The R4 is quite happy to accept a hostname rather than an IP address as the argument to USOCO but there is no way in the current SocketAddress API to pass this through the stack.

NeilMacMullen, 2018-08-02:
Copying @TeemuKultala  @MarceloSalazar

NeilMacMullen, 2018-08-02:
Basic changes to support hostname address type. I haven't checked closely whether all semantics are sensibly maintained.  Also, it may be unwise to overload get_ip_address in this way since clients may assume the maximum size of the returned string is the size of an IPV6 address.

```
--- a/features/netsocket/SocketAddress.cpp
+++ b/features/netsocket/SocketAddress.cpp
@@ -147,6 +147,15 @@ bool SocketAddress::set_ip_address(const char *addr)
         return true;
     } else {
         _addr = nsapi_addr_t();
+        if(addr)
+        {
+         _addr.version = NSAPI_HOSTNAME;
+         int len = strlen(addr);
+         _ip_address = new char [len +1];
+         strcpy(_ip_address,addr);
+
+
+        }
         return false;
     }
 }
@@ -234,7 +243,7 @@ SocketAddress::operator bool() const

         return false;
     } else {
-        return false;
+        return (_addr.version == NSAPI_HOSTNAME && _ip_address);
     }
 }

@@ -258,7 +267,9 @@ bool operator==(const SocketAddress &a, const SocketAddress &b)
     } else if (a._addr.version == NSAPI_IPv6) {
         return memcmp(a._addr.bytes, b._addr.bytes, NSAPI_IPv6_BYTES) == 0;
     }
-
+    else if (a._addr.version == NSAPI_HOSTNAME) {
+        return strcmp(a._ip_address, b._ip_address) == 0;
+    }
     MBED_UNREACHABLE;
 }
```

mudassar-ublox, 2018-08-03:
Host name resolution is supported for SARA-R4, [AT Command Manual](https://www.u-blox.com/sites/default/files/SARA-R4-SARA-N4_ATCommands_%28UBX-17003787%29.pdf).

In `gethostbyname` function, IP is retrieved and set as well. 
https://github.com/u-blox/mbed-os/blob/a8abeccdacb019c852335b19b652102155f322bc/features/cellular/framework/targets/UBLOX/AT/UBLOX_AT_CellularStack.cpp#L573

NeilMacMullen, 2018-08-03:
@mudassar-ublox It is NOT supported in the modules that are currently being shipped to us!

philwareublox, 2018-08-03:
+UDNSRN will be supported in later firmware versions, but for now we will need to work around this issue.

NeilMacMullen, 2018-08-12:
@SeppoTakalo   I understand the rationale for not wanting to change SocketAddress but I don't think your alternative addresses the issue.  The problem here is that for this particular modem (and modem firmware version) the IP stack is running entirely on-chip and is opaque to mbed _including_ the resolved IP addresses.  What is exposed to the mbed host is a socket-like endpoint that is opened with a hostname but which has no IP address visible to mbed.  Maybe this is really a change for the cellular team to address but it does seem to me that many network interfaces are capable of doing 'transparent' DNS lookup so the concept of an endpoint may be useful one.

SeppoTakalo, 2018-08-28:
Mbed OS already has internal DNS resolver that is independent of the Network stack.
https://os.mbed.com/docs/v5.9/reference/dns-resolver.html

If more optimal resolver is provided by the Module, it might override the `gethostbyname()` functions.

However, if the module, as in this case, is able to work with symbolic names instead of IP addresses, that causes API changes to the user level, which I'm not happy to support.
The benefit of the API change would be that modem does DNS query, instead of our driver.
Same amount of UDP packets send to DNS server.
Outcome?
Less UART traffic perhaps, and maybe Cortex-M core idling while DNS query ongoing.
So not much of optimization.

At least if we keep the fact in mind that DNS query is done only once per connection. And in case of long idling TCP streams, it might be less than one in a day.

Currently this is the model that ALL socket communication uses:
```
NetworkInterface *net = NetworkInterface::get_default_instance();

// This phase might happen only once in the connection lifetime
SocketAddress addr;
net->gethostbyname("server.example.com", &addr);
addr.set_port(port);

// This phase might happen more often.
TCPSocket sock(net);
sock.connect(addr);
```

closed 2018-10-02 by adbridge