ARMmbed/mbed-os #7682: UBLOX SARA R4 module requires difference command to open PPP data channel

NeilMacMullen, 2018-08-02:
When using LWIP, the following changes are required to enter PPP mode.  (Note there still appears to be an issue that prevents data flowing when this is done.)
```

--- a/features/cellular/framework/AT/AT_CellularNetwork.cpp
+++ b/features/cellular/framework/AT/AT_CellularNetwork.cpp
@@ -381,8 +381,15 @@ nsapi_error_t AT_CellularNetwork::open_data_channel()
 {
 #if NSAPI_PPP_AVAILABLE
     tr_info("Open data channel in PPP mode");
+#ifndef SARA_R4
     _at.cmd_start("AT+CGDATA=\"PPP\",");
     _at.write_int(_cid);
+#else
+     _at.cmd_start("ATD*99***1#");
+
+    _at.cmd_stop();
+    _at.cmd_start("ATD*99***1#");
+#endif
     _at.cmd_stop();
 
     _at.resp_start("CONNECT", true);
```

NeilMacMullen, 2018-08-02:
See also https://github.com/ARMmbed/mbed-os/issues/7464

NeilMacMullen, 2018-08-02:
Adding @TeemuKultala @MarceloSalazar

0xc0170, 2018-08-03:
cc @ARMmbed/team-ublox

AriParkkila, 2018-08-28:
@NeilMacMullen do you think we can close this issue. PR #7870 implements support for GPRS dial-up command and that's now merged.

closed 2018-10-02 by adbridge