ARMmbed/mbed-os #7684: socket API using same lock for send + recv

drahnr, 2018-08-03:
### Description
gcc 7.x.y
mbed-os 5.9.4

With the current API there is one lock used for send + recv, essentially rendering it unuseable and killing the full duplex feature of most communication channels.
Instead of two threads for recv() + queue + send() (for drivers which can do that) one now has to use essentially do polling with recv() and a small timeout value.

What is the motivation behind this?

### Issue request type
[x] Question  
[ ] Enhancement  
[x] Bug

SeppoTakalo, 2018-08-03:
> What is the motivation behind only having a single lock for send and recv?

Memory, mostly.
Also some drivers are not capable of reading and writing at the same time.

Can you tell me that case where this current behaviour is actually not working? Even if the Socket API is set to blocking mode, all underlying stack operations are non-blocking. All IP stacks we have are event based.
So in case when `recv()` or `send()` would block, it actually releases the mutex and starts to wait for event flag.
https://github.com/ARMmbed/mbed-os/blob/master/features/netsocket/TCPSocket.cpp#L142..L146

So even that they share the same mutex, they do not block each others.

drahnr, 2018-08-03:
The assumption there is that the driver implementing `virtual socket_recv` actually returns and yields an event, which might very well not be the case, in case of an SDK which is blocking.

SeppoTakalo, 2018-08-03:
It is pretty safe assumption that stack providers implement their stacks as documented.

Non-blocking behaviour is documented in `struct nsapi_stack_api` so there could not be any confusion here:
https://github.com/ARMmbed/mbed-os/blob/master/features/netsocket/nsapi_types.h#L483..L484

SeppoTakalo, 2018-08-03:
Also documented in `NetworkStack` class https://github.com/ARMmbed/mbed-os/blob/master/features/netsocket/NetworkStack.h#L274..L275

drahnr, 2018-08-03:
The mechanism of signaling as soon data is ready is the part that is not documented overly conclusive from an implementors POV.

SeppoTakalo, 2018-08-03:
True, its not very throughly explained.

The behaviour is explained in documentation of `attach()` API, so if someone does full porting of NetworkStack, they understand the behaviour only when `attach()` is ported.

https://github.com/ARMmbed/mbed-os/blob/master/features/netsocket/NetworkStack.h#L322..L335
https://github.com/ARMmbed/mbed-os/blob/master/features/netsocket/nsapi_types.h#L554..L569

When quickly looking up just one API documentation, this underlying behaviour is easily missed. We probably need better porting documentation of NetworkStack if there would be more external IP stacks to be ported. However its not that common, and so far all the devices seems to be doing it okayish.

drahnr, 2018-08-03:
I find this quite confusing because it is not clear what purpose this callback serves, it is explain when it may be called, but does not answer the purpose of what it does. It's also not clear why this is the only function having a callback so this is something discarded rather quickly.

(Also: I guess most embedded people skip that part entierly (open, connect, send, recv, close is the most common workflow for IoT devices I've run into so far) with an error value of NOT_IMPLEMENTED.

drahnr, 2018-08-06:
There is only this well hidden section on how to implement the feedback from the driver to the Socket API on state change. This is not obvious from the API documenation, but there is also no link to this page: https://os.mbed.com/docs/latest/reference/networkstack.html

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/ONME-3771*

closed 2018-11-20 by ciarmcom