ARMmbed/mbed-os #7697: Hard fault pointing into event queue code

pauluap, 2018-08-04:
### Description

mbed-os version 5.8.3
gcc version 6.3.1 20170215 (release) [ARM/embedded-6-branch revision 245512] (GNU Tools for ARM Embedded Processors 6-2017-q1-update) 

I have an application that has several threads sending events to a central dispatcher.  I'm running into a problem where the application crashes after multiple days of operation with a hard fault in `equeue_dequeue`

The crash handler output is:
```
++ MbedOS Fault Handler +++
FaultType: HardFault

Context:
R0   : 8AF85C66
R1   : 0257B693
R2   : 8D5012F9
R3   : 1550FFFF
R4   : C0003B78
R5   : 00000000
R6   : 00000000
R7   : 00000000
R8   : 00000000
R9   : 00000000
R10  : 00000000
R11  : 00000000
R12  : 0000B484
SP   : 20013360
LR   : 08002A65
PC   : 08002A5A
xPSR : 21000000
PSP  : 200132F8
MSP  : 2004FFB0
CPUID: 410FC241
HFSR : 40000000
MMFSR: 00000000
BFSR : 00000082
UFSR : 00000000
DFSR : 00000000
AFSR : 00000000
SHCSR: 00000000
BFAR : 15510013
Mode : Thread
Priv : Privileged
Stack: PSP

Thread Info:
Current:
State: 00000002 EntryFn: 08021421 Stack Size: 00001000 Mem: 200128A0 SP: 20013280
Next:
State: 00000002 EntryFn: 08021421 Stack Size: 00001000 Mem: 200128A0 SP: 20013280
Wait Threads:
State: 00000083 EntryFn: 080266B5 Stack Size: 00000300 Mem: 20002660 SP: 200028C8
Delay Threads:
State: 00000043 EntryFn: 08027081 Stack Size: 00000400 Mem: 20001100 SP: 20001340
Idle Thread:
State: 00000001 EntryFn: 08021565 Stack Size: 00000200 Mem: 20002460 SP: 200025F0

-- MbedOS Fault Handler --
```

Running the crash handler parser on the output obtains this:
```
Crash Info:
	Crash location = equeue_dequeue [0x08002A5A] (based on PC value)
	Caller location = equeue_dequeue [0x08002A65] (based on LR value)
	Stack Pointer at the time of crash = [20013360]
	Target and Fault Info:
		Processor Arch: ARM-V7M or above
		Processor Variant: C24
		Forced exception, a fault with configurable priority has been escalated to HardFault
```

And the section of the code corresponding to the addresses of the crash/caller locations is:
[image]

Looking at the contents of R3, `0x1550FFFF` when executing `0x08002A5A` it's supposed to point to either p or p->target, but given that the alignment of `0x1550FFFF` isn't on a 32-bit boundary, I'm figuring that something got corrupted, either the queue or the contents to the queue.  I raise some events using the `Event` class, others using the `queue->call()` function.

I have several failures land at the same location, so I'm assuming that generic memory corruption is not likely since if that was the case, I would have seen failures in other locations.

Hopefully somebody knows exactly what's going on, but if not, how would I go about finding out the original location of the call to dequeue - I'm figuring that it's part of the `dispatch_forever()` process of one of my queues.

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

kegilbert, 2018-08-04:
Pinging @geky

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-349*

evedon, 2019-05-24:
Sorry for the lack of response. As this is an old issue, I will close it. Please re-open if you can reproduce the issue with the latest Mbed OS release and you need further assistance.

closed 2019-05-24 by evedon