ARMmbed/mbed-os #7699: Stateful function objects for `Callback` size limitation

OmerRosler, 2018-08-05:
Is there a specific reason you ban stateful function objects of size larger then `sizeof(uintptr_t)` to be converted into `mbed::Callback`?

This assertion is defined in the macro `MBED_ENABLE_IF_CALLBACK_COMPATIBLE`

Similar callback libraries - `std::function`, `boost::function`, the propesed `inplace_function` and many more I don't know, enable arbitrary function objects sizes such that for objects too large for the SBO they need to be allocated on thd heap.

geky, 2018-08-06:
Yes, and it's an interesting problem.

The heap is not interrupt safe, and one of the primary uses for callbacks in Mbed OS is for interacting with interrupt context. By asserting, we can garuntee that any operations involving the callback class is interrupt safe.

If you need larger state, you can store it in a separate object or struct and pass it by pointer with the state overloads for the callback class.

There was also a discussion previously about providing a bind class that is opt-in for dynamic memory, but can still be passed around as a callback. However, there wasn't enough interest. This role can also be filled by the std::function class in the future.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-350*

closed 2019-10-31 by ciarmcom