ARMmbed/mbed-os #7702: Sub-optimal usage of volatile in CircularBuffer

cmorty, 2018-08-06:
### Description
The usage of volatile in `CircularBuffer` is suboptimal:

`_haed` and `_tail` are only accessed within critical sections. However them being `volatile` keeps the compiler from optimizing their access and causes them to be unnecessary accessed. For example 
```
_tail++;
_tail %= BufferSize;
```
will yield 2 read and 2 write accesses, while only one each is really necessary. There are two ways to fix this:
* Use a local variable `ltail=_tail; ltail++; ltail %= BufferSize; _tail=ltail;`
* Remove `volatile` from `_tail` and `_head` and use something like[ `ACCESS_ONCE`](https://lwn.net/Articles/508991/).

While the actual impact per call might be small, the circular buffer is typically called in loops.

Additionally `_pool` should actually be `volatile`, as it can change any time. While I currently cannot come up with a situation where the compiler could optimize accessing `_pool`, this does not mean there is none. Especially with improved compiler support for LTO.
Currently this should not be an issue as `_head` and `_tail` are volatile. But IMO it is a bug nonetheless.

### Issue request type
[ ] Question  
[ x ] Enhancement  
[ ] Bug  

[1]: https://lwn.net/Articles/508991/

pan-, 2018-08-06:
> Additionally _pool should actually be volatile, as it can change any time. While I currently cannot come up with a situation where the compiler could optimize accessing _pool, this does not mean there is none. 

No; `_pool` cannot change at any time given that accesses to the data member of the CircularBuffer are guarded by a `critical section`. Of course the `critical section` should set up the appropriate data and/or instruction barriers; I hope it does.

Note that `volatile` does offer little to no protection regards to the memory accessed as it doesn't issue memory barrier/synchronization instructions. It forces the compiler to add read or write accesses but these read or write access can very well be optimized by the hardware itself. 

`volatile` is only useful to ensure memory mapped I/O; it is certainly not a synchronization primitive. For synchronization; use a mutex, a critical section or even an atomic variable construct.

In the end, I think we can safely drop the volatile qualifier out of the `_head`, `_tail` and `_full` members; granted that critical section ensure the memory consistency.

0xc0170, 2018-08-06:
> In the end, I think we can safely drop the volatile qualifier out of the _head, _tail and _full members; granted that critical section ensure the memory consistency.

👍

cmorty, 2018-08-07:
> No; _pool cannot change at any time given that accesses to the data member of the CircularBuffer are guarded by a critical section. Of course the critical section should set up the appropriate data and/or instruction barriers; I hope it does.

`core_util_critical_section_enter` is only a mutex:
https://github.com/ARMmbed/mbed-os/blob/ffcb6ecfb5e769efecb1b5f3148b908d7e694a87/platform/mbed_critical.h#L59-L70 
and https://github.com/ARMmbed/mbed-os/blob/ffcb6ecfb5e769efecb1b5f3148b908d7e694a87/hal/critical_section_api.h#L32-L63

And why should it be anything else? If you don't understand how volatile works you shouldn't write critical sections. Again: This is the reason for Linux having the ACCESS_ONCE macro. Even if you have a hardware memory barrier you still need to tell the compiler not to optimize using volatile or some non-standard compiler intrinsic. 

I haven't yet worked with an OS where the critical sections does any memory stuff (and I've worked with quite a few). Depending on their complexity they often just turn off interrupts.
> 
> Note that volatile does offer little to no protection regards to the memory accessed as it doesn't issue memory barrier/synchronization instructions. It forces the compiler to add read or write accesses but these read or write access can very well be optimized by the hardware itself.

Of course it can. But it must not have any effect on the code execution (With exceptions of e.g. certain DMA implementations). That's the reason for rather complex [Cache coherence protocols](https://en.wikipedia.org/wiki/Cache_coherence).

> 
> volatile is only useful to ensure memory mapped I/O; 

No. No and No! If you ever wrote a multi-threaded real-time OS from scratch you'll know what I'm talking about. :) 

> it is certainly not a synchronization primitive. 

Here I can agree. It is a compiler optimization suppression directive. And one must still ensure that the variables are accessed atomically. This is one of the reasons why the critical section is needed here (besides the IMO suboptimal implementation of circular buffer using a _full variable). With a single writer/reader (which is often the case using single-threaded systems) and atomic access to the indexes, one does not even need a critical section.

>For synchronization; use a mutex, a critical section or even an atomic variable construct.

See above.

> In the end, I think we can safely drop the volatile qualifier out of the _head, _tail and _full members; granted that critical section ensure the memory consistency.

This assumptions is (fortunately) wrong. These were the first reasonable hits I found, but I'm sure there are hundreds of other hits out there, where people find out the hard way why one needs volatile in a ring buffer: https://stackoverflow.com/questions/32526202/strange-circular-buffer-behaviour and https://hackaday.com/2015/10/29/embed-with-elliot-going-round-with-circular-buffers/

cmorty, 2018-08-07:
@bmcdonnell-ionx Yes. that would work, too. ;) The binary code will be the same.

0xc0170, 2018-08-07:
> There are two ways to fix this

We could optimize there as suggested. Please send PR with the proposal as above.

> Additionally _pool should actually be volatile, as it can change any time. While I currently cannot come up with a situation where the compiler could optimize accessing _pool, this does not mean there is none. Especially with improved compiler support for LTO.

Not certain about this one.

pan-, 2018-08-07:
@cmorty 
So I've checked what disabling the interrupt actually means on Cortex M. [Here](https://github.com/ARMmbed/mbed-os/blob/1e676f6eda27c84f61fa97626b1980b9e410b769/cmsis/TARGET_CORTEX_M/cmsis_gcc.h#L129-L140) is the code from cmsis used to disable and enable irqs: 

```c
/**
  \brief   Enable IRQ Interrupts
  \details Enables IRQ interrupts by clearing the I-bit in the CPSR.
           Can only be executed in Privileged modes.
 */
__STATIC_FORCEINLINE void __enable_irq(void)
{
  __ASM volatile ("cpsie i" : : : "memory");
}

/**
  \brief   Disable IRQ Interrupts
  \details Disables IRQ interrupts by setting the I-bit in the CPSR.
           Can only be executed in Privileged modes.
 */
__STATIC_FORCEINLINE void __disable_irq(void)
{
  __ASM volatile ("cpsid i" : : : "memory");
}
```

The `volatile` part forces the compiler to keep the statement while the `memory` part do the following: 

> The "memory" clobber tells the compiler that the assembly code performs memory reads or writes to items other than those listed in the input and output operands (for example, accessing the memory pointed to by one of the input parameters). **To ensure memory contains correct values, GCC may need to flush specific register values to memory before executing the asm. Further, the compiler does not assume that any values read from memory before an asm remain unchanged after that asm; it reloads them as needed. Using the "memory" clobber effectively forms a read/write memory barrier for the compiler.** 

Last but not least, according to the [Cortex-M programming guide](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dai0321a/BIHDFBJF.html) there is no requirement to insert a memory barrier after a `cpsid` or `cpsie` instruction. In the end it is perfectly fine for the compiler to optimize accesses to the buffer data structure within the critical section as the critical section ensure that variables are _synchronized_ at the entry and exit of the critical section. 

So yes, our critical section is sufficient for this use case as it does what it should do. There is no requirement to add volatile statements.

-----

> I haven't yet worked with an OS where the critical sections does any memory stuff. 

I'm sure you have, you just don't know it ;). 

It mostly depends on the expectations of the operating system and the target architecture. As an example, when linux disable irqs on the DEC Alpha (it has a very weak memory model), it also issue a memory barrier (see [here](https://elixir.bootlin.com/linux/v4.18-rc8/source/arch/alpha/include/asm/irqflags.h#L32)). Similarly the functions `taskENTER_CRITICAL` and `taskENTER_CRITICAL_FROM_ISR` of free RTOS on cortex M issue data and instruction barriers ...

Mbed OS do not define a complete and fine grained memory model like linux do (see [here](https://www.kernel.org/doc/Documentation/memory-barriers.txt)). Cherry picking primitives from other systems is not an option either as often primitives needs to work together to achieve the desired use cases (`ACCESS_ONCE` is not a replacement for barriers; it is another tool in the box).

If we move to more recent C and C++ standard (11) then it will become possible to use a more efficient memory model defined by the language. Indeed the C and C++ standard have added atomic variables in their latter standard to workaround the limitations of volatile (emphasis mine): 

> The volatile type qualifier has a long history within C and C++, and changing its meaning is both risky and unnecessary. **In addition, the existing meaning of volatile, "may be modified by external agents", is somewhat orthogonal to "may be modified concurrently by the program**"

You can read the initial proposal and its rationale [here](http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2007/n2427.html)

---- 

Side note: 
* https://stackoverflow.com/questions/32526202/strange-circular-buffer-behaviour : that is **not** a valid implementation of a SPSC circular buffer. I encourage you to look at [this](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&ved=2ahUKEwjz--yH_NrcAhVF2KQKHb-SDP4QFjAAegQIABAC&url=https%3A%2F%2Fpdfs.semanticscholar.org%2F8c15%2Fb938e6fdbcbab6e1b5704857043e5c680734.pdf&usg=AOvVaw1AJorG8a9p4HM4zMybPy0C) paper and the references it contains. AFAIK there is not lock free data structure without atomic read modify write.
* https://hackaday.com/2015/10/29/embed-with-elliot-going-round-with-circular-buffers/ : Same as above.

cmorty, 2018-08-08:
@0xc0170 
>>    Additionally _pool should actually be volatile, as it can change any time. While I currently cannot come up with a situation where the compiler could optimize accessing _pool, this does not mean there is none. Especially with improved compiler support for LTO.
> Not certain about this one.

After a night of sleep: Even if the compiler is not able to optimize _pool, because it's pointer is volatile (due to `_head` and `_tail` being volatile), the value itself is volatile, too, as it is can be changed from a different context.

@pan- 
> IRQ for Arm

So what? I said disabling interrupts is one way of doing this. (Another option is to increase the priority.) Using inline assembler is "some non-standard compiler intrinsic.", as assembler is not part of the C-standard. Besides, you are mixing up correlation with causality. The memory parameter is needed bacause `cpsie i` changes memory mapped registers and the compiler must be made aware of this. I've also seen similar stuff because people are too stupid to use volatile correctly. But that's besides the point: The documentation says nothing about the memory. Thus one must not rely on the specific implementation (in this case disabling IRQ+using ARM-GCC). 

> taskENTER_CRITICAL having a memory barrier

Again: So what? It says nothing about a memory barrier in the documentation. Thus even if it has one - for whatever reason -, anyone is free to implement it without. Yet such an implementation would probably break a lot of stuff that "magically" works due to this, but where people did not put the brains into the code to see that their variables actually need a volatile.

> Atomic discussion

This is besides the point, again. Yes, Atomic is something totally different than volatile. And here we need volatile(!). In this case we really need the volatile/ACCESS_ONCE semantic and not the atomic, as atomic is covered by the critical section. 

> Links

The were to reference issues with volatile in ring buffers, not SPSC ring buffers.

pan-, 2018-08-08:
@cmorty Critical sections, like mutexes are constructs that prevent concurrent access to a shared resource. If there is pending loads and stores of an exited critical section at the entry of a critical section entered then the critical section implementation doesn't do it's job. Similarly if values fetched before the entry are used inside the critical section it doesn't do its job either. 

Stopping concurrent execution is one aspect of critical sections but remember critical sections and mutexes are mostly used to avoid **data race**. If memory is in an inconsistent state, how would you avoid this ? Expect the programmer to wrongfully use a keyword, `volatile`, that do not cover the "may be modified concurrently by the program" use case ? 

> The were to reference issues with volatile in ring buffers, not SPSC ring buffers.

Besides the aggressive tone and lack of argumentation; that's some really bad faith here. The first sentence of the first link you provide is: `I implemented a single-producer-single-consumer lock-free Circular Buffer`. Here is the code of the second link: 

```C++
volatile struct Buffer rx_buffer = {{}, 0, 0};
 
ISR(USART_RX_vect){
    bufferWrite(&rx_buffer, UDR0);
}
 
int main(void){
    uint8_t received_byte;  
    enum BufferStatus status;
 
    initUSART();
    initUSART_interrupts();
 
    while (1) {
        status = bufferRead(&rx_buffer, &received_byte);
        if (status == BUFFER_OK) {      /* we have data */
            do_something(received_byte);
        } 
    }
}
```

Yeah that is definitely not a writer in the handler mode and a reader in thread mode.

---- 

I encourage you to read [Volatile Considered Harmful](https://github.com/torvalds/linux/blob/master/Documentation/process/volatile-considered-harmful.rst) in the linux kernel doc.

kjbracey-arm, 2018-08-08:
I'm with @pan- on this - the semantics of enter/exit-critical must be assumed to contain all appropriate barriers such that `volatile` is not necessary, just as `volatile` is not necessary on things accessed inside mutexes.

If enter/exit critical does not correctly protect memory accesses within them, they would be broken, just as a mutex lock/unlock would be.

kjbracey-arm, 2018-08-08:
Okay, pan- made exactly the same point at the same time :)

PR #7727 (2018-08-09): CircularBuffer: volatile specifier removal
Volatile specifier in this case it not required as we currently have all accesses
to the buffer protected by critical section. This shall optimize accesses in
some cases to the buffer.

Fixes #7702

### Description

### Pull request type

    [X] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Feature
    [ ] Breaking change
commit 7a44185c (2018-08-09): CircularBuffer: volatile specifier removal

Volatile specifier in this case it not required as we currently have all accesses
to the buffer protected by critical section. This shall optimize accesses in
some cases to the buffer.

Fixes #7702
closed 2018-08-09 by cmonr