ARMmbed/mbed-os #7707: NRF52 PwmOut inverted

ChiSaw, 2018-08-06:
### Description
Switched from mbed-os-5.7 to mbed-os-5.9, suddenly PwmOut behaved inverted, meaning

`PwmOut led(SOME_PIN);`
`led.write(0); // LED on`

accordingly opposite cases for `.write(1000);`

Note: LED circuit is active-high.

Target: NRF52832
ARM_GCC 6-2017-q2-update 6.3.1 20170620 (release)
mbed-os: 5.9, mbed-cli
Steps to reproduce:
empty mbed-os project with main() containing

```
PwmOut led(p17);
led.write(0);
Thread::wait(3000); // to notice a change
led.write(1000);
```

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

MateuszMaz, 2018-08-07:
I'll check it.

ChiSaw, 2018-08-08:
No, I am using my own board where the NRF52832 serves as the main processor and there the LED is connected active high. But I see where this inverted behavior comes from, someone implemented PwmOut based on the DK and applied the active-low logic from this specific hardware? 
In my opinion, this should be changed to non-inverted, what do you think? One reason might be to use PWM not for LED, but for motor control, etc, where you can not always expect active-low logic.

jrobeson, 2018-08-08:
shouldn't this sort of thing be specified at the mbed-os platform level and thus be standard across all devices? If so, then the default should be whatever they do.

MateuszMaz, 2018-08-08:
cc @0xc0170 
Martin, the PWM waveform polarity generated by nRF-52 is inconsitient to other devices like K64F, Nucleo-F070RB, what do you think, should this behaviour be changed?

fkjagodzinski, 2018-08-08:
CC @ARMmbed/mbed-os-hal

0xc0170, 2018-08-08:
> the PWM waveform polarity generated by nRF-52 is inconsitient to other devices like K64F, Nucleo-F070RB, what do you think, should this behaviour be changed?

As this might not be yet specified (will be!), should follow what the rest of the boards do. They do write 0 in their init:

```
pwmout_write    (obj, 0);
```

I can't locate this for NRF52, I can see obj->percent is set to 0 but not used anywhere during `init` ?

MateuszMaz, 2018-08-08:
@0xc0170 ok

MateuszMaz, 2018-08-08:
@0xc0170  It looks like the problem is located in PR #6547 that is about updating Nordic NRF52 based targets to SDK 14.2

drahnr, 2018-08-08:
https://github.com/ARMmbed/mbed-os/pull/6547/commits/d11f74cb4c8b83dd72a5f16d77e2fc8ef0a1f1d8

MateuszMaz, 2018-08-09:
I found another problem that is connected to Pwm in target NRF-52 , I described it here  #7743

jrobeson, 2018-08-12:
I covered this in https://github.com/ARMmbed/mbed-os/issues/5683 I think. It was said to be fixed, but it doesn't look to be the case.

MateuszMaz, 2018-08-13:
here is pr #7779

bmcdonnell-ionx, 2018-09-14:
[Mbed OS 5.9.7 release notes](https://os.mbed.com/blog/entry/mbed-os-597-released/) links to this issue as "fixed". Consider closing.

closed 2018-09-21 by ChiSaw