ARMmbed/mbed-os #7709: Failing mbedtls selftests for a specific tool-chain

aashishc1988, 2018-08-06:
### Description
- Type: Bug
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
WIZWIKI_W7500

**Toolchain:**
IAR

**mbed-os sha:**
Failing since mbed-os-5.5.0

**Tests we are failing:**
tests-mbedtls-selftest                                                       | mbedtls_sha256_self_test

**Steps to reproduce**
mbed  test -m WIZWIKI_W7500 -t IAR -n tests-mbedtls-selftest -v

**log snippet**
```
> 
00:03:26.797 mbedgt: mbed-host-test-runner: started
00:03:27.607 [1533313335.62][root]Generating grammar tables from /usr/lib/python2.7/lib2to3/Grammar.txt
00:03:27.635 [1533313335.65][root]Generating grammar tables from /usr/lib/python2.7/lib2to3/PatternGrammar.txt
00:03:27.647 [1533313335.66][HTST][INF] host test executor ver. 1.4.0
00:03:27.647 [1533313335.66][HTST][INF] copy image onto target... SKIPPED!
00:03:27.650 [1533313335.66][HTST][INF] starting host test process...
00:03:27.665 [1533313335.67][CONN][INF] starting connection process...
00:03:27.665 [1533313335.68][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
00:03:27.666 [1533313335.68][CONN][INF] initializing global resource mgr listener...
00:03:27.685 [1533313335.70][HTST][INF] setting timeout to: 60 sec
00:03:27.728 [1533313335.74][GLRM][INF] remote resources initialization: remote(host=goku.austin.arm.com, port=8000)
00:03:27.729 [1533313335.74][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:03:27.731 [1533313335.74][urllib3.connectionpool]http://goku.austin.arm.com:8000 "POST /login HTTP/1.1" 200 213
00:03:27.734 [1533313335.74][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:03:27.735 [1533313335.75][urllib3.connectionpool]http://goku.austin.arm.com:8000 "GET /socket.io/?token=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJ1c2VybmFtZSI6InVzZXIiLCJncm91cCI6InVzZXIiLCJpYXQiOjE1MzMzMTMzMzUsImV4cCI6MTUzMzkxODEzNX0.N5xx6kWmF_sMpiBT78RTfDSVJfhOQvb6dKWG2YVRGKw&EIO=3&transport=polling&t=1533313335742-0 HTTP/1.1" 200 101
00:03:27.746 [1533313335.76][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:03:27.749 [1533313335.76][urllib3.connectionpool]http://goku.austin.arm.com:8000 "GET /this/resource HTTP/1.1" 200 3651
00:03:27.750 [1533313335.76][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:03:27.756 [1533313335.77][urllib3.connectionpool]http://goku.austin.arm.com:8000 "GET /resource HTTP/1.1" 200 55402
00:03:27.760 [1533313335.77][GLRM][INF] remote resources count: 39
00:03:27.777 [1533313335.79][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:03:27.779 [1533313335.79][urllib3.connectionpool]http://goku.austin.arm.com:8000 "GET /resource/2201000003b88c5b00000000000000000000000097969902 HTTP/1.1" 200 1834
00:03:27.780 [1533313335.79][GLRM][INF] remote resources flashing with 'BUILD/tests/WIZWIKI_W7500/IAR/TESTS/mbedtls/selftest/selftest.bin'...
00:03:27.781 [1533313335.79][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:03:27.786 [1533313335.80][urllib3.connectionpool]http://goku.austin.arm.com:8000 "POST /file HTTP/1.1" 200 244
00:03:41.668 [1533313349.68][GLRM][INF] opening connection to platform at baudrate='9600'
00:03:41.669 [1533313349.68][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:03:41.820 [1533313349.83][urllib3.connectionpool]http://goku.austin.arm.com:8000 "PUT /resource/2201000003b88c5b00000000000000000000000097969902/connect HTTP/1.1" 200 15
00:03:41.820 [1533313349.83][GLRM][INF] remote resources reset...
00:03:44.108 [1533313352.12][GLRM][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
00:03:44.108 [1533313352.12][CONN][INF] sending up to 10 __sync packets (specified with --sync=10)
00:03:44.108 [1533313352.12][CONN][INF] sending preamble '7c69ac78-ee69-48d0-b987-ff4d75a2ad20'
00:03:44.109 [1533313352.12][GLRM][TXD] {{__sync;7c69ac78-ee69-48d0-b987-ff4d75a2ad20}}
00:03:45.267 [1533313353.28][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
00:03:45.267 [1533313353.28][CONN][INF] found SYNC in stream: {{__sync;7c69ac78-ee69-48d0-b987-ff4d75a2ad20}} it is #0 sent, queued...
00:03:45.267 [1533313353.28][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
00:03:45.268 [1533313353.28][HTST][INF] sync KV found, uuid=7c69ac78-ee69-48d0-b987-ff4d75a2ad20, timestamp=1533313353.278023
00:03:45.268 [1533313353.28][HTST][INF] DUT greentea-client version: 1.3.0
00:03:45.367 [1533313353.38][CONN][RXD] >>> Running 2 test cases...
00:03:45.367 [1533313353.38][HTST][INF] setting timeout to: 120 sec
00:03:45.368 [1533313353.38][CONN][INF] found KV pair in stream: {{__timeout;120}}, queued...
00:03:45.368 [1533313353.38][HTST][INF] host test class: '<class 'mbed_host_tests.host_tests.default_auto.DefaultAuto'>'
00:03:45.368 [1533313353.38][HTST][INF] host test setup() call...
00:03:45.368 [1533313353.38][HTST][INF] CALLBACKs updated
00:03:45.368 [1533313353.38][HTST][INF] host test detected: default_auto
[... 25 lines cut ...]
00:03:46.179 [1533313354.19][CONN][RXD] {{__cpu_info     sleep time;0}}
00:03:46.179 [1533313354.19][CONN][RXD] {{__cpu_info deepsleep time;0}}
00:03:46.179 [1533313354.19][CONN][RXD] {{__cpu_info  %  sleep/deep;0;0}}
00:03:46.271 [1533313354.28][CONN][INF] found KV pair in stream: {{end;failure}}, queued...
00:03:46.271 [1533313354.28][CONN][INF] found KV pair in stream: {{__exit;0}}, queued...
00:03:46.281 [1533313354.29][HTST][INF] __exit(0)
00:03:46.282 [1533313354.29][HTST][INF] __notify_complete(False)
00:03:46.282 [1533313354.29][HTST][INF] __exit_event_queue received
00:03:46.282 [1533313354.29][HTST][INF] test suite run finished after 0.91 sec...
00:03:46.292 [1533313354.30][CONN][INF] received special event '__host_test_finished' value='True', finishing
00:03:46.295 [1533313354.31][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:03:46.438 [1533313354.45][urllib3.connectionpool]http://goku.austin.arm.com:8000 "PUT /resource/2201000003b88c5b00000000000000000000000097969902/disconnect HTTP/1.1" 200 2
00:03:46.438 [1533313354.45][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:03:49.666 [1533313357.68][urllib3.connectionpool]http://goku.austin.arm.com:8000 "PUT /resource/2201000003b88c5b00000000000000000000000097969902/release HTTP/1.1" 200 66
00:03:49.669 [1533313357.68][HTST][INF] CONN exited with code: 0
00:03:49.669 [1533313357.68][HTST][INF] Some events in queue
00:03:49.669 [1533313357.68][HTST][INF] stopped consuming events
00:03:49.669 [1533313357.68][HTST][INF] host test result() call skipped, received: False
00:03:49.669 [1533313357.68][HTST][INF] calling blocking teardown()
00:03:49.669 [1533313357.68][HTST][INF] teardown() finished
00:03:49.669 [1533313357.68][HTST][INF] {{result;failure}}
```
    [ ] Question
    [ ] Enhancement 
    [X ] Bug

studavekar, 2018-08-06:
cc : @andrewc-arm

andrewc-arm, 2018-08-10:
Hi, @khj098765 
Could you please review this problem? We need to pass IAR as well. I will discuss with Mr. Hur as well and notify you separately. Thank.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-6067*

closed 2021-05-12 by ciarmcom

0xc0170, 2021-05-17:
Closed as not relevant anymore (unsupported toolchain, >2 years)