ARMmbed/mbed-os #7710: Failing flashsim tests due to lack of memory

aashishc1988, 2018-08-06:
### Description
- Type: Bug
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
WIZWIKI_W7500

**Toolchain:**
IAR

**Tests we are failing:**
features-tests-filesystem-flashsim_block_device

**Steps to reproduce**
mbed  test -m WIZWIKI_W7500 -t IAR -n features-tests-filesystem-flashsim_block_device -v

**log snippet**
```
> 
00:07:17.808 mbedgt: mbed-host-test-runner: started
00:07:18.236 [1533313566.25][root]Generating grammar tables from /usr/lib/python2.7/lib2to3/Grammar.txt
00:07:18.264 [1533313566.28][root]Generating grammar tables from /usr/lib/python2.7/lib2to3/PatternGrammar.txt
00:07:18.274 [1533313566.29][HTST][INF] host test executor ver. 1.4.0
00:07:18.274 [1533313566.29][HTST][INF] copy image onto target... SKIPPED!
00:07:18.275 [1533313566.29][HTST][INF] starting host test process...
00:07:18.297 [1533313566.31][CONN][INF] starting connection process...
00:07:18.309 [1533313566.32][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
00:07:18.309 [1533313566.32][CONN][INF] initializing global resource mgr listener...
00:07:18.363 [1533313566.38][HTST][INF] setting timeout to: 60 sec
00:07:18.383 [1533313566.40][GLRM][INF] remote resources initialization: remote(host=goku.austin.arm.com, port=8000)
00:07:18.385 [1533313566.40][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:07:18.387 [1533313566.40][urllib3.connectionpool]http://goku.austin.arm.com:8000 "POST /login HTTP/1.1" 200 213
00:07:18.390 [1533313566.41][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:07:18.392 [1533313566.41][urllib3.connectionpool]http://goku.austin.arm.com:8000 "GET /socket.io/?token=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJ1c2VybmFtZSI6InVzZXIiLCJncm91cCI6InVzZXIiLCJpYXQiOjE1MzMzMTM1NjYsImV4cCI6MTUzMzkxODM2Nn0.4LKgWz7n9c4oKpYhJmY1ODx-preMjcZzo1BDuVjsni8&EIO=3&transport=polling&t=1533313566405-0 HTTP/1.1" 200 101
00:07:18.423 [1533313566.44][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:07:18.427 [1533313566.44][urllib3.connectionpool]http://goku.austin.arm.com:8000 "GET /this/resource HTTP/1.1" 200 3651
00:07:18.429 [1533313566.45][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:07:18.446 [1533313566.46][urllib3.connectionpool]http://goku.austin.arm.com:8000 "GET /resource HTTP/1.1" 200 54420
00:07:18.446 [1533313566.46][GLRM][INF] remote resources count: 39
00:07:18.462 [1533313566.48][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:07:18.464 [1533313566.48][urllib3.connectionpool]http://goku.austin.arm.com:8000 "GET /resource/2201000003b88c5b00000000000000000000000097969902 HTTP/1.1" 200 1846
00:07:18.465 [1533313566.48][GLRM][INF] remote resources flashing with 'BUILD/tests/WIZWIKI_W7500/IAR/features/TESTS/filesystem/flashsim_block_device/flashsim_block_device.bin'...
00:07:18.465 [1533313566.48][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:07:18.477 [1533313566.49][urllib3.connectionpool]http://goku.austin.arm.com:8000 "POST /file HTTP/1.1" 200 257
00:07:31.619 [1533313579.63][GLRM][INF] opening connection to platform at baudrate='9600'
00:07:31.619 [1533313579.64][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:07:31.757 [1533313579.77][urllib3.connectionpool]http://goku.austin.arm.com:8000 "PUT /resource/2201000003b88c5b00000000000000000000000097969902/connect HTTP/1.1" 200 15
00:07:31.757 [1533313579.77][GLRM][INF] remote resources reset...
00:07:34.000 [1533313582.02][GLRM][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
00:07:34.000 [1533313582.02][CONN][INF] sending up to 10 __sync packets (specified with --sync=10)
00:07:34.000 [1533313582.02][CONN][INF] sending preamble '872bf636-ace0-4175-96b2-5c45bda15471'
00:07:34.004 [1533313582.02][GLRM][TXD] {{__sync;872bf636-ace0-4175-96b2-5c45bda15471}}
00:07:35.173 [1533313583.19][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
00:07:35.173 [1533313583.19][CONN][INF] found SYNC in stream: {{__sync;872bf636-ace0-4175-96b2-5c45bda15471}} it is #0 sent, queued...
00:07:35.174 [1533313583.19][HTST][INF] sync KV found, uuid=872bf636-ace0-4175-96b2-5c45bda15471, timestamp=1533313583.190327
00:07:35.174 [1533313583.19][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
00:07:35.174 [1533313583.19][HTST][INF] DUT greentea-client version: 1.3.0
00:07:35.273 [1533313583.29][CONN][RXD] >>> Running 1 test cases...
00:07:35.274 [1533313583.29][CONN][INF] found KV pair in stream: {{__timeout;30}}, queued...
00:07:35.274 [1533313583.29][HTST][INF] setting timeout to: 30 sec
00:07:35.274 [1533313583.29][CONN][INF] found KV pair in stream: {{__host_test_name;default_auto}}, queued...
00:07:35.274 [1533313583.29][HTST][INF] host test class: '<class 'mbed_host_tests.host_tests.default_auto.DefaultAuto'>'
00:07:35.274 [1533313583.29][HTST][INF] host test setup() call...
00:07:35.274 [1533313583.29][HTST][INF] CALLBACKs updated
00:07:35.274 [1533313583.29][HTST][INF] host test detected: default_auto
00:07:35.374 [1533313583.39][CONN][RXD]
00:07:35.374 [1533313583.39][CONN][INF] found KV pair in stream: {{__testcase_name;FlashSimBlockDevice functionality test}}, queued...
[... 21 lines cut ...]
00:07:36.080 [1533313584.10][CONN][RXD] {{__cpu_info     sleep time;0}}
00:07:36.081 [1533313584.10][CONN][RXD] {{__cpu_info deepsleep time;0}}
00:07:36.178 [1533313584.19][CONN][RXD] {{__cpu_info  %  sleep/deep;0;0}}
00:07:36.178 [1533313584.19][CONN][INF] found KV pair in stream: {{end;failure}}, queued...
00:07:36.178 [1533313584.20][CONN][INF] found KV pair in stream: {{__exit;0}}, queued...
00:07:36.188 [1533313584.20][HTST][INF] __exit(0)
00:07:36.188 [1533313584.20][HTST][INF] __notify_complete(False)
00:07:36.188 [1533313584.21][HTST][INF] __exit_event_queue received
00:07:36.188 [1533313584.21][HTST][INF] test suite run finished after 0.91 sec...
00:07:36.200 [1533313584.22][CONN][INF] received special event '__host_test_finished' value='True', finishing
00:07:36.201 [1533313584.22][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:07:36.259 [1533313584.28][urllib3.connectionpool]http://goku.austin.arm.com:8000 "PUT /resource/2201000003b88c5b00000000000000000000000097969902/disconnect HTTP/1.1" 200 2
00:07:36.261 [1533313584.28][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:07:39.417 [1533313587.43][urllib3.connectionpool]http://goku.austin.arm.com:8000 "PUT /resource/2201000003b88c5b00000000000000000000000097969902/release HTTP/1.1" 200 66
00:07:39.421 [1533313587.44][HTST][INF] CONN exited with code: 0
00:07:39.421 [1533313587.44][HTST][INF] Some events in queue
00:07:39.421 [1533313587.44][HTST][INF] stopped consuming events
00:07:39.421 [1533313587.44][HTST][INF] host test result() call skipped, received: False
00:07:39.421 [1533313587.44][HTST][INF] calling blocking teardown()
00:07:39.421 [1533313587.44][HTST][INF] teardown() finished
00:07:39.421 [1533313587.44][HTST][INF] {{result;failure}}
```
    [ ] Question
    [ ] Enhancement 
    [X ] Bug

aashishc1988, 2018-08-06:
Filed the PR for it

https://github.com/ARMmbed/mbed-os/pull/7696

andrewc-arm, 2018-08-10:
Hi, @khj098765 
Could you please also follow this issue as well? Thanks.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-6068*

closed 2021-05-12 by ciarmcom