ARMmbed/mbed-os #7712: Failing filesystem tests due to memory allocation

aashishc1988, 2018-08-06:
### Description
- Type: Bug
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
WIZWIKI_W7500

**Toolchain:**
GCC_ARM

**Tests we are failing:**
features-tests-filesystem-buffered_block_device

**Steps to reproduce**

mbed  test -m WIZWIKI_W7500 -t IAR -n features-tests-filesystem-buffered_block_device -DMBED_HEAP_STATS_ENABLED=1 -DMBED_TRAP_ERRORS_ENABLED=1 -v

So, it seems that the call to new would fail, and even though we have nothrow exception defined, the mbed-os error handler is not very forgiving. The failure gets trapped by the error handler.

**log snippet**
```
> 
00:16:21.879 [1533362738.44][root]Generating grammar tables from /usr/lib/python2.7/lib2to3/Grammar.txt
00:16:21.932 [1533362738.49][root]Generating grammar tables from /usr/lib/python2.7/lib2to3/PatternGrammar.txt
00:16:21.972 [1533362738.53][HTST][INF] host test executor ver. 1.4.0
00:16:21.972 [1533362738.53][HTST][INF] copy image onto target... SKIPPED!
00:16:21.978 [1533362738.54][HTST][INF] starting host test process...
00:16:21.985 [1533362738.55][CONN][INF] starting connection process...
00:16:21.995 [1533362738.55][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
00:16:21.995 [1533362738.55][CONN][INF] initializing global resource mgr listener...
00:16:22.059 [1533362738.62][HTST][INF] setting timeout to: 60 sec
00:16:22.129 [1533362738.69][GLRM][INF] remote resources initialization: remote(host=goku.austin.arm.com, port=8000)
00:16:22.130 [1533362738.69][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:16:22.139 [1533362738.70][urllib3.connectionpool]http://goku.austin.arm.com:8000 "POST /login HTTP/1.1" 200 213
00:16:22.142 [1533362738.70][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:16:22.147 [1533362738.71][urllib3.connectionpool]http://goku.austin.arm.com:8000 "GET /socket.io/?token=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJ1c2VybmFtZSI6InVzZXIiLCJncm91cCI6InVzZXIiLCJpYXQiOjE1MzMzNjI3MzgsImV4cCI6MTUzMzk2NzUzOH0.tWkn4qzmacsiwfDq1g5P6ggBK0ceGRmmf-OdK4Y4VxM&EIO=3&transport=polling&t=1533362738702-0 HTTP/1.1" 200 101
00:16:22.188 [1533362738.75][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:16:22.196 [1533362738.76][urllib3.connectionpool]http://goku.austin.arm.com:8000 "GET /this/resource HTTP/1.1" 200 3651
00:16:22.196 [1533362738.76][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:16:22.210 [1533362738.77][urllib3.connectionpool]http://goku.austin.arm.com:8000 "GET /resource HTTP/1.1" 200 54065
00:16:22.210 [1533362738.77][GLRM][INF] remote resources count: 39
00:16:22.227 [1533362738.79][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:16:22.231 [1533362738.79][urllib3.connectionpool]http://goku.austin.arm.com:8000 "GET /resource/0200000013894d440085101affff0003ffffffff97969900 HTTP/1.1" 200 1815
00:16:22.231 [1533362738.79][GLRM][INF] remote resources flashing with 'BUILD/tests/KL25Z/GCC_ARM/TESTS/mbedmicro-rtos-mbed/threads/threads.bin'...
00:16:22.231 [1533362738.79][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:16:22.243 [1533362738.80][urllib3.connectionpool]http://goku.austin.arm.com:8000 "POST /file HTTP/1.1" 200 243
00:16:36.697 [1533362753.26][GLRM][INF] opening connection to platform at baudrate='9600'
00:16:36.697 [1533362753.26][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:16:36.830 [1533362753.39][urllib3.connectionpool]http://goku.austin.arm.com:8000 "PUT /resource/0200000013894d440085101affff0003ffffffff97969900/connect HTTP/1.1" 200 15
00:16:36.830 [1533362753.39][GLRM][INF] remote resources reset...
00:16:39.004 [1533362755.56][GLRM][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
00:16:39.004 [1533362755.56][CONN][INF] sending up to 10 __sync packets (specified with --sync=10)
00:16:39.004 [1533362755.56][CONN][INF] sending preamble '0e80ccb1-2532-45c9-8190-3e61a97a84cf'
00:16:39.007 [1533362755.57][GLRM][TXD] {{__sync;0e80ccb1-2532-45c9-8190-3e61a97a84cf}}
00:16:40.187 [1533362756.75][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
00:16:40.286 [1533362756.85][CONN][INF] found SYNC in stream: {{__sync;0e80ccb1-2532-45c9-8190-3e61a97a84cf}} it is #0 sent, queued...
00:16:40.286 [1533362756.85][HTST][INF] sync KV found, uuid=0e80ccb1-2532-45c9-8190-3e61a97a84cf, timestamp=1533362756.847568
00:16:40.286 [1533362756.85][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
00:16:40.287 [1533362756.85][HTST][INF] DUT greentea-client version: 1.3.0
00:16:40.287 [1533362756.85][CONN][INF] found KV pair in stream: {{__timeout;20}}, queued...
00:16:40.287 [1533362756.85][HTST][INF] setting timeout to: 20 sec
00:16:40.386 [1533362756.95][CONN][RXD] >>> Running 34 test cases...
00:16:40.386 [1533362756.95][CONN][INF] found KV pair in stream: {{__host_test_name;default_auto}}, queued...
00:16:40.386 [1533362756.95][CONN][INF] found KV pair in stream: {{__testcase_name;Testing single thread}}, queued...
00:16:40.387 [1533362756.95][HTST][INF] host test class: '<class 'mbed_host_tests.host_tests.default_auto.DefaultAuto'>'
00:16:40.387 [1533362756.95][HTST][INF] host test setup() call...
00:16:40.387 [1533362756.95][HTST][INF] CALLBACKs updated
[... 150 lines cut ...]
00:16:47.251 [1533362763.81][CONN][RXD]
00:16:47.331 [1533362763.89][CONN][RXD] Location: 0x4CAB
00:16:47.331 [1533362763.89][CONN][RXD] Error Value: 0x180
00:16:47.425 [1533362763.99][CONN][RXD] Current Thread: Id: 0x20001804 Entry: 0x79E9 StackSize: 0x180 StackMem: 0x20001AD8 SP: 0x20001B88
00:16:47.425 [1533362763.99][CONN][RXD] -- MbedOS Error Info --
00:17:00.454 [1533362777.02][HTST][INF] test suite run finished after 20.17 sec...
00:17:00.454 [1533362777.02][CONN][INF] received special event '__host_test_finished' value='True', finishing
00:17:00.455 [1533362777.02][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:17:00.504 [1533362777.07][urllib3.connectionpool]http://goku.austin.arm.com:8000 "PUT /resource/0200000013894d440085101affff0003ffffffff97969900/disconnect HTTP/1.1" 200 2
00:17:00.505 [1533362777.07][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:17:03.657 [1533362780.22][urllib3.connectionpool]http://goku.austin.arm.com:8000 "PUT /resource/0200000013894d440085101affff0003ffffffff97969900/release HTTP/1.1" 200 66
00:17:03.659 [1533362780.22][HTST][INF] CONN exited with code: 0
00:17:03.659 [1533362780.22][HTST][INF] No events in queue
00:17:03.659 [1533362780.22][HTST][INF] stopped consuming events
00:17:03.659 [1533362780.22][HTST][INF] host test result(): None
00:17:03.659 [1533362780.22][HTST][WRN] missing __exit event from DUT
00:17:03.660 [1533362780.22][HTST][WRN] missing __exit_event_queue event from host test
00:17:03.660 [1533362780.22][HTST][ERR] missing __exit_event_queue event from host test and no result from host test, timeout...
00:17:03.660 [1533362780.22][HTST][INF] calling blocking teardown()
00:17:03.660 [1533362780.22][HTST][INF] teardown() finished
00:17:03.660 [1533362780.22][HTST][INF] {{result;timeout}}
```
    [ ] Question
    [ ] Enhancement 
    [X ] Bug

aashishc1988, 2018-08-06:
One way to handle this, as suggested by @SenRamakri , is that we over-write the mbed-os error handler with a dummy procedure that does nothing when the call to "new" fails. Here is the snippet of what code would look like (with some commentary about the changes)

`
/**
  * If the flag MBED_TRAP_ERRORS_ENABLED is defined, and the part cannot allocate enough memory for the test, the call to memory allocator will fail and mbed-os error function will be called, and we will be stuck in a loop, which ideally is good for debugging purposes, but since we are aware of the reasons for failure, and we don't want the CI to get stuck due to that, we will overwrite "error" function call with a dummy procedure that returns SUCCESS. Please be wary while using this, as you do not want to use this for cases that can do without it because it will mask some reason failures due to memory allocation, therefore, if a test has multiple cases and not all of them needs this overwrite, consider splitting the test in two files.
*/

#if defined(MBED_TRAP_ERRORS_ENABLED) && MBED_TRAP_ERRORS_ENABLED
void error(const char* format, ...) {
    (void) format;
}

//Override the set_error function to trap the errors 
mbed_error_status_t mbed_error(mbed_error_status_t error_status, const char *error_msg, unsigned int error_value, const char *filename, int line_number) 
{
    return MBED_SUCCESS;
}
#endif
`
However, this change is only needed for some specific parts, and tool chain. Like the test, features-tests-filesystem-buffered_block_device is failing for WIZWIKI_W7500 on GCC_ARM toolchain. Adding the code snippet above can/will mask failures for other devices that do not need this change in. So, making this change would require having some more ifdef in the test script, and to me it looks like we are entering into a new territory of modifying the test according to specific part numbers which doesn't look as appealing.

0xc0170, 2018-08-07:
Why is it failing? not enough memory? There's 48 kB SRAM (if no socket are used, not sure if these are allocated by default)

davidsaada, 2018-08-07:
@aashishc1988 The test you are running does not match the output. 
The command you showed is for `buffered_block_device` test, which handles the allocation problems properly. I have tried this test on your board and it works properly.
However, the output you pasted is for the `tests-mbedmicro-rtos-mbed-threads` test. I haven't handled this test yet, which is indeed described in [#7535](https://github.com/ARMmbed/mbed-os/issues/7535).
So this issue actually duplicates [#7535](https://github.com/ARMmbed/mbed-os/issues/7535).

davidsaada, 2018-08-07:
Just to add: `std::nothrow` works perfectly. No need to change anything in the error handler. Shown failure is in a test that doesn't include it yet (and it will take some time to fix, as it's a complicated test).

0xc0170, 2018-08-07:
Closing as duplicate, lets continue in 7535 and close that one

aashishc1988, 2018-08-07:
@davidsaada My bad, I pasted the wrong output. I was filing multiple issues and got confused. Sorry about that. Please take a look at the updated log. The current test runs with `std::nothrow`, however, in our tests, we run with `-DMBED_TRAP_ERRORS_ENABLED` flag enabled, which will get triggered when memory allocation fails, irrespective of `std::nothrow.`  The idea behind filing this issue is to see if we can figure out a way around it (look at my second comment).

studavekar, 2018-08-07:
re-opening as per above comment from @aashishc1988

davidsaada, 2018-08-07:
@aashishc1988 This issue found me asking the following questions:
1. Is the `-DMBED_TRAP_ERRORS_ENABLED` included in our CI builds? Must say I'm not familiar with it.
2. If so, what's the reason we include it? It seems to nullify the `std::nothrow` directive.
3. I'm running the same command you do (with this flag included) on the same board, but still don't see the problem. Getting the same output, but without the exception. Tried on both IAR and GCC_ARM. What can be the reason for it?
4. I see that in your command the test name doesn't include `mbed-os` as a prefix. This hints that you have an older version of mbed tools. Could this be the reason for the different results?

aashishc1988, 2018-08-07:
@davidsaada I am little unfamiliar with all the terms and environment setup. I will try my best to answer questions to the best of my knowledge and will also tag relevant people.

1. When you say CI builds, I am assuming you meant the build artifacts generated by Jenkins (aws)? If yes, then `-DMBED_TRAP_ERRORS_ENABLED` is included. @studavekar 

2. It does seem to nullify std::throw directive in this case, but my assumption is that the addition of this flag and the corresponding error handler is to catch hold of lot of other problems besides memory allocation failures (I would need to dig in deeper), and I am not even sure if all the tests that call `new` for memory allocation actually use std::throw directive, however we can discuss it further to see if this check for the failure is actually needed in the error handler, given that if proper precaution is taken from the host with regards to memory allocation. Actually, I am not sure that if we expose this flag to customers, it very could be just for internal use. @SenRamakri 

3. Are you also including this flag `-DMBED_HEAP_STATS_ENABLED=1` I think we enable some heap traces that might use more memory. So, are you saying that you dont see `:34::SKIP: Not enough memory for test` , or you do see that but dont see mbed_os error info like in the output log above?

4. I dont quite understand. Can you show me what that command actually looks like? Has the directory structure for the test changed?

davidsaada, 2018-08-07:
Regarding 3: Yes - I'm using the same flags as you do. I simply copied your command (with the exception I broke it to two, in order to run it on a remote board). I am seeing the `:34::SKIP: Not enough memory for test` line, and I expect to see it. It is just the line that tells us that the test was skipped due to insufficient memory. That's OK and it doesn't fail the test. What I don't see is the exception that follows it - `Error Message: Operator new[] out of memory` for instance, which makes your tests fail.

Regarding 4: it's the same as yours, but instead of `features-tests-filesystem-buffered_block_device` I need `mbed-os-features-tests-filesystem-buffered_block_device`. AFAIK, with the newer version of the tools, I am obliged to run it with this mbed-os prefix, something that wasn't true in the older version.

aashishc1988, 2018-08-07:
Interesting.
I updated point 3 above. It says **The addition of mbed_os error handler is fairly new, could it be that you are on older version of mbed_os?**
 Regarding mbed-os as prefix, I need to investigate it more as to why I dont need it. The tools that I am using are all updated to there latest version.

davidsaada, 2018-08-07:
Well. I am updated with the latest mbed-os changes. In addition, I'm pretty sure that the PR that included the std::nothrow changes has passed CI with the error handling mechanism already in (it is fairly new). I really don't know what to say. Only way I can tackle it is if I reproduce it, something that didn't happen yet.
As for the tools, I'm no expert here, so my take on it could be wrong. Saw it in one of the previous PRs I was involved in.

aashishc1988, 2018-08-07:
@davidsaada Can you point me to the CI test log where it passed so that i can compare what is different between the two setups? It's possible that I might be missing something here.

aashishc1988, 2018-08-07:
@davidsaada Can you also try this:
Delete the BUILD directory (or the part directory)
Do a fresh build with this command:
`mbed test --compile -m WIZWIKI_W7500 -t GCC_ARM -DMBED_HEAP_STATS_ENABLED=1 -DMBED_STACK_STATS_ENABLED=1 -DMBED_TRAP_ERRORS_ENABLED=1 -DMBED_ALL_STATS_ENABLED`

davidsaada, 2018-08-08:
Thanks @aashishc1988. I have managed to reproduce the problem with your command. Apparently, it has nothing to do with the `MBED_TRAP_ERRORS_ENABLED` flag, but with the `MBED_STACK_STATS_ENABLED` flag (which is also turned on if the `MBED_ALL_STATS_ENABLED` is set). This flag unleashes a whole bunch of monitoring code, which probably allocates some stuff dynamically (likely without std::nothrow). I'll try to find out where and fix it.

davidsaada, 2018-08-08:
@aashishc1988 [#7730](https://github.com/ARMmbed/mbed-os/pull/7730) should fix it. Please verify.

aashishc1988, 2018-08-08:
@davidsaada https://github.com/ARMmbed/mbed-os/pull/7730 fixes it. Thanks for investigating (and fixing) it :)

davidsaada, 2018-08-08:
@aashishc1988 Thanks for verifying. That PR still needs some work, but glad it's in the right direction.

closed 2018-08-24 by 0xc0170