ARMmbed/mbed-os #7718: LoRaWAN: Not Receiving TX_ERROR after Exhausting Confirmed Message Retries

mattbrown015, 2018-08-07:
### Description
I believe a `TX_ERROR` event should occur if confirmed message retries are exhausted.  Although I haven't found this explicitly stated in the documentation in `LoRaWANInterface.h`.

As far as I can tell, the `TX_DONE` event is generated even if the retries are exhausted.  The stack prints debug saying the the retries have been exhausted:

    [ERR ][LSTK]: Retries exhausted for Class A device

This is scenario is a bit difficult to reproduce; I created the problem by switching off the gateway after successfully connecting.

I tried confirmed messages back in May '18 and I believe I saw the `TX_ERROR` event then.

I shall continue investigating and confirm what I'm saying but I wanted to open an issue in case someone else knows anything about this.

**SHA** 546dafb

Hi @hasnainvirk, could this be an issue in the work you've been doing recently?

### Issue request type
[ ] Question  
[ ] Enhancement  
[X] Bug

hasnainvirk, 2018-08-07:
@mattbrown015 I re-ran the tests and I can confirm that we will receive `TX_ERROR` event when all the retries are exhausted for a CONFIRMED message. Yeah, may be the documentation needs to be improved. TX_ERROR is documented as a general TX_ERROR but it is happening specifically for the case when we do not receive an ack back.

mattbrown015, 2018-08-07:
@hasnainvirk Hhmm, ok, thanks!

I've been distracted by something else this morning and therefore haven't done any more investigation.  Hopefully I'll get back on it soon and let you know what's going on with me. :-)

I tried 0 and 1 retries, BTW.

mattbrown015, 2018-08-08:
Hi @hasnainvirk, I hate to disagree but...

I can see where my problem is and I wonder if the there's a difference between not receiving a response and receiving a response that isn't an ack.  I'm testing my confirmed message by ensuring no response is received.

The RX2 window times out and we end up in `LoRaWANStack::process_reception_timeout`.  `LoRaWANStack::process_reception_timeout` calls `LoRaMac::on_radio_rx_timeout` and `LoRaMac::on_radio_rx_timeout` sets `_mcps_confirmation.status` to `LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT`.

A bit further down `LoRaWANStack::process_reception_timeout` calls `LoRaMac::post_process_mcps_req` and the first thing `LoRaMac::post_process_mcps_req` does is set `_mcps_confirmation.status` to `LORAMAC_EVENT_INFO_STATUS_OK`.  This looks like the mistake to me but I don't know what should happen.

    void LoRaWANStack::process_reception_timeout(bool is_timeout)
    {
        ...
        _loramac.on_radio_rx_timeout(is_timeout);
        // confirmed status correctly set to timeout <==
        ...
        if (slot == RX_SLOT_WIN_2) {
            _loramac.post_process_mcps_req();
            // confirmed status now incorrectly set to OK <==

            if (_loramac.get_mcps_confirmation()->req_type == MCPS_CONFIRMED) {
                if (_loramac.continue_sending_process()) {
                    return;
                } else {
                    tr_error("Retries exhausted for Class A device");
                }
            }

            state_controller(DEVICE_STATE_STATUS_CHECK);
            // status check incorrectly calls TX_DONE because status is OK <==
            state_machine_run_to_completion();
        }
    }

mattbrown015, 2018-08-08:
I tried to simulate receiving a response that didn't contain an ack, to see if that would cause a `TX_ERROR` event, by commenting out the code that see the ack bit.  I.e.

    diff --git a/features/lorawan/lorastack/mac/LoRaMac.cpp 
    b/features/lorawan/lorastack/mac/LoRaMac.cpp
    index 6823e7403..ee7df4151 100644
    --- a/features/lorawan/lorastack/mac/LoRaMac.cpp
    +++ b/features/lorawan/lorastack/mac/LoRaMac.cpp
    @@ -585,8 +586,8 @@ void LoRaMac::handle_data_frame(const uint8_t *const payload,
         if (_params.is_node_ack_requested) {
             if (fctrl.bits.ack) {
                 _mac_commands.clear_command_buffer();
    -            _mcps_confirmation.ack_received = fctrl.bits.ack;
    -            _mcps_indication.is_ack_recvd = fctrl.bits.ack;
    +            // _mcps_confirmation.ack_received = fctrl.bits.ack;
    +            // _mcps_indication.is_ack_recvd = fctrl.bits.ack;
             }
         } else {
             _mac_commands.clear_command_buffer();

But, even with this modified code I see `TX_DONE`.

In other words, I've tried quite hard now to see the `TX_ERROR` and never have.  It makes me wonder what kind of tests your doing?

kivaisan, 2018-08-09:
@mattbrown015 We do test both scenarios you mentioned above;
1) Node sends CONFIRMED message, but the received message is corrupted (in our case means MIC calculation fails)
2) Node sends CONFIRMED message, but server never responds
In both of these cases TX_ERROR event is sent to application.

mattbrown015, 2018-08-09:
@kivaisan Hhhmm, you're sure it does work and I'm sure it doesn't work, tricky! :-)

Can you see the mistake in my reasoning above?

I'll update Mbed, although I don't think any of this has changed for some time.

I'm interested in how you perform your testing because testing is something we've been talking about.

We've created a LoRaWAN end-device but the gateways we use are just off-the-shelf.  So my testing scenarios are pretty much limited to gateway on or gateway off.  Hence testing things like missing a confirmation or ADR is pretty difficult or impossible.

I've tried to test the confirmed messages in 2 ways, switching off the gateway and/or adding _temporary_ debug code to the stack that means I can mess up the RX window opening for the odd transaction.

mattbrown015, 2018-08-09:
@kivaisan and @hasnainvirk Sorry, my bad!

Something has changed between [546dafb](https://github.com/ARMmbed/mbed-os/commit/546dafbadcd7db487f5a111b7ea73c557866b917) (6 days ago) and [1ab05c2](https://github.com/ARMmbed/mbed-os/commit/1ab05c28f68e8e5da0275e671ddaa4ffb7518e94) (today).

Oh, it's this [Mark an error in post processing if ack not recvd](https://github.com/ARMmbed/mbed-os/commit/24c5c58d4f473d37eda2f4de54f9151448e3eb95).  Which @hasnainvirk fixed a couple of weeks ago but obviously it was only merged in the last 6 days.

Sorry. :-)

I'm still interested in testing LoRaWAN end-devices though!

mattbrown015, 2018-08-09:
Here's the PR which describes the fix [LoRaWAN: Reduced priority for automatic uplinks & higher data rate usage for connection establishment #7601](https://github.com/ARMmbed/mbed-os/pull/7601).  Merged 2 days ago.

hasnainvirk, 2018-08-09:
@mattbrown015  Sorry, when I ran tests I was using my pipeline branch which had the commits from the  opened PR. Maybe that's why I didn't see it and I kind of forgot that I was using pipeline branch. That PR has been in the pipeline longer than expected. I hope you are good to go now.

mattbrown015, 2018-08-09:
Hi @hasnainvirk, thanks, I am basically 'good to go now' but I've just noticed something else I don't understand...

I've got confirmed messages selected now but my RF conditions are good so most of the time the first transmission is ack'ed and I see `TX_DONE`.

Occasionally the ack for the first transmission is missed, which in itself isn't a surprise.

But, the retry appears to happen at a different DR which I think doesn't match the comment in LoRaWANInterface.h:

     * Note, that if number of retries is set to 1 or 2, MAC will not decrease
     * the datarate, if the LoRaMAC layer did not receive an acknowledgment.

I've got retries set to 1.

I think your debug output illustrates my point:

    [INFO][LMAC]: RTS = 10 bytes, PEND = 0, Port: 15
    [DBG ][LMAC]: Frame prepared to send at port 15
    [DBG ][LMAC]: TX: Channel=0, DR=5
    [DBG ][LSTK]: Transmission completed
    [DBG ][LSTK]: Awaiting ACK
    [DBG ][LMAC]: Opening RX1 Window
    [DBG ][LMAC]: Opening RX2 Window, Frequency = 869525000
    [DBG ][LMAC]: ACK_TIMEOUT Elapses, Retrying ...
    [DBG ][LMAC]: Trading datarate for range
    [DBG ][LMAC]: TX: Channel=5, DR=4
    [DBG ][LSTK]: Transmission completed
    [DBG ][LMAC]: Opening RX1 Window
    [DBG ][LSTK]: Ack=OK, NbTrials=1

I was expecting the retry to also be at DR5.

What is the intended data-rate adaption during message retransmissions?  Was it your intention to implement the LoRaWAN V1.0 recommendation (section 18.4)?

hasnainvirk, 2018-08-09:
@mattbrown015 Yeah. The documentation is not updated I guess. We try to reduce DR after every 2 transmissions if ack is not received. Mostly, if the network sees that your channel conditions are good and you just happened to drop the ack for some reason (mostly timing mismatch), it will send you a mac command to restore your data-rate. This is how it works at the moment. However, this is one of the items we wish to go through once again. The tradeoff have pros and cons but we wish to study it in a bit more detail.

kivaisan, 2018-08-10:
@mattbrown015 I did some debugging on this and modified our test case a bit and I think there is still some problem in TX_ERROR handling especially in the case where there is no any response from gateway. I need to debug this more.

kivaisan, 2018-08-13:
After debugging, I noticed the problem actually is in case when while waiting ack, we receive somehow corrupted message. TX_ERROR is sent too early (after RX1).

mattbrown015, 2018-08-13:
My specific problem solved and new issue opened for something I hadn't seen so I think this issue can be closed. :-)

closed 2018-08-13 by mattbrown015