ARMmbed/mbed-os #7723: Build system failure (not due to compilation issues)

andresag01, 2018-08-07:
### Description
Compilation of an mbed-os-example-tls example fails with the following error:
```
DEBUG] Output: ./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/api/sleep.c: 7 warnings, 0 errors
Traceback (most recent call last):
  File "/home/andama01/Repos/mbed-os-example-tls/benchmark/mbed-os/tools/make.py", line 293, in <module>
    ignore=options.ignore
  File "/home/andama01/Repos/mbed-os-example-tls/benchmark/mbed-os/tools/build_api.py", line 563, in build_project
    res, _ = toolchain.link_program(resources, build_path, name)
  File "/home/andama01/Repos/mbed-os-example-tls/benchmark/mbed-os/tools/toolchains/__init__.py", line 624, in link_program
    if path.endswith(self.LINKER_EXT)][-1]
IndexError: list index out of range
[mbed] ERROR: "/usr/bin/python" returned error.
       Code: 1
       Path: "/home/andama01/Repos/mbed-os-example-tls/benchmark"
       Command: "/usr/bin/python -u /home/andama01/Repos/mbed-os-example-tls/benchmark/mbed-os/tools/make.py -t uARM -m K64F --source . --build ./BUILD/K64F/UARM -c -v"
       Tip: You could retry the last command with "-v" flag for verbose output
```
I do not know whether the sequence of steps I used to compile the program (see below) is legal, but I would expect mbed-cli to either succeed building or print a suitable error.

Target: K64F
Toolchain: uARM
Tool: mbed-cli v1.7.5
Mbed OS SHA: 273a052 Merge pull request #7691 from 0xc0170/fix_events_style
Steps to reproduce:
* Clone the mbed-os-example-tls repo
* Type the following commands:
```
cd benchmark
mbed deploy
cd mbed-os
git checkout 273a052
mbed compile -t uARM -m K64F
```

[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-08-08:
Thanks for reporting. The problem is that K64F does not support uARM. It fails later during linker finding (thus the error)

References:

linker scripts are located: https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/device
target definition: https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L623

andresag01, 2018-08-08:
@0xc0170: Thank you for your reply.

As I mentioned, I was not sure if what I was trying was legal. However, given that it isn't, ideally the tool would display a more descriptive error which actually points to the problem instead of failing with a python exception. As you pointed out, the information to identify if this combinations are illegal is there in the config files, etc so it might even be possible to detect these errors before even starting compilation

0xc0170, 2018-08-08:
I think so, tools team will review this issue

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-357*

PR #8250 (2018-10-22): Tools: Don't traceback on missing linker script
### Description

The `mbed compile` would traceback when no linker script is found.
This PR changes that behavior to make that into a NotSupportedException,
which has decent user behavior.

Fixes #7723

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Breaking change
commit f8f5bc40 (2018-10-22): Tools: Don't traceback on missing linker script

### Description

The `mbed compile` would traceback when no linker script is found.
This PR changes that behavior to make that into a NotSupportedException,
which has decent user behavior.

Fixes #7723

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Breaking change
closed 2018-10-22 by cmonr