ARMmbed/mbed-os #7736: Retarget Default Console to Log File

AGlass0fMilk, 2018-08-09:
### Description
I'm not sure if this is a currently supported thing, but it would be nice to be able to redirect the standard output (printf, debug, etc) to a log file after a filesystem has been initialized. In my application, I first output debug information to SWO. But to do field testing, I need to log data to an on-board SD card. My libraries make extensive use of the "debug" function that prints out from mbed so I'd like to just capture all information output by that.

To do this, I have made a type of FileHandle class that allows you to swap the underlying FileHandle in runtime (in case submodules obtain the default console once and save the handle). Then I return this in the overloaded `mbed_override_console`. It works at first (outputting debug information over SWO) but I'm not getting anything written out to the file. It is created successfully but there's no data.

Any idea of how to implement such a thing or is it just asking for trouble?

Code for reference:

```
/*
 * SwitchedFile.h
 *
 *  Created on: Aug 9, 2018
 *      Author: gdbeckstein
 */

#ifndef SWITCHEDFILE_H_
#define SWITCHEDFILE_H_

#include "filesystem/FileSystem.h"
#include "platform/FileHandle.h"

//#include "rtos/"

/** SwitchedFile class
 */
class SwitchedFile : public mbed::FileHandle {

public:
    /**
     * Dynamically switched file pointer
     */
    SwitchedFile();

    /**
     * Dynamic file
     * @param[in] ptr Initial pointer used
     */
    SwitchedFile(mbed::FileHandle* ptr) : _ptr(ptr) { }

    /**
     * Switch the file used in operations
     * @param[in] state left or right state (file)
     */
    void switch_file(mbed::FileHandle* ptr) {
    	_ptr = ptr;
    }

    /** Destroy a file
     *
     */
    virtual ~SwitchedFile() { }

    /** Close a file
     *
     *  @return         0 on success, negative error code on failure
     */
    virtual int close() { return _ptr->close(); }

    /** Read the contents of a file into a buffer
     *
     *  @param buffer   The buffer to read in to
     *  @param size     The number of bytes to read
     *  @return         The number of bytes read, 0 at end of file, negative error on failure
     */

    virtual ssize_t read(void *buffer, size_t size) { return _ptr->read(buffer, size); }

    /** Write the contents of a buffer to a file
     *
     *  @param buffer   The buffer to write from
     *  @param size     The number of bytes to write
     *  @return         The number of bytes written, negative error on failure
     */
    virtual ssize_t write(const void *buffer, size_t size) { return _ptr->write(buffer, size); }

    /** Flush any buffers associated with the file
     *
     *  @return         0 on success, negative error code on failure
     */
    virtual int sync() { return _ptr->sync(); }

    /** Check if the file in an interactive terminal device
     *
     *  @return         True if the file is a terminal
     */
    virtual int isatty() { return _ptr->isatty(); }

    /** Move the file position to a given offset from from a given location
     *
     *  @param offset   The offset from whence to move to
     *  @param whence   The start of where to seek
     *      SEEK_SET to start from beginning of file,
     *      SEEK_CUR to start from current position in file,
     *      SEEK_END to start from end of file
     *  @return         The new offset of the file
     */
    virtual off_t seek(off_t offset, int whence = SEEK_SET) { return _ptr->seek(offset, whence); }

    /** Get the file position of the file
     *
     *  @return         The current offset in the file
     */
    virtual off_t tell() { return _ptr->tell(); }

    /** Rewind the file position to the beginning of the file
     *
     *  @note This is equivalent to file_seek(file, 0, FS_SEEK_SET)
     */
    virtual void rewind() { return _ptr->rewind(); }

    /** Get the size of the file
     *
     *  @return         Size of the file in bytes
     */
    virtual off_t size() { return _ptr->size(); }

private:

    mbed::FileHandle* _ptr;
};

#endif /* SWITCHEDFILE_H_ */
```

main.cpp excerpt
```
mbed::File log_file;

static mbed::SerialWireOutput swo;

SwitchedFile debug_out(&swo);

mbed::FileHandle *mbed::mbed_override_console(int fd) {
	return &debug_out;
}

// Later on, in filesystem_init()...
error = log_file.open(&fat_fs, "log.txt", O_WRONLY | O_CREAT);
	if(error == 0)
	{
		debug("filesystem: successfully opened log file!\n");
		debug("            switching output to log file...\n");
		debug_out.switch_file(&log_file);
	}

```
### Issue request type

[ X ] Question  
[ ] Enhancement  
[ ] Bug

AGlass0fMilk, 2018-08-09:
Simply trying to write to the log file, I discovered I needed to call .sync after .write. However, I'm not sure how to do this for the stderr output since mbed_debug does not call .sync after .write.

How does one synchronize the file in this case?

kjbracey-arm, 2018-08-17:
I believe you should be able to `freopen(stdout)` to send to a file, so your rather clever trick shouldn't be necessary.  That should work fine for anything with a pathname, such as a real file on a `FileSystem`.

Only limitation I'm aware of is that there's no `fdreopen` for when you only have a `FileHandle` without a pathname, eg a plain `UARTSerial` device.

As to the sync of `stderr`, you can do that with `fsync(STDERR_FILENO)`. I wouldn't necessarily recommend that after every `write` though, I would guess it might cause excessive wear, depending on the `FileSystem`. I'd be inclined to rate-limit that in some way.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-360*

closed 2019-10-31 by ciarmcom