ARMmbed/mbed-os #7743: NRF52 Can not initialize Pwm

MateuszMaz, 2018-08-09:
### Description
After following commit: 2ba88ba8f6a98a80f22187f0bb47aba0b96f2773, that was made soon after mbed-os-5.9.0 it is no longer possible to create an object of PwmOut class for device NRF52832. The error exist in every version of mbed-os since 5.9.1 to current 1ab05c28f68e8e5da0275e671ddaa4ffb7518e94.

**Mbed error string:**
>++ MbedOS Error Info ++
Error Status: 0x80FF0100 Code: 256 Module: 255
Error Message: Fatal Run-time error
Location: 0x3683D
Error Value: 0x0
Current Thread: Id: 0x2000534C Entry: 0x32D45 StackSize: 0x1000 StackMem: 0x20005398 SP: 0x200062C8 
-- MbedOS Error Info --
nrf failure at .\mbed-os\targets\TARGET_NORDIC\TARGET_NRF5x\TARGET_SDK_14_2\drivers_nrf\pwm\nrf_drv_pwm.c:192

**Target:** 
NRF52832
**Mbed-OS SHA**
current master: 1ab05c28f68e8e5da0275e671ddaa4ffb7518e94
**Toolchain**
GCC_ARM 7 2018-q2-update

**Steps to reproduce**

1. Create a new mbed program
2. checkout mbed-os to 1ab05c28f68e8e5da0275e671ddaa4ffb7518e94
3. Create main.cpp with following declaration PwmOut led(LED1); 
4. Run and read error from terminal

### Issue request type
[ ] Question  
[ ] Enhancement  
[x] Bug

0xc0170, 2018-08-09:
Why that error is happening ? Based on the referenced commit, I would assume that previously it was happening , it was just silenced ?

MateuszMaz, 2018-08-09:
During pwm initialization we check if no other object of current pwm is already initialized
>/* Make sure PWM instance is not running before making changes. */
    nrf_drv_pwm_uninit(&nordic_nrf5_pwm_instance[obj->instance]);

The problem is since nref asserts were redirected to mbed error (2ba88ba8f6a98a80f22187f0bb47aba0b96f2773), 
line 192 in nrf_drv_pwm.c of function nrf_drv_pwm_uninit() visible below:
>ASSERT(p_cb->state != NRF_DRV_STATE_UNINITIALIZED);

now returns an error because:

>void assert_nrf_callback(uint16_t line_num, const uint8_t *p_file_name)
{
    ASSERT_TRUE(false, (void) 0);
}

was changed to:

>void assert_nrf_callback(uint16_t line_num, const uint8_t *p_file_name)
{
    error("nrf failure at %s:%d", p_file_name, line_num);
}

It was silenced. When we initialize object first time: this thing     nrf_drv_pwm_uninit(&nordic_nrf5_pwm_instance[obj->instance]); 
in pwmout_api.c always returned failed asert

0xc0170, 2018-08-10:
@MateuszMaz Can you fix it ?

MateuszMaz, 2018-08-13:
here is pr #7779

0xc0170, 2018-08-24:
@AGlass0fMilk Does https://github.com/ARMmbed/mbed-os/pull/7779 fixes the issues you are having?

bmcdonnell-ionx, 2018-09-14:
[Mbed OS 5.9.7 release notes](https://os.mbed.com/blog/entry/mbed-os-597-released/) links to this issue as "fixed". Consider closing.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTDEV-1598*