ARMmbed/mbed-os #7756: OS2 tests scripts doesn't work any more

jeromecoutant, 2018-08-10:
### Description

Since #7183, OS2 tests can't be executed

@theotherjimmy @cmonr 

````
python tools/singletest.py -M muts_test.json -i muts_compilation.json -V -j 4

Traceback (most recent call last):
  File "tools/singletest.py", line 262, in <module>
    if (singletest_in_cli_mode(single_test)):
  File "C:\github\mbed\tools\test_api.py", line 1551, in singletest_in_cli_mode
    test_summary, shuffle_seed, test_summary_ext, test_suite_properties_ext, build_report, build_properties = single_test.execute()
  File "C:\github\mbed\tools\test_api.py", line 655, in execute
    self.execute_thread_slice(q, target, toolchains, clean, test_ids, self.build_report, self.build_properties)
  File "C:\github\mbed\tools\test_api.py", line 584, in execute_thread_slice
    handle_results = self.handle(test_spec, target, toolchain, test_loops=test_loops)
  File "C:\github\mbed\tools\test_api.py", line 1029, in handle
    handle_result = self.handle_mut(mut, data, target_name, toolchain_name, test_loops=test_loops)
  File "C:\github\mbed\tools\test_api.py", line 944, in handle_mut
    if not exists(image_path):
  File "C:\Python27\lib\genericpath.py", line 26, in exists
    os.stat(path)
TypeError: coercing to Unicode: need string or buffer, list found
````

### Issue request type

[ ] Question  
[ ] Enhancement  
[x] Bug

jeromecoutant, 2018-08-14:
Hi
I am available to test some patch :-)

cmonr, 2018-08-14:
Thanks @jeromecoutant!

Would you happen to have tests for a specific target that we could test against so that we can reproduce the issues?

jeromecoutant, 2018-08-16:
No, issue is not related to a specific target, you can try "python tools/singletest.py ..." to the target you have

jeromecoutant, 2018-09-18:
Note that OS2 tests was OK with last rev 163 lib as #7183 has not been included in 5.9 branch.
Next mbed_lib can't be released before solving this issue.

0xc0170, 2018-09-18:
Thanks @jeromecoutant ,we will review this issue

@ARMmbed/mbed-os-tools

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-395*

jeromecoutant, 2018-10-08:
Is OS2 support stopped for 5.10 release ?

0xc0170, 2018-10-08:
> Is OS2 support stopped for 5.10 release ?

No. We should fix this

jeromecoutant, 2018-11-16:
Hi

I could see that mbed_lib_rev164 has been released
Whereas no test is possible....?

screamerbg, 2018-12-14:
@cmonr @jeromecoutant @bridadan Any update on this?

bridadan, 2018-12-14:
Fix up: https://github.com/ARMmbed/mbed-os/pull/9107

PR #9107 (2018-12-17): Fixing execution of OS2 tests
### Description

This fixes #7756.

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change
closed 2018-12-17 by cmonr