ARMmbed/mbed-os #7784: features/nanostack/mbed-mesh-api/source/NanostackMemoryManager.cpp:25: incomplete type is not allowed

andrewc-arm, 2018-08-14:
Hi,

### Description
My partner WIZnet compiled with IAR 7.40 and got following error.

> Compile [ 50.9%]: NanostackMemoryManager.cpp
> Compile: C:\Program Files (x86)\IAR Systems\Embedded Workbench 7.2\arm\bin\iccarm --no_wrap_diagnostics -e --diag_suppress=Pa050,Pa084,Pa093,Pa082 -Oh --enable_restrict --cpu Cortex-M0 --thumb --dlib_config DLib_Config_Full.h --c++ --no_rtti --no_exceptions --guard_calls --no_static_destruction -D__CORTEX_M0 -DCMSIS_VECTAB_VIRTUAL -D__MBED__=1 -DTARGET_LIKE_MBED -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -D__CMSIS_RTOS -DTARGET_WIZwiki_W7500 -DTARGET_W7500x -DARM_MATH_CM0 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_M0 -DTARGET_WIZWIKI_W7500 -DTOOLCHAIN_IAR -DMBED_HEAP_STATS_ENABLED=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_RELEASE -DDEVICE_SERIAL=1 -DMBED_BUILD_TIMESTAMP=1534207149.34 -DTARGET_WIZNET -DDEVICE_PORTIN=1 -DTARGET_FF_ARDUINO -DDEVICE_SPI=1 -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h" -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DTARGET_LIKE_CORTEX_M0 -DTARGET_CORTEX_M -f D:\Wiznet\Git\mbed\mbed-os\BUILD\tests\WIZWIKI_W7500\IAR\.includes_d28b14682197ac3995b1d491852c207b.txt --preinclude=D:\Wiznet\Git\mbed\mbed-os\BUILD\tests\WIZWIKI_W7500\IAR\mbed_config.h --dependencies BUILD\tests\WIZWIKI_W7500\IAR\features\nanostack\mbed-mesh-api\source\NanostackMemoryManager.d -l BUILD\tests\WIZWIKI_W7500\IAR\features\nanostack\mbed-mesh-api\source\NanostackMemoryManager.s.txt -o BUILD\tests\WIZWIKI_W7500\IAR\features\nanostack\mbed-mesh-api\source\NanostackMemoryManager.o .\features\nanostack\mbed-mesh-api\source\NanostackMemoryManager.cpp
> [Error] NanostackMemoryManager.cpp@27,0: [Pe070]: incomplete type is not allowed
> [DEBUG] Return: 2
> [DEBUG] Output: 
> [DEBUG] Output:       uint8_t mem[];
> [DEBUG] Output:               ^
> [DEBUG] Output: "D:\Wiznet\Git\mbed\mbed-os\features\nanostack\mbed-mesh-api\source\NanostackMemoryManager.cpp",27  Error[Pe070]: incomplete type is not allowed
> Failed to build library

I think the code is incorrect. I think it should have been "uint8_t *mem;" instead. Am I missing something?
Could you please review the code?

Thanks,
Andrew

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug

pan-, 2018-08-14:
@andrewc-arm IAR 7.40 is not supported; we do support IAR 7.80. 

The issue you mention refers to a C99 flexible array declaration in C++ mode. 7.40 had some [issues](http://supp.iar.com/FilesPublic/UPDINFO/011261/arm/doc/infocenter/iccarm_history.ENU.html) with flexible array members.

cmonr, 2018-08-14:
@andrewc-arm For reference: https://os.mbed.com/docs/v5.9/tools/exporting.html

Is there a reason that your partner is using 7.40 instead of 7.8?

andrewc-arm, 2018-08-16:
OK. I will ask WIZnet to try IAR 7.80 instead of 7.40. Thanks for the quick feedback!

cmonr, 2018-08-16:
@andrewc-arm Feel free to open another issue if the problem persists.

closed 2018-08-16 by cmonr