ARMmbed/mbed-os #7800: What is the proper way to disable RTOS

pmancele, 2018-08-16:
### Description
Hello,
I know that [this page](https://os.mbed.com/blog/entry/Reducing-memory-usage-by-tuning-RTOS-con/) is usually linked to explain how to diable RTOS but it there a way to disable RTOS through mbed compilation time configuration ? (Through the `mbed_app.json` file)

I tried this:
```json
{
	"target_overrides": {
	    "*": {
		"rtos.present": "0",
		"events.present": "0"
	    }
	}
}
```
--> The RTOS and EVENTS are not compiled but the others features that needs RTOS or EVENTS throw compilation errors.

The only way for me to disable RTOS is to create a `.mbedignore` file with the following content:

```
mbed-os/rtos/*
mbed-os/features/cellular/
mbed-os/features/FEATURE_CLIENT/*
mbed-os/features/FEATURE_COMMON_PAL/*
mbed-os/features/FEATURE_UVISOR/*
mbed-os/features/frameworks/*
mbed-os/features/net/*
mbed-os/features/netsocket/*
mbed-os/features/nanostack/*
mbed-os/features/nvstore/*
mbed-os/features/storage/*
mbed-os/features/device_key/*
mbed-os/features/lorawan/*
mbed-os/events/*
mbed-os/features/lwipstack/*
```

Is it the best way ?
Thanks, Pierre-Marie.

    [X] Question
    [ ] Enhancement 
    [ ] Bug

deepikabhavnani, 2018-08-20:
Yes `.mbedignore` is the way to disable RTOS, as we do not scan features based on RTOS support.

deepikabhavnani, 2018-08-24:
> Yes .mbedignore is the way to disable RTOS, as we do not scan features based on RTOS support.

I believe this answers the question, hence closing. Feel free to reopen if required.

bsturgess, 2018-11-22:
How can i do this on the online compiler?

pmancele, 2018-11-23:
@bsturgess I tried to create a `.mbedignore` file online but its not working indeed ..

So I don't know how to do it on the online compile

adbridge, 2019-03-04:
I've changed the labels on this so that it will open a new Jira issue and mirror correctly

j3hill, 2019-03-08:
@pmancele It's been a few months since we've seen any activity on this issue, and it appears that it has been resolved.  Please confirm, if possible.

linlingao, 2019-05-29:
The question has been answered. Close issue.

closed 2019-05-29 by linlingao