ARMmbed/mbed-os #7804: RF52 serious behavioural regressions in RTC

NeilMacMullen, 2018-08-16:
GCC_ARM
nRF52 platform.
mbed-os-5.9.4

See https://github.com/ARMmbed/mbed-os/issues/7763 for the likely breaking change.

The following code demonstrates that the seconds and minutes are NOT being correctly reset when calling set_time. 

Further, other strange behaviour includes that described in https://github.com/ARMmbed/mbed-os/issues/7763 where setting the clock (backwards?) can sometimes cause it to 'pause' for 35 minutes (1<<31 us).

I'm pretty sure I've also seen the clock  drift significantly (several minutes over the course of an hour).

```
#include "mbed.h"

#include "segger_rtt.h"

int main()
{
    for (int hour = 23; hour >= 0; hour--)
    {
        for (int z = 0; z < 12; z++)
        {
            SEGGER_RTT_printf(0, "resetting time using hour %d...\n", hour);

            struct tm t
            {
            };
            t.tm_year = 2018 - 1900;
            t.tm_mon = 8 - 1;
            t.tm_mday = 16;
            t.tm_hour = hour;
            t.tm_min = 1;
            t.tm_sec = 1;
            set_time(mktime(&t));

            for (int c = 0; c < 10; c++)
            {
                time_t seconds = time(NULL);
                SEGGER_RTT_printf(0, "%s", ctime(&seconds));
                wait_ms(1000);
            }
        }
    }
    return 0;
}
```

NeilMacMullen, 2018-08-16:
@MarceloSalazar  @pan-  @mprse

NeilMacMullen, 2018-08-16:
set_time seems to have been broken between 5.8.6 ( expected behaviour) and 5.9.0 (incorrect)

NeilMacMullen, 2018-08-16:
For good measure I verified it is also broken on 5.9.5

TacoGrandeTX, 2018-08-17:
@NeilMacMullen We're still root-causing this issue, but a work-around to the problem with set_time() is to re-introduce the "RTC" property to the "Device Has" list in `targets.json` for the MCU_NRF52832.  We see that working properly with your test code.  However, it's not clear what other ramifications that will have on your end app.

NeilMacMullen, 2018-08-17:
Thanks @TacoGrandeTX I'll do some testing with this today and let you know of any relevant findings.

NeilMacMullen, 2018-08-17:
On the basis of some light testing, this does look to have fixed the issues I was seeing and even works with the low-power workaround I describe in #7763.  Thanks!

TacoGrandeTX, 2018-08-17:
@NeilMacMullen Thanks for the report - that's good to hear.  Even without DEVICE_RTC we are still using RTC1 (running at 32kHz) for updating our internal ticker for time-keeping.  Our internal rtc_lp_base is being set appropriately by set_time() but we might be mishandling the internal ticker. We will have more for you early next week.

NeilMacMullen, 2018-08-18:
It seems I spoke too soon :(  Although I haven't managed to find a simple repro case, the esp8266-driver is now failing consistently (with RTC added).  It looks like it's using the Timer implementation and this is not returning sensible values for read_ms but I haven't been able to verify directly.

NeilMacMullen, 2018-08-20:
Further to the above comment..... The difference in behaviour actually occurs when there are two threads in the system, a producer 'A' and a consumer 'B' .   

A is running a tight loop using wait_ms(1) to interject some idle time between checks of a peripheral.  
B is busy-waiting in a while(1) loop (checking a status flag set by B.   

In this scenario, when the RTC feature is added, thread A stalls on the wait_ms(1) presumably because the scheduler is not pre-empting B at all.  

Adding a small wait inside the  B loop allows the code to work correctly (as does reverting the RTC feature).

Clearly, a better design would be for the code in B to use waits but it's interesting the behaviour changes as RTC is added/removed.

FYI, the B code is the get/putc code in https://os.mbed.com/teams/ESP8266/code/esp8266-driver/file/6946b0b9e323/ESP8266/ATParser/ATParser.cpp/

TacoGrandeTX, 2018-08-20:
@NeilMacMullen I'm testing a fix that is looking good.  The timer used to keep track of usec was not being reset on set_time():

```
void set_time(time_t t)
{
    _mutex->lock();
    if (_rtc_init != NULL) {
        _rtc_init();
    }
    if (_rtc_write != NULL) {
        _rtc_write(t);
        _rtc_lp_timer->reset();
    }
    _mutex->unlock();
}
```

 Will update again ASAP (hopefully with a pull request).

TacoGrandeTX, 2018-08-21:
@NeilMacMullen - please see https://github.com/ARMmbed/mbed-os/pull/7849

NeilMacMullen, 2018-08-22:
Thanks Ralph - does this mean the DEVICE_RTC change is now redundant?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTDEV-1603*

0xc0170, 2018-11-09:
> @NeilMacMullen - please see PR #7849

@NeilMacMullen Is this still valid issue and present on master?

closed 2018-11-12 by ciarmcom