ARMmbed/mbed-os #7809: Cortex-M Fault Exception handlers - Not compatible to v8M

deepikabhavnani, 2018-08-16:
### Description
Cortex-M Fault Exception handlers, when enabled, assembly file build (except.S) fails with v8M devices.

[ERROR] warning: 'thumbv4t' is unsupported in this version of the product
error: -mcmse is not supported for arm7tdmi
Fatal error: A1905U: Pre-processor step failed for '.\mbed-os\rtos\TARGET_CORTEX\TARGET_CORTEX_M\TOOLCHAIN_ARM\except.S'
1 Error, 0 Warnings

### Issue request type
[ ] Question  
[ ] Enhancement  
[X] Bug

deepikabhavnani, 2018-08-16:
CC @gaborkertesz - Instruction set used in `except.S` is not compatible to ARMv8.

deepikabhavnani, 2018-08-17:
```
--- a/rtos/TARGET_CORTEX/TARGET_CORTEX_M/mbed_rtx_fault_handler.c
+++ b/rtos/TARGET_CORTEX/TARGET_CORTEX_M/mbed_rtx_fault_handler.c
@@ -88,7 +88,7 @@ MBED_NOINLINE void print_context_info(void)
     //Capture CPUID to get core/cpu info
     mbed_error_printf("\nCPUID: %08X", SCB->CPUID);

-#if !defined(TARGET_M0) && !defined(TARGET_M0P)
+#if !defined(TARGET_M0) && !defined(TARGET_M0P) && !defined(TARGET_M23)
     //Capture fault information registers to infer the cause of exception
     mbed_error_printf("\nHFSR : %08X"
                     "\nMMFSR: %08X"

```

SenRamakri, 2018-08-17:
@deepikabhavnani and I debugged this issue, and we found that adding the above fix can fix compilation issue we were originally seeing with M2351. And with recent fixes Deepika mentioned we are able to compile for Musca. But in general we are not able to enable fault-handler dump for ARMv8M as we are going to secure side on a fault exception and we don't have the fault handler implementation on secure side. There needs to be more discussions on what needs to be enabled/brought into secure side and thus for now we may keep the fault handler disabled for Musca. Other option is move this fault-handler implementation to TF-M side for v8Ms.

gaborkertesz, 2018-08-17:
I also can build for Musca with these fixes and renaming the startup code to *.S from *.s as suggested.
I'll discuss the fault handlers issue with the TFM team, but probably the fault handlers should be in the TFM secure side implementation anyway.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-376*

closed 2019-04-25 by ciarmcom