ARMmbed/mbed-os #7811: Support SPI flash on LPC546XX

maclobdell, 2018-08-17:
### Description
Target: LPCXpresso54628 or LPCXpresso54608
Toolchain: N/A
Mbed Os Version: 5.9.5

To support applications that require storage (e.g. Mbed Cloud Client) it is required to have SPI flash, SD Card, or extra internal flash, etc.  

Luckily, the LPCXpresso546xx boards have a SPI flash on board.  

However, on the schematic the pins are connected to the SPIFI interface, not SPI.  

To support this memory, a SPI implementation needs to be added to map SPI functions to SPIFI, or better yet add support for Quad SPI interface using the SPIFI.

### Issue request type
[ ] Question  
[X] Enhancement  
[ ] Bug

maclobdell, 2018-08-17:
@mmahadevan108

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-6202*

mmahadevan108, 2019-01-11:
@maclobdell SPI support is now available on LPC546XX

maclobdell, 2019-01-14:
@mmahadevan108 - Just to confirm, the LPCXpresso546XX now has support for QSPI.  Thanks a lot for that!  The next step is to test it with the QSPIF block device driver.  You have done that, right?  The next step after that is to add QSPIF storage as the default storage for this platform.  In targets.json for the "LPC546XX" target, you can add this: 

```
 "components_add": ["QSPIF"]   
```

Then this can return a pointer to the block device.  
```
BlockDevice *bd = BlockDevice::get_default_instance();
```

Can you confirm that the default configuration of QSPIF will work?  See \mbed-os\components\storage\blockdevice\COMPONENT_QSPIF\mbed_lib.json

specifically, I'd like to confirm that these default settings are okay.  
```
       "QSPI_POLARITY_MODE": 0,
        "QSPI_FREQ": "40000000"
```

maclobdell, 2019-01-14:
I went ahead and tested it and made a pull request!  
https://github.com/ARMmbed/mbed-os/pull/9374

Thanks!  
I'm closing this issue.  It is resolved.

closed 2019-01-14 by maclobdell