ARMmbed/mbed-os #7827: STM32L4 enter critical section for a long time

kevinong0108, 2018-08-19:
### Description
set_time() will occur system enter critical section for a long time (around 1 second) when DEVICE_LPTICKER is enabled on STM32L476.

The following code is in rtc_api.c, there will spend time to wait in enter critical section.
Why need this part?
```
#if DEVICE_LPTICKER && !MBED_CONF_TARGET_LPTICKER_LPTIM
    while (Read_SubSeconds != (RTC->SSR)) {
    }
#endif /* DEVICE_LPTICKER && !MBED_CONF_TARGET_LPTICKER_LPTIM */
```
I am not sure how about other series.

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug

0xc0170, 2018-08-20:
@ARMmbed/team-st-mcd Please review

LMESTM, 2018-08-20:
@kevinong0108 Thanks for the question. 

This code is active is the below case
#if DEVICE_LPTICKER && !MBED_CONF_TARGET_LPTICKER_LPTIM
which applies to devices who want to use LPTICKER feature but do not have a LPTIM HW. 

In this configuration, the RTC hardware will be used for 2 purposes: For the standard time & date management, but also the sub-second part is used as Low Power Ticker which will be used for scheduling purposes. I think the code tries to sync the new time with and the ticker to avoid mis-alignments of some low period that have been scheduler. But maybe there is a side effect that needs to be analyzed. My colleague who wrote this code is out of office and will be back in 2 weeks, I'd prefer to wait for his return so that he can confirm .

In the meantime, when it comes to STM32L4, I think that the LPTIM HW is available so that I would advise to also add the below config to your device. this should avoid that your L4 device is impacted by this side effect. 

            "lpticker_lptim": {
                "help": "This target supports LPTIM. Set value 1 to use LPTIM for LPTICKER, or 0 to use RTC wakeup timer",
                "value": 1
            }

Br
Laurent

mattbrown015, 2018-08-20:
There's a discussion about the difference between using LPTIM and the RTC on the STM32L4 here, [How to Achieve Long Periods of Deep Sleep on STM32 #7156](https://github.com/ARMmbed/mbed-os/issues/7156).

In summary, on the STM32L4 the LPTIM is only 16-bits and therefore it will only achieve a deepsleep of a few seconds.  The RTC has to be used in order to achieve longer periods of deepsleep.

The code highlighted is important to me because I have enabled tickless in RTC mode and I use `settime()`.  Having said that, I haven't noticed it causing a problem in practice. :-)

kevinong0108, 2018-08-20:
@LMESTM and @mattbrown015  Thanks for your reply, fully understand :)

closed 2018-09-01 by kevinong0108