ARMmbed/mbed-os #7829: REALTEK_RTL8195AM IPv6 build fails

juhaylinen, 2018-08-20:
### Description
Target: REALTEK_RTL8195AM
Toolchain: GCC_ARM
mbed-cli 1.5.0
Mbed OS versions 5.9, 5.9.5 and master (sha f15dbf2c3)
Steps to reproduce:
For example, import mbed-os-example-wifi
```
mbed import mbed-os-example-wifi
cd mbed-os-example-wifi
```
Enable IPv6
```
"target_overrides": {
        "*": {
            "platform.stdio-convert-newlines": true,
            "lwip.ipv4-enabled": false,
            "lwip.ipv6-enabled": true
        },
```
Build
`mbed compile -t GCC_ARM -m REALTEK_RTL8195AM`

Fixed once in this commit https://github.com/ARMmbed/mbed-os/pull/7056

This commit https://github.com/ARMmbed/mbed-os/pull/7669 brings back the "dhcps.h" and "dhcps.c" files which cause the issue in Mbed OS master

[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-08-21:
cc @ARMmbed/team-realtek

samchuarm, 2018-08-22:
Hi @prashantrar @M-ichae-l , can you please help address this build fail asap? Thanks

M-ichae-l, 2018-08-23:
@samchuarm 
For the #7669, It brings back the "dhcps.h" and "dhcps.c".  Please ask the PR creator to remove the two files (remove empty folder also). RTL8195AM does not need the two files. Realtek team would like to review the related commit before it merged. Thank you!

0xc0170, 2018-08-23:
Thanks @M-ichae-l . I'll revert them soon

samchuarm, 2018-08-28:
It seems this is fixed now?

commit 16d23702 (2018-08-28): Merge pull request #7873 from 0xc0170/fix_realtek_ipv6

realtek: fix #7829 - remove dhcps files
commit ab029b5f (2018-08-28): realtek: fix #7829 - remove dhcps files
closed 2018-08-28 by 0xc0170