ARMmbed/mbed-os #7842: Context switch during driver access

KarolisMil, 2018-08-21:
### Description
This is more of conceptual discussion topic.
Regarding current OS configuration, it is running with preemptive round robin scheduler.
There is a possibility, that context switch will happen when CPU is executing drivers' code, accessing registers etc., which may sometimes be time critical or cannot (must not) be delayed for undefined amount of time other context may take.
Mutexes synchronize the shared resource, but does not serve for issue described above.
It is a proposal/discussion topic to add unified API/mechanism to make it available to disable context switching during critical code accesses.
Disabling IRQs (OS critical sections) does not serve this purpose, because that may overflow SysTick, miss OS tick interrupt and mess up system time management. IRQ usually does allowed when executing drivers' code, since ISRs are very fast, not like scheduling higher priority routines etc.
RTX5 support scheduler locking. This may be the thing to use. This won't block OS time following and time intervals counting.
Maybe something else, not necessary scheduler locking would solve that.
Also, this is an open topic and I may be wrong. Please feel free to add/correct/comment this topic.

### Issue request type
[ ] Question  
[ X ] Enhancement  
[ ] Bug

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTDEV-1617*

evedon, 2019-09-13:
Closing topic due to inactivity.

closed 2019-09-13 by evedon