ARMmbed/mbed-os #7845: CI may need to exclude low resource boards

davidsaada, 2018-08-21:
### Description
Upon the addition of new boards with low resources (mainly RAM, but also flash), few of our tests started failing on these boards. In order to prevent these CI failures, tests were modified to get skipped if these resources cannot be allocated. This solution is sub-optimal and adds overhead in our test code. We should think of a mechanism to distinguish such boards from the "normal" ones and exclude them from our CI, either altogether or partially.

### Issue request type
[ ] Question  
[x] Enhancement  
[ ] Bug

davidsaada, 2018-08-22:
@dannybenor @OPpuolitaival @bulislaw

dannybenor, 2018-08-22:
@neilwjackson

bulislaw, 2018-08-22:
Guys to accelerate that it would be better if you use our PREQ process. Create a PREQ and assign it to OP for review.

cmonr, 2018-08-22:
@c1728p9

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-379*

closed 2019-10-31 by ciarmcom