ARMmbed/mbed-os #7852: Conflicting CPU architectures when building with gcc for NUCLEO_F429ZI.

BenjaminPMLovegood, 2018-08-22:
### Description
I used mbed-cli to download "mbed-os-example-blinky" and tried building it for target 'NUCLEO_F429ZI' with arm-none-eabi-gcc, but it failed when linking
```
link: mbed-os-example-blinky.elf
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/crt0.o: Conflicting CPU architectures 13/1
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/crt0.o
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libc.a(lib_a-setjmp.o): Conflicting CPU architectures 13/1
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libc.a(lib_a-setjmp.o)
collect2: error: ld returned 1 exit status
```

toolchain : ARM_GCC(arm-none-eabi-gcc 6.3.1)
tools : mbed-cli 1.7.5 on Python 3.6.5
mbed : #f8b140f8d
os : ubuntu 18.04.1 LTS

### Issue request type

[ ] Question  
[ ] Enhancement  
[x] Bug

0xc0170, 2018-08-22:
This looks like an environment bug. Is this related : https://bugs.launchpad.net/ubuntu/+source/gcc-arm-none-eabi/+bug/1767223 ?

BenjaminPMLovegood, 2018-08-23:
Hi, @0xc0170 
Yes it was a problem about gcc-arm-none-eabi on ubuntu 18.04.1.
I've installed another version of it manually and solved this problem.
Thanks for your advice.

closed 2018-08-23 by 0xc0170

ispkorte, 2018-12-19:
i have same problem on MINT 19; wich of GCC-ARM Version can use ?

jeromecoutant, 2018-12-20:
Hi

GCC download link is available https://os.mbed.com/docs/v5.11/tools/index.html

melvyniandrag, 2019-03-25:
Error still present on ubuntu 18.04. Installed gcc with 'apt-get install gcc-arm-none-eabi'. Worked on ubuntu 16 a few days ago. Both standard installs, no modifications of apt sources.

cmonr, 2019-03-25:
@melvyniandrag This sounds like a problem with either the `gcc-arm-none-eabi` package, or Ubuntu 18.04. We don't manage either package.

You can try what @BenjaminPMLovegood did (https://github.com/ARMmbed/mbed-os/issues/7852#issuecomment-415263190) or contact the maintainers for either the package or Ubuntu in their forums.