ARMmbed/mbed-os #7859: PR submission form to include [ ] Test as an option

paul-szczepanek-arm, 2018-08-22:
### Description

Please add:
[ ] Test
as an option on the PR submission form or add instructions in the comment explaining which existing option should be chosen for PR with Tests only.

### Issue request type

[ ] Question  
[X] Enhancement  
[ ] Bug

paul-szczepanek-arm, 2018-08-22:
@ARMmbed/mbed-os-maintainers I guess functionally this would be the same as refactor?

0xc0170, 2018-08-22:
I was considering it as well. At the moment, updating/adding tests - it's fix most of the time (does not change functionality, no breaking change there), only when adding a new test then it is not that clear.

What others think?

cmonr, 2018-08-24:
@paul-szczepanek-arm Something we want to avoid is adding too many checkboxes/options to the list, since we want to encourage users to only check one thing. I think this was the original reason why we left "Tests" off as its own thing.

> @ARMmbed/mbed-os-maintainers I guess functionally this would be the same as refactor?

It's more nuanced than that. Sometimes new tests are added, making them closer to features, whereas other times tests are being fixed because they're not operating correctly.

I think that adding this in makes sense since we're now going to have even more tests getting added with the addition of unit tests. BUT, with this addition, we (maintainers) should also get stricter on making PRs only have one thing, unless there's a good reason (tests next to a new feature as a single PR?).

paul-szczepanek-arm, 2018-09-04:
Looks like a fair compromise to me. Is this moving forward then?

cmonr, 2018-09-04:
I'd be fine with seeing the change in a new PR. @adbridge @NirSonnenschein Thoughts?

0xc0170, 2018-09-06:
> Tests whether new or fixed can go into both patch and minor releases and thus I think having a 'Test changes' checkbox could work. However I am wondering if this will get confusing for people when we insist on a new test alongside new functionality....

Bigger wins in this case (functionality change). From what I see recently as PR, and expect more - docs and tests.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTMORF-2457*

0xc0170, 2018-11-07:
This was resolved via 8237

closed 2018-11-07 by 0xc0170