ARMmbed/mbed-os #7863: Intelisense in VS Code

Alex-EEE, 2018-08-22:
### Description

VS Code export is working great for me, thanks for that!  Has anyone been able to get full Intelisense working?  Mine keeps complaining about finding header files for the c runtime library.  @janjongboom I know you're a fan, any ideas?

### Issue request type

[x ] Question  
[ ] Enhancement  
[ ] Bug

janjongboom, 2018-09-03:
@Alex-EEE You have to include the header files from the GCC ARM Toolchain, plus only include the files for your target (the exporter does that) but it never worked 100% for me... @arekzaluski you've set this all up for Mbed Studio, how do you do it there?

arekzaluski, 2018-09-13:
@janjongboom @Alex-EEE Indeed you need to manually include path to header files located in GCC ARM. We are working on a solution that will give path to all required folders that need to be included. I will open a PR in mbed-os tools as soon as we will have it ready.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTDEV-1618*

DBS06, 2018-10-22:
Yes I did, but it was a fight against dragons.
First update your vscode and all plugins.
Backup you c_cpp_properties.json file, remove everything and add and fill out the following:
```c++
{
    "configurations": [
        {
            "name": "MBED",
            "intelliSenseMode": "clang-x64",
            "compilerPath": "PATH_TO_YOUR_COMPILER" // i.e. C:/.../GNU Tools ARM Embedded/6 2017-q2-update/bin/arm-none-eabi-g++.exe",
            "limitSymbolsToIncludedHeaders": true,
            "cStandard": "c99",
            "cppStandard": "c++11",
            "defines": [
                // put the exporter generated definitions here and your own definitions
                "MBED_BUILD_TIMESTAMP=1520864281.56",
                "MBED_CONF_RTOS_PRESENT=1",
                "MBED_TICKLESS",
                
            ],
            "includePath": [
                // put the exporter generated include paths here and your own path
                "${workspaceRoot}/*",
                "${workspaceRoot}/mbed-os/*",
                // ...
            ],
            "browse": {
                "path": [
                    // same as "includePath"
                ]
            }
        }
    ],
    "version": 4
}
```

Do not include compiler paths in "includePath" and "browse" let vscode do it for you with  "compilerPath".
Note: Some includes are still red underlined, like i.e. "mbed.h" ....
If you are working on linux and windows just make a second configuration with a different name like "MBED Linux" and change the compiler path, the rest should be the same.

BTW: for debugging the "Cortex-Debug" extension is highly recommended! It works better, faster and provides you more debugging information.

Hope this helps

DBS06, 2018-10-22:
Mhm my comment disappeared...

But yes I did! It was a fight against dead dragons which are spitting blue fire 😄 😠 

1. Be sour your vscode and extensions are up to date.
2. Backup you "c_cpp_properties.json" file.
3. Clean everything in your "c_cpp_properties.json" file.
4. Insert and fill out the following in your "c_cpp_properties.json" file:
```c++
{
    "configurations": [
        {
            "name": "MBED",
            "intelliSenseMode": "clang-x64",
            "compilerPath": "PATH_TO_YOUR_COMPILER" // i.e. C:/.../GNU Tools ARM Embedded/6 2017-q2-update/bin/arm-none-eabi-g++.exe",
            "limitSymbolsToIncludedHeaders": true,
            "cStandard": "c99",
            "cppStandard": "c++11",
            "defines": [
                // put the exporter generated definitions here and your own definitions
                "MBED_BUILD_TIMESTAMP=1520864281.56",
                "MBED_CONF_RTOS_PRESENT=1",
                "MBED_TICKLESS",
                
            ],
            "includePath": [
                // put the exporter generated include paths here and your own paths
                "${workspaceRoot}/*",
                "${workspaceRoot}/mbed-os/*",
                // ...
            ],
            "browse": {
                "path": [
                    // same as "includePath"
                ]
            }
        }
    ],
    "version": 4
}
```

**Do not include compiler paths in "includePath" or "browse"! Otherwise you will enter the hell of GCC-Paths! Let vscode do it for you with setting the "compilerPath"**

But some includes are still red underlined i.e. "mbed.h" and I was never able to fix that. Try to solve it or just ignore it, I think this are following errors, but at least intellisense works and "CTRL-Left-Click" on the include file works.

BTW: for Compiling use the "Cortex-Debug" extensions, its faster, more stable, provides more debug infos and is better 😄 You are even able to debug with Nordic-Softdevices!

Hope this helps!

Alex-EEE, 2018-10-22:
thansk @DBS06 I'll give this a shot!

JojoS62, 2018-10-23:
I'm playing also with the settings for the Intellisense. The suggestion from @DBS06 works fine, and additionally:

- there is a difference between "C_Cpp.intelliSenseEngine": "Tag Parser" and "Default" in settings.json. With Tag parser the result shows no errors in parsing, with Default I always get unknown 'DigitalOut' class (AnalogOut is known, so the drivers path is found). When I remove one of the TARGET_STM includes, the DigitalOut is known but mbed.h is incomplete. Strange, but looks like a parser problem.

- the configuration for windows can use also '/' for path, the triple configuration that the exporter creates is not necessary

- the varialbe name ${workspaceRoot} is depricated, ${workspaceFolder} should be used now.

- conditional compiling is not marked by using current defines, or are other magic settings neccessary for this?

@arekzaluski Will mbed Studio be a VSCode Extension? There was a call for mbed Studio alpha testers but then there was no feedback on this. I have asked in #8415 about VSCode.

DBS06, 2018-10-24:
@JojoS62 Your remarks are correct.
I also have the Intellisense Fallback Engine enabled, which is the Tag-Parser therefore I added also the "browse" paths.

```
    "C_Cpp.intelliSenseEngine": "Default",
    "C_Cpp.intelliSenseEngineFallback": "Enabled",
```

BTW: mbed Studio as a VSCode Extension would be really nice!!!

janjongboom, 2018-10-29:
@JojoS62 @DBS06 No, it will not be based on Visual Studio Code, but on something similar. There was an alpha test at Mbed Connect a few weeks back, hope to open the general alpha at some point too, but not my call ;-)

linlingao, 2019-05-30:
It appears the question has been answered. I'll close this issue.

closed 2019-05-30 by linlingao