ARMmbed/mbed-os #7866: [export][cmake-gcc-arm] Online compiler - mbed_config.h is added to ADD_EXECUTABLE incorrectly

yennster, 2018-08-22:
### Description

### Issue request type

[ ] Question  
[ ] Enhancement  
[ ] Bug

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-381*

theotherjimmy, 2018-10-05:
This is no longer reproduce-able with the online compiler because I fixed it when we migrated to the 5.10.0 tools.

closed 2018-10-05 by theotherjimmy

lemmerelassal, 2019-01-29:
> This is no longer reproduce-able with the online compiler because I fixed it when we migrated to the 5.10.0 tools.

I do still get it. I fixed it, but now get e=87 despite adding ``LOCAL_SHORT_COMMANDS := true`` to ``Makefile`` .

Joebeazelman, 2024-01-31:
I am also getting this error. How do I go about fixing it? Is there a workaround? I couldn't find any Mbed command to regenerate the file.