ARMmbed/mbed-os #7867: Failing flashiap tests on DISCO_L475VG_IOT01A

aashishc1988, 2018-08-22:
### Description
- Type: Bug
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
DISCO_L475VG_IOT01A

**Toolchain:**
GCC_ARM

**Tests we are failing:**
tests-mbed_drivers-flashiap

**Steps to reproduce**

mbed  test -m DISCO_L475VG_IOT01A -t GCC_ARM -n tests-mbed_drivers-flashiap -DMBED_HEAP_STATS_ENABLED=1 -DMBED_STACK_STATS_ENABLED=1 -DMBED_TRAP_ERRORS_ENABLED=1 -DMBED_ALL_STATS_ENABLED -v

**log snippet**
```
> 
00:36:46.718 	calling mbedhtrun: mbedhtrun -m DISCO_L475VG_IOT01A -p DUMMY:9600 -f "BUILD/tests/DISCO_L475VG_IOT01A/GCC_ARM/TESTS/mbed_drivers/flashiap/flashiap.bin" -e "TESTS/host_tests" --grm raas_client:goku.austin.arm.com:8000 -C 4 --sync 10 -P 120
00:36:46.718 mbedgt: mbed-host-test-runner: started
00:36:47.637 [1534961277.58][root]Generating grammar tables from /usr/lib/python2.7/lib2to3/Grammar.txt
00:36:47.697 [1534961277.64][root]Generating grammar tables from /usr/lib/python2.7/lib2to3/PatternGrammar.txt
00:36:47.727 [1534961277.67][HTST][INF] host test executor ver. 1.4.0
00:36:47.727 [1534961277.67][HTST][INF] copy image onto target... SKIPPED!
00:36:47.728 [1534961277.67][HTST][INF] starting host test process...
00:36:47.766 [1534961277.71][CONN][INF] starting connection process...
00:36:47.766 [1534961277.71][CONN][INF] notify event queue about extra 120 sec timeout for serial port pooling
00:36:47.766 [1534961277.71][CONN][INF] initializing global resource mgr listener...
00:36:47.783 [1534961277.73][HTST][INF] setting timeout to: 120 sec
00:36:47.873 [1534961277.82][GLRM][INF] remote resources initialization: remote(host=goku.austin.arm.com, port=8000)
00:36:47.874 [1534961277.82][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:36:47.884 [1534961277.83][urllib3.connectionpool]http://goku.austin.arm.com:8000 "POST /login HTTP/1.1" 200 213
00:36:47.887 [1534961277.83][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:36:47.890 [1534961277.83][urllib3.connectionpool]http://goku.austin.arm.com:8000 "GET /socket.io/?token=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJ1c2VybmFtZSI6InVzZXIiLCJncm91cCI6InVzZXIiLCJpYXQiOjE1MzQ5NjEyNzcsImV4cCI6MTUzNTU2NjA3N30.8QnknQdsau-q88KX7wmRBUfkmaFbPEfwRpMkvo7kluQ&EIO=3&transport=polling&t=1534961277829-0 HTTP/1.1" 200 101
00:36:47.936 [1534961277.88][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:36:47.938 [1534961277.88][urllib3.connectionpool]http://goku.austin.arm.com:8000 "GET /this/resource HTTP/1.1" 200 3660
00:36:47.939 [1534961277.88][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:36:47.950 [1534961277.89][urllib3.connectionpool]http://goku.austin.arm.com:8000 "GET /resource HTTP/1.1" 200 60033
00:36:47.950 [1534961277.89][GLRM][INF] remote resources count: 40
00:36:47.959 [1534961277.90][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:36:47.963 [1534961277.90][urllib3.connectionpool]http://goku.austin.arm.com:8000 "GET /resource/07640221144368082B07FE51 HTTP/1.1" 200 1626
00:36:47.963 [1534961277.91][GLRM][INF] remote resources flashing with 'BUILD/tests/DISCO_L475VG_IOT01A/GCC_ARM/TESTS/mbed_drivers/flashiap/flashiap.bin'...
00:36:47.963 [1534961277.91][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:36:47.967 [1534961277.91][urllib3.connectionpool]http://goku.austin.arm.com:8000 "POST /file HTTP/1.1" 200 244
00:37:00.060 [1534961290.00][GLRM][INF] opening connection to platform at baudrate='9600'
00:37:00.060 [1534961290.00][urllib3.connectionpool]Starting new HTTP connection (1): goku.austin.arm.com:8000
00:37:00.251 [1534961290.19][urllib3.connectionpool]http://goku.austin.arm.com:8000 "PUT /resource/07640221144368082B07FE51/connect HTTP/1.1" 200 15
00:37:00.251 [1534961290.19][GLRM][INF] remote resources reset...
00:37:02.572 [1534961292.51][GLRM][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
00:37:02.572 [1534961292.51][CONN][INF] sending up to 10 __sync packets (specified with --sync=10)
00:37:02.572 [1534961292.51][CONN][INF] sending preamble '5550bb3e-1f57-481b-a077-e24267829bde'
00:37:02.573 [1534961292.52][GLRM][TXD] {{__sync;5550bb3e-1f57-481b-a077-e24267829bde}}
00:37:03.657 [1534961293.60][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
00:37:03.657 [1534961293.60][CONN][INF] found SYNC in stream: {{__sync;5550bb3e-1f57-481b-a077-e24267829bde}} it is #0 sent, queued...
00:37:03.657 [1534961293.60][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
00:37:03.657 [1534961293.60][HTST][INF] sync KV found, uuid=5550bb3e-1f57-481b-a077-e24267829bde, timestamp=1534961293.600190
00:37:03.657 [1534961293.60][HTST][INF] DUT greentea-client version: 1.3.0
00:37:03.883 [1534961293.83][CONN][INF] found KV pair in stream: {{__timeout;120}}, queued...
00:37:03.883 [1534961293.83][HTST][INF] setting timeout to: 120 sec
00:37:03.883 [1534961293.83][HTST][INF] host test class: '<class 'mbed_host_tests.host_tests.default_auto.DefaultAuto'>'
00:37:03.883 [1534961293.83][HTST][INF] host test setup() call...
00:37:03.883 [1534961293.83][HTST][INF] CALLBACKs updated
00:37:03.883 [1534961293.83][HTST][INF] host test detected: default_auto
00:37:03.883 [1534961293.83][CONN][INF] found KV pair in stream: {{__host_test_name;default_auto}}, queued...
00:37:03.884 [1534961293.83][CONN][INF] found KV pair in stream: {{__testcase_count;5}}, queued...
[... 376 lines cut ...]
00:43:48.373 [1534961698.31][HTST][INF] CONN exited with code: 0
00:43:48.373 [1534961698.31][HTST][INF] No events in queue
00:43:48.373 [1534961698.31][HTST][INF] stopped consuming events
00:43:48.373 [1534961698.31][HTST][INF] host test result(): None
00:43:48.373 [1534961698.31][HTST][WRN] missing __exit event from DUT
00:43:48.373 [1534961698.31][HTST][WRN] missing __exit_event_queue event from host test
00:43:48.373 [1534961698.31][HTST][ERR] missing __exit_event_queue event from host test and no result from host test, timeout...
00:43:48.373 [1534961698.31][HTST][INF] calling blocking teardown()
00:43:48.373 [1534961698.31][HTST][INF] teardown() finished
00:43:48.373 [1534961698.31][HTST][INF] {{result;timeout}}
00:43:48.396 mbedgt: retry mbedhtrun 3/3
00:43:48.396 mbedgt: ['mbedhtrun', '-m', 'DISCO_L475VG_IOT01A', '-p', 'DUMMY:9600', '-f', u'"BUILD/tests/DISCO_L475VG_IOT01A/GCC_ARM/TESTS/mbed_drivers/flashiap/flashiap.bin"', '-e', u'"TESTS/host_tests"', '--grm', 'raas_client:goku.austin.arm.com:8000', '-C', '4', '--sync', '10', '-P', '120'] failed after 3 count
00:43:48.410 mbedgt: checking for GCOV data...
00:43:48.411 mbedgt: mbed-host-test-runner: stopped and returned 'TIMEOUT'
00:43:48.411 mbedgt: test on hardware with target id: DUMMY
00:43:48.411 mbedgt: test suite 'tests-mbed_drivers-flashiap' ..................................................... TIMEOUT in 140.06 sec
00:43:48.411 	test case: 'FlashIAP - init' ................................................................. OK in 0.00 sec
00:43:48.411 	test case: 'FlashIAP - program' .............................................................. OK in 0.10 sec
00:43:48.411 	test case: 'FlashIAP - program across sectors' ............................................... OK in 0.10 sec
00:43:48.411 	test case: 'FlashIAP - program errors' ....................................................... OK in 0.21 sec
00:43:48.411 	test case: 'FlashIAP - timing' ............................................................... ERROR in 0.00 sec
```
    [ ] Question
    [ ] Enhancement 
    [X ] Bug

0xc0170, 2018-08-23:
cc @ARMmbed/team-st-mcd

adustm, 2018-08-23:
Hello, 
Could you precise your issue please ? 
`mbed test -m DISCO_L475VG_IOT01A -t GCC_ARM -n tests-mbed_drivers-flashiap -v`
gives correct results
```
mbedgt: test suite 'tests-mbed_drivers-flashiap' ..................................................... OK in 17.08 sec
        test case: 'FlashIAP - init' ................................................................. OK in 0.05 sec
        test case: 'FlashIAP - program' .............................................................. OK in 0.17 sec
        test case: 'FlashIAP - program across sectors' ............................................... OK in 0.11 sec
        test case: 'FlashIAP - program errors' ....................................................... OK in 0.06 sec
        test case: 'FlashIAP - timing' ............................................................... OK in 0.78 sec
mbedgt: test case summary: 5 passes, 0 failures
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.0359470136
mbedgt: test suite report:
+-----------------------------+---------------------+-----------------------------+--------+--------------------+-------------+
| target                      | platform_name       | test suite                  | result | elapsed_time (sec) | copy_method |
+-----------------------------+---------------------+-----------------------------+--------+--------------------+-------------+
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | tests-mbed_drivers-flashiap | OK     | 17.08              | default     |
+-----------------------------+---------------------+-----------------------------+--------+--------------------+-------------+
```

FYI my setup is the following
mbed-os  sha1 = `603c4f9`
GCC version is `6 2017-q2-update`
```
mbed-cli                      1.7.5
mbed-connector-api            1.0.4
mbed-greentea                 1.4.0
mbed-host-tests               1.4.1
mbed-ls                       1.5.1
mbed-test-wrapper             1.0.0
```

I have tried your command line 
`mbed test -m DISCO_L475VG_IOT01A -t GCC_ARM -n tests-mbed_drivers-flashiap -DMBED_HEAP_STATS_ENABLED=1 -DMBED_STACK_STATS_ENABLED=1 -DMBED_TRAP_ERRORS_ENABLED=1 -DMBED_ALL_STATS_ENABLED -v`
but it does not work 
```
Build successes:
  * DISCO_L475VG_IOT01A::GCC_ARM::MBED-BUILD
  * DISCO_L475VG_IOT01A::GCC_ARM::TESTS-MBED_DRIVERS-FLASHIAP
[mbed] Exec "mbedgt --test-spec .\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json -n tests-mbed_drivers-flashiap -V -DMBED_HEAP_STATS_ENABLED=1 -DMBED_STACK_STATS_ENABLED=1
-DMBED_TRAP_ERRORS_ENABLED=1 -DMBED_ALL_STATS_ENABLED" in "C:\MCD8\mbed-os"
Usage: mbedgt [options]

mbedgt: error: no such option: -D
[mbed] ERROR: "mbedgt" returned error.
       Code: 2
       Path: "C:\MCD8\mbed-os"
       Command: "mbedgt --test-spec .\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json -n tests-mbed_drivers-flashiap -V -DMBED_HEAP_STATS_ENABLED=1 -DMBED_STACK_STATS_ENABLE
D=1 -DMBED_TRAP_ERRORS_ENABLED=1 -DMBED_ALL_STATS_ENABLED"
       Tip: You could retry the last command with "-v" flag for verbose output
```

I have created a mbed_app.json containing
```
{
    "macros": [
        "MBED_HEAP_STATS_ENABLED=1",
        "MBED_STACK_STATS_ENABLED=1",
        "MBED_TRAP_ERRORS_ENABLED=1",
	"MBED_ALL_STATS_ENABLED"
    ]
}
```
and it gives the same error as the one you described
Adding statistics is clearly impacting the results...

aashishc1988, 2018-08-23:
@adustm That's correct. My findings were along the similar lines as to adding just the flag to collect all the stats (-DMBED_ALL_STATS_ENABLED) causes the issue. I am still investigating as to why that is. My understanding is that that flag collects all memory information and might be doing some allocation/deallocation and may be much more, so at this point I am not completely sure if this is an issue with the target or something we are doing while collecting the stats.

aashishc1988, 2018-08-23:
The issue here is memory. When all the stats are enabled, we dont have enough memory on this part to run this test. It fails to allocate 512 bytes. When trying to run flashiap_timing_test, we need to allocate bunch of memory, we could add a change in that skips the test if we cannot get enough memory for it, something along the lines of this PR: https://github.com/ARMmbed/mbed-os/pull/7696

MateuszMaz, 2018-08-24:
Tested few other devices with tests-mbed_drivers-flashiap and those flags here is output:

device | flash | ram | result
-- | -- | -- | --
stm32f429zi (nucleo) | 2048 KB | 256 +4 KB | ok
nucleo f070r0b | 128 KB | 16 KB | ok
nucleo -l476rg | 1 MB | 128 KB | error (timeout)
k64f | 1 MB | 256 KB | ok
DISCO-F769NI | 512 Mbit | 128 Mbit | ok
NRF52_DK (nrf52832) | 512 KB | 64 kB | ok

looks like only targets stm32-47x have this issue

adustm, 2018-08-24:
Hi @MateuszMaz 
Thanks for this test status.

Hi @0xc0170 
This is the assumption I was having before reading this report also.
The test is using 76553 flash and 11844 ram, and the device has 1Mb flash + 128k ram. It is supposed to be able to use statistics
I am trying to print memory values with the following function, but it does not work
```
void print_heap(const char* info) {
    printf(info);
    mbed_stats_heap_get(&stats);
    printf("  current_size =  0x%8X\r\n", stats.current_size);
    printf("      max_size =  0x%8X\r\n", stats.max_size);
    printf("    total_size =  0x%8X\r\n", stats.total_size);
    printf(" reserved_size =  0x%8X\r\n", stats.reserved_size);
    printf("     alloc_cnt =  0x%8X\r\n", stats.alloc_cnt);
    printf("alloc_fail_cnt =  0x%8X\r\n", stats.alloc_fail_cnt);
}

```
Let's keep searching

MateuszMaz, 2018-08-24:
To be a little more precise  problem occurs for flag -DMBED_HEAP_STATS_ENABLED=1
(-DMBED_ALL_STATS_ENABLED=1 just sets  ,sys, stack,cpu,heap,thread stats enable)

adustm, 2018-08-28:
Hello,

Those devices (STM32L476 / 475 / 486 ) have 2 non contigous SRAMs.
Heap is located in SRAM1 (from 0x20000000) and stack is located in SRAM2 ( from 0x10000000).
Could someone check if the heap statistic code supports this split ?

[Crash analysis document](https://os.mbed.com/docs/v5.9/tutorials/analyzing-mbed-os-crash-dump.html#analyzing-crash-dump) explains that 

> You can also use the stack size (Stack Size), stack top (Mem) and current stack pointer (SP) value to determine if there is thread stack overflow. For example, if the SP value is smaller than the Mem value, it indicates stack overflow for that thread.

The provided dump
`00:39:26.927 [1534961436.87][CONN][RXD] Current Thread: Id: 0x20001A0C Entry: 0x8004629 StackSize: 0x1000 StackMem: 0x20001A58 SP: 0x10007F70`
has sp pointer < StackMem... Is it an issue ?

Kind regards

adustm, 2018-08-30:
hello again,
I would like to share some findings. We don't understand the impact of our modifications on the behavior of the test. Maybe some of you will.

Crash log parser says:
```
/c/mbed-os  (master)$ python tools/debug_tools/crash_log_parser/crash_log_parser.py hardfault.txt \
BUILD/tests/DISCO_L475VG_IOT01A/GCC_ARM/TESTS/mbed_drivers/flashiap/flashiap.elf \
BUILD/tests/DISCO_L475VG_IOT01A/GCC_ARM/TESTS/mbed_drivers/flashiap/flashiap.map.old

Crash Info:
        Crash location = _free_r [0x08009C2A] (based on PC value)
        Caller location = osMutexAcquire [0x08005B03] (based on LR value)
        Stack Pointer at the time of crash = [200028E0]
        Target and Fault Info:
                Processor Arch: ARM-V7M or above
                Processor Variant: C24
                Forced exception, a fault with configurable priority has been escalated to HardFault
                A precise data access error has occurred. Faulting address: 1FFD1F5E
```
This mean that the error occurs here in mbed_alloc_wrappers.cpp
```
extern "C" void free_wrapper(struct _reent *r, void *ptr, void *caller)
{
#ifdef MBED_MEM_TRACING_ENABLED
    mbed_mem_trace_lock();
#endif
#ifdef MBED_HEAP_STATS_ENABLED
    malloc_stats_mutex->lock();                  // * * * * * ERROR OCCURS HERE
    alloc_info_t *alloc_info = NULL;
    if (ptr != NULL) {
        alloc_info = ((alloc_info_t *)ptr) - 1;
        heap_stats.current_size -= alloc_info->size;
        heap_stats.alloc_cnt -= 1;
    }
```

We've been able to find 2 ways of fixing the problem (no explanation so far...)
1/ mbed_alloc_wrappers.cpp
```
diff --git a/platform/mbed_alloc_wrappers.cpp b/platform/mbed_alloc_wrappers.cpp
index 87fcd95..cf9965f 100644
--- a/platform/mbed_alloc_wrappers.cpp
+++ b/platform/mbed_alloc_wrappers.cpp
@@ -43,7 +43,7 @@ are active, the second one (MBED_MEM_TRACING_ENABLED) will trace the first one's
 /* Size must be a multiple of 8 to keep alignment */
 typedef struct {
     uint32_t size;
-    uint32_t pad;
+    //uint32_t pad;
 } alloc_info_t;

```
=> test is ok
2/ l4_retarget.c
```
diff --git a/targets/TARGET_STM/TARGET_STM32L4/l4_retarget.c b/targets/TARGET_STM/TARGET_STM32L4/l4_retarget.c
index c80d087..b14cff1 100644
--- a/targets/TARGET_STM/TARGET_STM32L4/l4_retarget.c
+++ b/targets/TARGET_STM/TARGET_STM32L4/l4_retarget.c
@@ -38,7 +38,7 @@
 extern uint32_t __mbed_sbrk_start;
 extern uint32_t __mbed_krbs_start;

-#define STM32L4_HEAP_ALIGN               32
+#define STM32L4_HEAP_ALIGN               4
 #define STM32L4_ALIGN_UP(X, ALIGN)       (((X) + (ALIGN) - 1) & ~((ALIGN) - 1))
```
=> test is ok

Both modifications fixes the test. 

Please note that adding a printf in main.cpp / flashiap_init_test function makes the program fail during flashiap_init function instead of flashiap_timing_test function

On the other hand, we 've checked heap settings
```
__heap_size = 0x   151f8  
__mbed_sbrk_start  = 0x20002e08
__mbed_krbs_start  = 0x20018000
__HeapLimit = 0x20018000
```
There is 84k of heap... should be sufficient...

I hope this will help you finding the issue.
Kind regards

MateuszMaz, 2018-08-30:
I found that this problem is not related neither to greentea nor FlashIAP, and thanks to @maciejbocianski  help I also know that it occurs only for GCC_ARM. It is indeed related to memory allocation.

Just make a simple program, and compile it with -DMBED_HEAP_STATS_ENABLED=1  -t GCC_ARM

```
#include "mbed.h"

int main(){

int FIRST_SIZE = 8;
int SECOND_SIZE = 2000;

uint8_t *table1 = new uint8_t[FIRST_SIZE];
uint8_t *table2 = new uint8_t[SECOND_SIZE];

return 0;
}
```
this one will result in something similar:

> ++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
Location: 0x80014B7
Error Value: 0x8005A4A
Current Thread: Id: 0x200014BC Entry: 0x800150D StackSize: 0x1000 StackMem: 0x20001508 SP: 0x10007F70 
-- MbedOS Error Info --

LMESTM, 2018-08-30:
HI all
Can someone explain why alloc_info_t _NEEDS_ to be aligned to 8 ? 

```
diff --git a/platform/mbed_alloc_wrappers.cpp b/platform/mbed_alloc_wrappers.cpp
index 87fcd95..cf9965f 100644
--- a/platform/mbed_alloc_wrappers.cpp
+++ b/platform/mbed_alloc_wrappers.cpp
@@ -43,7 +43,7 @@ are active, the second one (MBED_MEM_TRACING_ENABLED) will trace the first one's
 /* Size must be a multiple of 8 to keep alignment */
 typedef struct {
     uint32_t size;
-    uint32_t pad;
+    //uint32_t pad;
 } alloc_info_t;
```

LMESTM, 2018-08-31:
@bogdanm @c1728p9 @heky
I think that you 3 were at the origin of the code for handling memory tracing. We have an issue here and a question about why alloc_info_t must be aligned to 8 - could you maybe help out ? 
thanks
Laurent

c1728p9, 2018-08-31:
Hi @LMESTM this was done in an abundance of caution. The default implementation only returns 8 byte aligned data and some things, such as stacks require 8 byte aligned memory. To ensure there were no regressions the 8 byte alignment was kept.

LMESTM, 2018-08-31:
Thanks for quick feedback @c1728p9.  In our particular case, removing the padding, therefore removing this extra caution step and keeping structure with 4 bytes only, somehow solves our issue ... 
and I have no why  ... any idea ?  could it be that the pad in structure is sometimes optimized out as this is never used nor referenced ?

c1728p9, 2018-08-31:
I created #7950 to fix this. Please test and confirm it fixes the problem.

LMESTM, 2018-09-03:
@c1728p9 based on recent observations, we were heading in a similar direction and planned to remove the alignment but we can not explain why the alignment was causing the failure. you suggestion is : 

> the alignment done in __wrap__sbrk won't match what newlib expected, which is likely the cause of the crash

so ok we'll consider sbrk  needs to allocate exactly what is requested and no more.  Fine with me.

aashishc1988, 2018-09-05:
@c1728p9 I have verified that https://github.com/ARMmbed/mbed-os/pull/7950 fixes the problem.

closed 2018-09-18 by 0xc0170