ARMmbed/mbed-os #7885: numerous compiler warnings

bmcdonnell-ionx, 2018-08-24:
### Description

There are a number of compiler warnings. Below is a sample excerpt of when I compiled for 2 targets, with an empty `main()`. Please fix these. I suggest you also then check the reset of the targets and fix any remaining.

Targets: Probably all
Toolchain: Tested only GCC_ARM
Mbed CLI
`f8b140f8d (HEAD, tag: mbed-os-5.9.5, tag: latest, origin/mbed-os-5.9) Merge pull request #7753 from ARMmbed/release-candidate`

```
me@pc MINGW64 /c/dev_temp/mbed-warnings (master)
$ mbed compile -c
Building project mbed-warnings (LPC4088, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
Scan: FEATURE_LWIP

<snip>

Compile [ 25.1%]: UBLOX_PPP.cpp
[Warning] UBLOX_PPP.h@37,7: attribute ignored in declaration of 'class mbed::UBLOX_LISA_U' [-Wattributes]
[Warning] UBLOX_PPP_CellularNetwork.h@39,7: attribute ignored in declaration of 'class mbed::UBLOX_LISA_U_CellularNetwork' [-Wattributes]
[Warning] UBLOX_PPP_CellularPower.h@38,7: attribute ignored in declaration of 'class mbed::UBLOX_LISA_U_CellularPower' [-Wattributes]
Compile [ 25.3%]: UBLOX_PPP_CellularNetwork.cpp
[Warning] UBLOX_PPP_CellularNetwork.h@39,7: attribute ignored in declaration of 'class mbed::UBLOX_LISA_U_CellularNetwork' [-Wattributes]
Compile [ 25.4%]: UBLOX_PPP_CellularPower.cpp
[Warning] UBLOX_PPP_CellularPower.h@38,7: attribute ignored in declaration of 'class mbed::UBLOX_LISA_U_CellularPower' [-Wattributes]

<snip>

Compile [ 49.8%]: coap_connection_handler.c
[Warning] coap_security_handler.h@132,45: statement with no effect [-Wunused-value]
[Warning] coap_security_handler.h@129,111: statement with no effect [-Wunused-value]
[Warning] coap_security_handler.h@132,45: statement with no effect [-Wunused-value]
[Warning] coap_connection_handler.c@531,12: 'timer_status' defined but not used [-Wunused-function]
[Warning] coap_connection_handler.c@508,13: 'start_timer' defined but not used [-Wunused-function]
[Warning] coap_connection_handler.c@453,12: 'secure_session_recvfrom' defined but not used [-Wunused-function]
[Warning] coap_connection_handler.c@421,12: 'secure_session_sendto' defined but not used [-Wunused-function]

<snip>

Compile [ 59.5%]: thread_discovery.c
[Warning] thread_discovery.c@562,50: statement with no effect [-Wunused-value]

<snip>

Compile [ 86.7%]: mbed_ticker_api.c
[Warning] mbed_ticker_api.c@47,22: comparison between signed and unsigned integer expressions [-Wsign-compare]

<snip>

Compile [ 93.3%]: mbed_rtx_handlers.c
[Warning] mbed_rtx_handlers.c@44,18: unused variable 'tid' [-Wunused-variable]

<snip>

Compile [ 96.7%]: rtx_thread.c
[Warning] rtx_thread.c@188,14: 'osRtxThreadListRoot' defined but not used [-Wunused-function]

<snip>

Compile [ 97.9%]: serial_api.c
[Warning] serial_api.c@236,43: 'parity_select' may be used uninitialized in this function [-Wmaybe-uninitialized]
[Warning] serial_api.c@235,43: 'parity_enable' may be used uninitialized in this function [-Wmaybe-uninitialized]

<snip>

Compile [100.0%]: test_env.cpp
Link: mbed-warnings
Elf2Bin: mbed-warnings
+------------------+-------+-------+------+
| Module           | .text | .data | .bss |
+------------------+-------+-------+------+
| [fill]           |   105 |     0 |   13 |
| [lib]\c.a        | 17243 |  2472 |   89 |
| [lib]\gcc.a      |  3168 |     0 |    0 |
| [lib]\misc       |   252 |    12 |   28 |
| main.o           |     8 |     0 |    0 |
| mbed-os\drivers  |    56 |     0 |    0 |
| mbed-os\hal      |  1757 |     4 |   68 |
| mbed-os\platform |  2685 |   260 |  133 |
| mbed-os\rtos     |  8512 |   168 | 6073 |
| mbed-os\targets  |  3096 |    16 |   16 |
| Subtotals        | 36882 |  2932 | 6420 |
+------------------+-------+-------+------+
Total Static RAM memory (data + bss): 9352 bytes
Total Flash memory (text + data): 39814 bytes

Image: .\BUILD\LPC4088\GCC_ARM\mbed-warnings.bin
```

```
me@pc MINGW64 /c/dev_temp/mbed-warnings (master)
$ mbed compile -t GCC_ARM -m NUCLEO_F410RB
Building project mbed-warnings (NUCLEO_F410RB, GCC_ARM)
Scan: .
Scan: mbed
Scan: env

<snip>

Compile [  9.9%]: UBLOX_PPP.cpp
[Warning] UBLOX_PPP.h@37,7: attribute ignored in declaration of 'class mbed::UBLOX_LISA_U' [-Wattributes]
[Warning] UBLOX_PPP_CellularNetwork.h@39,7: attribute ignored in declaration of 'class mbed::UBLOX_LISA_U_CellularNetwork' [-Wattributes]
[Warning] UBLOX_PPP_CellularPower.h@38,7: attribute ignored in declaration of 'class mbed::UBLOX_LISA_U_CellularPower' [-Wattributes]
Compile [ 10.1%]: UBLOX_PPP_CellularNetwork.cpp
[Warning] UBLOX_PPP_CellularNetwork.h@39,7: attribute ignored in declaration of 'class mbed::UBLOX_LISA_U_CellularNetwork' [-Wattributes]
Compile [ 10.3%]: UBLOX_PPP_CellularPower.cpp
[Warning] UBLOX_PPP_CellularPower.h@38,7: attribute ignored in declaration of 'class mbed::UBLOX_LISA_U_CellularPower' [-Wattributes]

<snip>

Compile [ 44.1%]: thread_discovery.c
[Warning] thread_discovery.c@562,50: statement with no effect [-Wunused-value]

<snip>

Compile [ 70.2%]: mbed_ticker_api.c
[Warning] mbed_ticker_api.c@47,22: comparison between signed and unsigned integer expressions [-Wsign-compare]

<snip>

Compile [ 76.6%]: mbed_rtx_handlers.c
[Warning] mbed_rtx_handlers.c@44,18: unused variable 'tid' [-Wunused-variable]

<snip>

Compile [ 79.8%]: rtx_thread.c
[Warning] rtx_thread.c@188,14: 'osRtxThreadListRoot' defined but not used [-Wunused-function]

<snip>

Compile [100.0%]: test_env.cpp
Link: mbed-warnings
Elf2Bin: mbed-warnings
+------------------+-------+-------+------+
| Module           | .text | .data | .bss |
+------------------+-------+-------+------+
| [fill]           |   121 |     4 |   17 |
| [lib]\c.a        | 24981 |  2472 |   89 |
| [lib]\gcc.a      |  3168 |     0 |    0 |
| [lib]\misc       |   252 |    16 |   28 |
| main.o           |     8 |     0 |    0 |
| mbed-os\drivers  |   192 |     4 |  100 |
| mbed-os\hal      |  1653 |     4 |   68 |
| mbed-os\platform |  2736 |   260 |  133 |
| mbed-os\rtos     |  8510 |   168 | 6073 |
| mbed-os\targets  |  6469 |     4 |  348 |
| Subtotals        | 48090 |  2932 | 6856 |
+------------------+-------+-------+------+
Total Static RAM memory (data + bss): 9788 bytes
Total Flash memory (text + data): 51022 bytes

Image: .\BUILD\NUCLEO_F410RB\GCC_ARM\mbed-warnings.bin
```

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-08-27:
@ARMmbed/team-ublox @ARMmbed/mbed-os-wan Please address warnings in cellular/lorawan

@ARMmbed/mbed-os-core Please address warnings in the platform/drivers/rtos

deepikabhavnani, 2018-08-27:
```
Compile [ 97.9%]: serial_api.c
[Warning] serial_api.c@236,43: 'parity_select' may be used uninitialized in this function [-Wmaybe-uninitialized]
[Warning] serial_api.c@235,43: 'parity_enable' may be used uninitialized in this function [-Wmaybe-uninitialized]
```

Above warning is from target specific code, and is present for all NXP devices. Parity is not set for `default` case.

https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NXP/TARGET_LPC408X/TARGET_LPC4088/serial_api.c#L229

kegilbert, 2018-08-27:
I was looking over the osRtxthreadListRoot issue, looks like a problem with the `EvrRtx*` calls in the case structure here: https://github.com/ARMmbed/mbed-os/blob/master/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_thread.c#L344-L369

The call defaults to a void macro in default conditions removing the call to `osRtxThreadListRoot`.

closed 2018-10-02 by adbridge