ARMmbed/mbed-os #7886: improve process to prevent warnings

bmcdonnell-ionx, 2018-08-24:
### Description

The number of compiler warnings has grown over time. Please consider implementing one or both of the following:

- Improve documentation to encourage contributors to check for and resolve any compiler warnings before merge.
- Add CI tests to enforce no compiler warnings allowed. (Preferred, I think.)

Related: #7885, #6305

### Issue request type

[ ] Question  
[X] Enhancement  
[ ] Bug

mattbrown015, 2018-08-24:
Also related, [mbed should be compilable with "-Werror" #7312](https://github.com/ARMmbed/mbed-os/issues/7312)

0xc0170, 2018-08-27:
> Improve documentation to encourage contributors to check for and resolve any compiler warnings before merge.

This should be covered by docs, we will review (cc @AnotherButler)

> Add CI tests to enforce no compiler warnings allowed. (Preferred, I think.)

This is definitely needed. We already discussed this. The referenced PR above by @mattbrown015 . There is this comment: https://github.com/ARMmbed/mbed-os/issues/7312#issuecomment-400980507.   believe we should put something in place soon and keep it under control and fix one by one.

cc @OPpuolitaival

bmcdonnell-ionx, 2018-08-27:
I don't quite understand https://github.com/ARMmbed/mbed-os/issues/7312#issuecomment-400980507. What is the part about +15 warnings?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/MBEDOSTEST-165*

closed 2019-10-31 by ciarmcom