ARMmbed/mbed-os #7887: Comments in mbed_app.json

jflynn129, 2018-08-24:
### Description

QUESTION: Is there a way to insert comments into an mbed_app.json file. I would basically like to 
comment out some lines occasionally but haven't found a way to simply comment out a line, eg. start a line with # or ' or something and it is ignored.

### Issue request type

[X ] Question  
[ ] Enhancement  
[ ] Bug

andrewleech, 2018-08-24:
I too would love this change. 
As much as it's non-standard json vs-code does this, allowing comments in json Configs and it's incredibly useful.

jflynn129, 2018-08-30:
Its an ugly work around but the following will work...

In the mbed_app.json file, put  comments in a separate set of name/value pairs in the config section. I tried to add this to the macros section but it wouldn't work.  example:
 
"config": {
        "config Comments": "Add the to the macros section above to enable memory stats output",
        "heap_stats"           : "MBED_HEAP_STATS_ENABLED",
        "stack_stats"           : "MBED_STACK_STATS_ENABLED",
        "thread_stats"         : "MBED_THREAD_STATS_ENABLED",
        "network-interface": {
            "help": "Options are ETHERNET, CELLULAR_WNC14A2A, CELLULAR_BG96",
            "value": "ETHERNET"
        },

jflynn129, 2018-09-06:
since I found a work-around, I'm closing this. if an alternative way to add comments is ever implemented, please let me know, thanks!

carlosperate, 2019-05-20:
Any chance this could be considered? It'd be great to have comments to explain some of the selected settings.

loverdeg-ep, 2019-05-20:
I want this feature but would probably prefer changing to a notation that supports comments.

andrewleech, 2019-05-20:
Comments, improved readability and ease of writing are the main reasons I switched from json to yaml years ago for all my python config needs, originally with pyyaml then I built this to make it even easier to read and write config files: https://pypi.org/project/structured-config/

The best part is yaml is a superset of json so the yaml / structured-config parsers can read json in and use it the same way, or if desired write it back as clean yaml without a single line of conversion code

0xc0170, 2019-05-21:
> Comments, improved readability and ease of writing are the main reasons I switched from json to yaml years ago for all my python config needs, originally with pyyaml then I built this to make it even easier to read and write config files: https://pypi.org/project/structured-config/

Same here, used yaml where I could.

Config provides "help" to describe settings. Without changing the config format, is there anything else where "help" member is not present or we could improve the experience?

0xc0170, 2019-05-21:
cc @ARMmbed/mbed-os-tools

loverdeg-ep, 2019-05-21:
@0xc0170 Personally I'd like for a well defined standard tools be used for builds, package management, and build configuration.

Mbed OS was ahead of its time in this area but has fallen behind to the projects now available to handle this stuff.

[Buckaroo](https://github.com/LoopPerfect/buckaroo) looks promising among others. CMSIS packs looks like its coming along but I get the feeling it needs significant rework to properly accommodate the new requirements of embedded development. @njlr, @nikhedonia

mark-edgeworth, 2019-05-21:
Ok, happy to reopen this issue; we'll take another look at some point.

nikhedonia, 2019-05-22:
@loverdeg-ep we are happy to help with the adoption of buckaroo and make buckaroo more suitable for mbed-os.

However currently, we are not familiar of mbed-os ecosystem, and your help would be very appreciated:

- How are dependencies currently managed?
- How do you integrate dependencies that live outside the mbed ecosystem?
- How are you currently building the project? 
- What are your toolchain requirements?
- What kind of improvements do you expect from using a different dependency management solution?

mark-edgeworth, 2019-05-22:
Hi, thanks for the offer of help here. This is not something that we are actively working on today but we will revisit the question of dependency and library management at some point. I'm not able to answer the questions you pose but others may well pitch in here...

loverdeg-ep, 2019-06-03:
@nikhedonia I bumped an email exchange we had a while back on this topic.
Our team is already operating outside the normal Mbed OS tooling and will not mind evaluating alternative dependency/package/configuration management options.

@maclobdell @trowbridgec @farrenv  @AGlass0fMilk

0xc0170, 2019-06-04:
@loverdeg-ep Any example of blinky with the tools? What tools do you use and how they interact with targets/components ?

mark-edgeworth, 2019-06-04:
All this discussion is somewhat off-topic from the original report which was to allow comments in the mbed config files. These are in Json, and json does not have a way of expressing comments. It is unlikely that this will change.

closed 2019-06-04 by mark-edgeworth