ARMmbed/mbed-os #7888: Memory tracing suggestion

jflynn129, 2018-08-24:
### Description

I have been using the memory tracing features of mbed-os but found the current implementation a bit overwhelming in terms of the raw data produced.  I wanted to trace only certain portions of the code so I implemented a flag (mem_trace_enable) so that I could enable/disable tracing during execution.  It was implemented by modifing mbed_mem_tracing.cpp in the function calls, i.e.

```
bool mem_trace_enable = true;
.
.
.
    if (mem_trace_cb && mem_trace_enable) {

```
This allows dynamic enabled/disable of memory tracing which greatly assists in isolating issues.

### Issue request type

[ ] Question  
[X ] Enhancement  
[ ] Bug

kegilbert, 2018-08-24:
Hi @jflynn129 ! Thanks for your enhancement request. Tracking every memory operation in the OS can lead to quite a bit of output, good point on having a better way of controlling what is profiled. I can look into adding a proper enable/disable logging function in the tracing class module. In the meantime, you can also try setting the callback to void when you don't want output (take the following that allocates, waits, and deallocates dynamic memory in a loop. Every 5 iterations it disables the logging, then re-enables it 5 loops later. Entire loop posted below with output, but the TL;DR call to make to disable/enable the reporting is:

```c++
mbed_mem_trace_set_callback(0);    // Disable reporting
mbed_mem_trace_set_callback(/*cb*/);  // Enable reporting, use mbed_mem_trace_default_callback in place of `cb` for the default profiling
```

```c++
int main() {
    printf("RUNNING....\r\n");

    mbed_mem_trace_cb_t cb = mbed_mem_trace_default_callback;

    mbed_mem_trace_set_callback(cb);
    uint16_t count = 0;

    while(true) {
        void *p = malloc(50);
        wait(0.5f);
        free(p);
        count += 1;

        if ((count % 5) == 0) {
            if (cb) {
                printf("Setting cb to void...\r\n");
                cb = 0;
            } else {
                printf("Setting cb to default...\r\n")
                cb = mbed_mem_trace_default_callback;
            }

            mbed_mem_trace_set_callback(cb);
        }
    }
```

```
RUNNING....

#m:0x20003ae8;0x17e1-50
#f:0x0;0x17f5-0x20003ae8
#m:0x20003ae8;0x17e1-50
#f:0x0;0x17f5-0x20003ae8
#m:0x20003ae8;0x17e1-50
#f:0x0;0x17f5-0x20003ae8
#m:0x20003ae8;0x17e1-50
#f:0x0;0x17f5-0x20003ae8
#m:0x20003ae8;0x17e1-50
#f:0x0;0x17f5-0x20003ae8
Setting cb to void...

Setting cb to default...

#m:0x20003ae8;0x17e1-50
#f:0x0;0x17f5-0x20003ae8
#m:0x20003ae8;0x17e1-50
#f:0x0;0x17f5-0x20003ae8
#m:0x20003ae8;0x17e1-50
#f:0x0;0x17f5-0x20003ae8
#m:0x20003ae8;0x17e1-50
#f:0x0;0x17f5-0x20003ae8
#m:0x20003ae8;0x17e1-50
#f:0x0;0x17f5-0x20003ae8
Setting cb to void...

Setting cb to default...

#m:0x20003ae8;0x17e1-50
#f:0x0;0x17f5-0x20003ae8
#m:0x20003ae8;0x17e1-50
#f:0x0;0x17f5-0x20003ae8
#m:0x20003ae8;0x17e1-50
#f:0x0;0x17f5-0x20003ae8
```

deepikabhavnani, 2018-08-30:
@jflynn129 - @kegilbert described the mechanism available to enable/disable memory tracing logs if it is good enough we can close the enhancement request.

jflynn129, 2018-08-31:
@deepikabhavnani I have a workable method to trace only the code I was interested in profiling. @kegilbert option is similar and works in a similar manner. I had submitted this as a suggestion only.  Adding kegilbert's description to the documentation or maybe a blog article may be helpful to the community.

closed 2018-08-31 by jflynn129

deepikabhavnani, 2018-08-31:
> Adding kegilbert's description to the documentation or maybe a blog article may be helpful to the community.

👍 https://github.com/ARMmbed/mbed-os-5-docs/issues/697.