ARMmbed/mbed-os #7889: Cannot compile mbed with RTOS disabled

andrewleech, 2018-08-24:
### Description
I regularly need to run mbed without the rtos on smaller targets that do not have enough ram/flash to support the os.

Currently the only way I know of to do this is adding the rtos and events libraries to .mbedignore
```
mbed-os/rtos/*
mbed-os/events/*
```
During testing of PR #7794 I found that this longer works. It was not caused by that PR but something committed to master shortly beforehand.
It appears to be related to some of the new networking `features`

Building a clean blinky project in GCC with the above .mbedignore results in the following error.
```
Compile [  0.3%]: main.cpp
[Fatal Error] Mutex.h@26,23: cmsis_os2.h: No such file or directory
[ERROR] In file included from ./mbed-os/features/netsocket/InternetSocket.h:25:0,
                 from ./mbed-os/features/netsocket/UDPSocket.h:23,
                 from ./mbed-os/features/netsocket/nsapi.h:40,
                 from ./mbed-os/mbed.h:26,
                 from .\main.cpp:1:
./mbed-os/rtos/Mutex.h:26:23: fatal error: cmsis_os2.h: No such file or directory
 #include "cmsis_os2.h"
                       ^
compilation terminated.
```

### Issue request type
[ ] Question  
[ ] Enhancement  
[X] Bug

kegilbert, 2018-08-24:
Hi @andrewleech ! Very good question, you are on the right track. Unfortunately we don't have an easy way at the moment to disable the RTOS completely. Please see the following for the current "best" way to disable the RTOS (it does exactly what you're doing but disables all modules that use the RTOS to allow the build to compile):

https://github.com/ARMmbed/mbed-os/issues/7800#issue-351121684

andrewleech, 2018-08-24:
Yes I guess that list of features is a bit more selected that disabling all of them eg. 
https://github.com/ARMmbed/mbed-os/pull/7794#issuecomment-415466314

deepikabhavnani, 2018-08-27:
`/master/features/netsocket` - for sure needs rtos

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-386*

SenRamakri, 2019-03-26:
This is already addressed in mbed-os master, and the documentation will be available as part of 5.12 release. For preview of the upcoming documentation please see "Enabling Mbed OS bare metal" section in - https://os.mbed.com/docs/mbed-os/development/tutorials/migrating-to-mbed-os-5.html

closed 2019-03-26 by SenRamakri