ARMmbed/mbed-os #7897: Travis : astyle does not report increase or decrease in files

0xc0170, 2018-08-27:
### Description

It currently reports only total number of files that needs formatting. This was noticed recently, previosly was implemented as only +/- based on master number.

cc @geky 

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-08-27:
Looking at the log, this is there now

```
[0K$ CURR=$(cat astyle-branch.out | grep Formatted | wc -l)
    PREV=$(curl https://api.github.com/repos/$TRAVIS_REPO_SLUG/status/master \
        | jq -re "select(.sha != \"$TRAVIS_COMMIT\")
            | .statuses[] | select(.context == \"travis-ci/$NAME\").description
            | capture(\", (?<files>[0-9]+) files\").warnings" \
        || echo 0)
    
    STATUSM="Passed, ${CURR} files"
    if [ "$PREV" -ne 0 ]
    then
        STATUSM="$STATUSM ($(python -c "print '%+d' % ($CURR-$PREV)") files)"
    fi
    
/home/travis/.travis/job_stages: line 86: [: null
0: integer expression expected
```

0xc0170, 2018-08-27:
PREV is being reported as 0, that is why it is failing to report properly.

0xc0170, 2018-08-29:
Fixes #7898 - reports it again

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-382*

cmonr, 2018-11-20:
@0xc0170 I'm thinking this can finally be closed.

closed 2018-11-20 by ciarmcom