ARMmbed/mbed-os #7900: 'armv4t' warning when compiling with ARMC6

deepikabhavnani, 2018-08-27:
### Description
I also see below warning in detail build dumps, not sure what is causing this.

[DEBUG] Errors: warning: 'armv4t' is unsupported in this version of the product

https://github.com/ARMmbed/mbed-os/pull/7792#issuecomment-415430907

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-383*

0xc0170, 2020-07-13:
Just came to this, still the warning is there plus one more if I use K64F for instance:

```
TARGET_MCU_K64F\device\TOOLCHAIN_ARM_STD\startup_MK64F12.o
warning: 'armv4t' is unsupported in this version of the product
warning: 'arm7tdmi' is unsupported in this version of the product
```