ARMmbed/mbed-os #7927: mbed-os source file access in on-line compiler

jflynn129, 2018-08-29:
### Description

I am attempting to use the on-line compiler to compile a program that requires a modified config.h file within the mbed-os/features/mbedtls/inc/mbedtls folder.  When the program is imported into the on-line compiler, there appears to be no access to this area.  How do I access the mbed-os source files from within the on-line tool so I can make my required modifications?

If I am unable to make the modification, it will require me to restrict users to the CLI GCC_ARM environment.

### Issue request type

[X ] Question  
[ ] Enhancement  
[ ] Bug

mstfkaratas, 2018-08-29:
You can include your custom config file according to mbedtls config.h
https://github.com/ARMmbed/mbed-os/blob/7d871bfff5cc5fd7d068e6acdfbf8805d8d45076/features/mbedtls/inc/mbedtls/config.h#L2957-L2968

So you can use a config paramater in mbed_app.json like the one below to include your custom config header

```
{
    "config": {
        "mbedtls_custom_config": {
            "macro_name": "MBEDTLS_USER_CONFIG_FILE",
            "value": "\"mbedtls_custom_config.h\""
        }
    }
}
```

jflynn129, 2018-08-30:
Thank you that was the info I needed.

closed 2018-08-30 by jflynn129