ARMmbed/mbed-os #7928: USBDevice issues

bmcdonnell-ionx, 2018-08-29:
### Description

Regarding https://github.com/ARMmbed/mbed-os/pull/5874:

I don't understand some aspects of https://github.com/ARMmbed/mbed-os/pull/5874/commits/6decbedbb813b71ea098018c03d9f969a02b9217. These questions are still applicable to the latest/next Mbed OS release, 5.9.6 (0fdfcf735).

1. This commit makes several functions return `const` pointers (e.g. `USBDevice::deviceDesc()`), but then casts away the constness in several places (https://github.com/ARMmbed/mbed-os/pull/5874/commits/6decbedbb813b71ea098018c03d9f969a02b9217#diff-d5a044d3a20b0c2dae580a7d5cfea271). **Why?** Isn't this [bad practice](https://github.com/isocpp/CppCoreGuidelines/blob/master/CppCoreGuidelines.md#Res-casts-const)?

2. This commit adds `memcpy()`s inside some functions which return a pointer to the destination. Such functions are called repeatedly inside of other function calls. (e.g. `USBDevice::deviceDesc()` is called up to 4 times inside `USBDevice::requestGetDescriptor()`.) This is wasteful repeated copying, right? **Why** are these copies necessary? Are they **all** necessary?
    - Can this be reworked to remove some of the `memcpy()`s?
    - For those that must stay, is there a sane way to change the interface to discourage the repeated calls? If not, can the result at least be cached to avoid the repeated copies?

@c1728p9

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug

c1728p9, 2018-08-30:
Hi @bmcdonnell-ionx its good to see you interested in USB. Below are my answers to your questions.

1. Casting away const may not be the best practice, but in this case it is safe. The variable `transfer.ptr` is used for both sending data and receiving data over USB. Because it handles reading data it needs to be able to take a non-const buffer for HOST_TO_DEVICE transfers. In the case of descriptor transfers the direction is DEVICE_TO_HOST and the data `transfer.ptr` points to never changed.  If you are concerned with this you could try refactoring the code to have two seperate pointers - something like a `transfer.ptr_host_to_device' and const 'transfer.ptr_device_to_host'.

2. These copies aren't necessary on every transfer. If you are concerned about it you could refactor the code to setup all the buffers in the constructor. Since descriptors are read infrequently I left the memcpy in the call for readability. Feel free to update this.

bmcdonnell-ionx, 2018-09-13:
@c1728p9, I had some time, so started looking at things a bit.

1. There is a deeper issue here, which is that all the interfaces in all the layers down to `usb/targets/TARGET_*/` are not const correct for sending data to the host. Is this going to be fixable on the Mbed side of things, or is there upstream code that must also be modified?

   If there's upstream code, I guess the compromise is to make the interfaces under the Mbed team's control const correct, submit tickets for upstream, and cast away constness only as necessary before passing to the non-const lower layers to fake it for now.

c1728p9, 2018-09-13:
The should be fixable all the way down to the USBPhy layer by modifying the write functions to take const data rather than mutable data. Below the USBPhy is the vendor code that would be harder to fix. Casts could be used in the short term on the platforms that need it.

bmcdonnell-ionx, 2018-09-13:
> Below the USBPhy is the vendor code that would be harder to fix.

Is the vendor code all under `mbed-os/targets/`, or is there some under `mbed-os/usb/`, e.g. `mbed-os/usb/device/targets/`?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-397*

closed 2019-10-31 by ciarmcom