ARMmbed/mbed-os #7929: NXP system file - incorrect USBClock; update?

bmcdonnell-ionx, 2018-08-29:
### Description

Target: NXP LPC4088
Mbed OS 5.9.5

`system_LPC407x_8x_177x_8x.c` [computes `USBClock`](https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.9.5/targets/TARGET_NXP/TARGET_LPC408X/device/system_LPC407x_8x_177x_8x.c#L434) incorrectly. Problems include but may not be limited to the following:

 - Per Table 21 in section 3.3.3.3 of [UM10562.pdf](https://www.nxp.com/docs/en/user-guide/UM10562.pdf), the only valid values of bits [4:0] of `USBCLKSEL` are 0x0 to 0x3. However, the [`switch` statement](https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.9.5/targets/TARGET_NXP/TARGET_LPC408X/device/system_LPC407x_8x_177x_8x.c#L437) in the code treats 0, 4, and 6 as the only valid values.

Is this file received from upstream? Where / how / from whom?

I don't know where the canonical version lives. I noticed however that the version included in the [Keil MDK5 Software Pack](https://www.keil.com/dd2/pack/) appears to be more recent, but it doesn't address this issue.

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-08-30:
> Is this file received from upstream? Where / how / from whom?

As you noted, comes from cmsis pack - device pack to be specific. If this is t he issue there and yet not addressed, should be reviewed by @ARMmbed/team-nxp

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-6353*

0xc0170, 2019-10-28:
@ARMmbed/team-nxp Can you review?

0xc0170, 2021-01-26:
I'll close this as won't fix. The target is no longer supported. 

It is better to ask on NXP forum where these device support files are coming from and how to fix them. I would start rather there (if the issue is still valid for their startup files).

closed 2021-01-26 by 0xc0170