ARMmbed/mbed-os #7930: USB tear-down functions for hotplugging

bmcdonnell-ionx, 2018-08-29:
### Description

Mbed OS 5.9.5
Target & toolchain agnostic / all

#### Background

[Rumor has it](https://forums.mbed.com/t/usbdevice-in-mbed-os-5/2179/5) that "USB mainline support is coming [to Mbed OS] soon." I don't know what changes are in the works, but here is an issue with the current `unsupported` implementation, which I hope you will address soon - maybe even before the supported release.

#### Issue

Which USB classes/modules should have additional tear-down functions? On a cursory review, one thing I notice is that `USBDevice` and most of its derivatives lack destructors.

### Issue request type

[X] Question
[ ] Enhancement
[ ] Bug

c1728p9, 2018-08-30:
Hi @bmcdonnell-ionx you can get early access to the USB changes by checking out the mbed-os branch feature-hal-spec-usb-device. Feel free to try out that branch and provide feedback on it. All the classes there have the appropriate tear-down functions so they can be created and destroyed safely.

c1728p9, 2018-08-30:
Also @bmcdonnell-ionx below are the preliminary docs. Any feedback you have on these would also be appreciated.

[Overview of USB classes](https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/reference/api/usb/usb_component.md)

[Technology page](https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/reference/technology/USB.md)

[USBAudio](https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/reference/api/usb/USBAudio.md)
[USBCDC](https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/reference/api/usb/USBCDC.md)
[USBHID](https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/reference/api/usb/USBHID.md)
[USBMIDI](https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/reference/api/usb/USBMIDI.md)
[USBKeyboard](https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/reference/api/usb/USBKeyboard.md)
[USBMSD](https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/reference/api/usb/USBMSD.md)
[USBMouse](https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/reference/api/usb/USBMouse.md)
[USBMouseKeyboard](https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/reference/api/usb/USBMouseKeyboard.md)
[USBSerial](https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/reference/api/usb/USBSerial.md)

bmcdonnell-ionx, 2018-08-30:
@c1728p9, thanks for the info.

I'm not sure how much time I'll have to dig in, since my current usage of USB is kind of a one-off. However, to the extent that I can, what is the appropriate venue(s) for giving feedback?

c1728p9, 2018-08-31:
@bmcdonnell-ionx feel free to open issues in mbed-os like you have been for reporting USB feedback.

bmcdonnell-ionx, 2018-08-31:
@c1728p9, if I and/or others were to open a few or several issues, isn't that kind of noisy? Since it's not released yet.

I've seen other instances where devs open a PR tagged `do not merge`, and get review/comments there. Do you think that might fit here, too?

c1728p9, 2018-08-31:
@bmcdonnell-ionx for that you can either add `component: usb` or mention me and I'll add it.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-402*