ARMmbed/mbed-os #7931: USBMSD needs disk_deinitialize()

bmcdonnell-ionx, 2018-08-29:
### Description

Mbed OS 5.9.5
Target & toolchain agnostic / all

A common usage of `USBMSD` will probably involve [sharing a `BlockDevice` as USB Mass Storage Device](https://github.com/bmcdonnell-ionx/mbed-os-demo-usbmsd).

I think the [`USBMSD`](https://github.com/ARMmbed/mbed-os/tree/master/features/unsupported/USBDevice/USBMSD) class should have a `disk_deinitialize()` member function, like `BlockDevice` has a [`deinit()` member function](https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.9.5/features/filesystem/bd/BlockDevice.h#L56), to facilitate hot plugging and unplugging.

Related / bigger picture: https://github.com/ARMmbed/mbed-os/issues/7930

### Issue request type

[ ] Question
[ ] Enhancement
[X] Bug

c1728p9, 2018-08-30:
The [updated USBMSD class](https://github.com/ARMmbed/mbed-os/blob/feature-hal-spec-usb-device/usb/device/USBMSD/USBMSD.cpp#L108) calls `deinit()` in its destructor. Is this sufficient for your use case?

bmcdonnell-ionx, 2018-08-30:
`USBDevice::deinit()` is the mirror of `USBDevice::init()`, not `USBMSD::disk_initialize()`.

Shouldn't there be symmetry here, so that everything that gets powered up can be powered down?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-401*