ARMmbed/mbed-os #7933: Nordic nRF 52

wlamprecht, 2018-08-30:
### Description

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPAN-237*

yogeshk19, 2020-05-21:
@0xc0170  I was wondering if the mbed/cordio team have a suggestion for this issue. I am trying to build an OTA Bootloader and I was hoping to up the MTU to the max IPhone supports, so at 2M PHY I can try updating at a faster rate than with the current MTU size of just 23?

Thanks,
Yogesh

pan-, 2020-10-05:
@yogeshk19 MTU negotiation is well supported on Cordio. There is compile time parameters that impacts the result: 
- [desired-att-mtu](https://github.com/ARMmbed/mbed-os/blob/ad40b1b2672d59b058f7c104fdc612d0dddf3ac3/connectivity/FEATURE_BLE/source/cordio/mbed_lib.json#L50)
- [rx-acl-buffer-size](https://github.com/ARMmbed/mbed-os/blob/ad40b1b2672d59b058f7c104fdc612d0dddf3ac3/connectivity/FEATURE_BLE/source/cordio/mbed_lib.json#L54)
- [max-acl-size](https://github.com/ARMmbed/mbed-os/blob/ad40b1b2672d59b058f7c104fdc612d0dddf3ac3/connectivity/FEATURE_BLE/libraries/TARGET_CORDIO_LL/mbed_lib.json#L36)

I'm closing this issue as it targets Softdevice which is not supported in Mbed OS 6.

closed 2020-10-13 by ciarmcom