ARMmbed/mbed-os #7936: New TCPSocket API always returns NULL from accept function

nikolak94, 2018-08-30:
Hi,

I am using new TCPSocket API that now has accept function (accept() has been moved to abstract socket class and implemented in TCPSocket).  There was an advice in new release to deprecate TCPServer in favor of just TCPSocket::accept(). But i think there is a problem with accept() function in TCPSocket. 
Function always returns NULL pointer instead of created TCPSocket within the function.

TCPSocket *TCPSocket::accept(nsapi_error_t *error)
{
    _lock.lock();
    TCPSocket *connection = NULL;
    nsapi_error_t ret;

    _readers++;

    while (true) {
        if (!_socket) {
            ret = NSAPI_ERROR_NO_SOCKET;
            break;
        }

        _pending = 0;
        void *socket;
        SocketAddress address;
        ret = _stack->socket_accept(_socket, &socket, &address);

        if (0 == ret) {
            TCPSocket *connection = new TCPSocket();
            connection->_lock.lock();
            connection->_factory_allocated = true; // Destroy automatically on close()
            connection->_remote_peer = address;
            connection->_stack = _stack;
            connection->_socket = socket;
            connection->_event = mbed::Callback<void()>(connection, &TCPSocket::event);
            _stack->socket_attach(socket, &mbed::Callback<void()>::thunk, &connection->_event);

            connection->_lock.unlock();
            break;
        } else if ((_timeout == 0) || (ret != NSAPI_ERROR_WOULD_BLOCK)) {
            break;
        } else {
            uint32_t flag;

            // Release lock before blocking so other threads
            // accessing this object aren't blocked
            _lock.unlock();
            flag = _event_flag.wait_any(READ_FLAG, _timeout);
            _lock.lock();

            if (flag & osFlagsError) {
                // Timeout break
                ret = NSAPI_ERROR_WOULD_BLOCK;
                break;
            }
        }
    }

    _readers--;
    if (!_socket) {
        _event_flag.set(FINISHED_FLAG);
    }
    _lock.unlock();
    if (error) {
        *error = ret;
    }
    return connection;
}

kjbracey-arm, 2018-09-04:
Hi, if you're trying to point out a code flaw by pasting in the code, I'm not seeing it.

Are you sure you're not just getting a connection error? Check the `error` output parameter.

nikolak94, 2018-09-05:
Hi Kevin, 

I think that there is a problem with TCPSocket pointer in the above code. 
TCPSocket *connection = NULL; is defined at the begining        
Than: 
if (0 == ret) {
        TCPSocket *connection = new TCPSocket();

is defined again, but at the end the NULL pointer from the begining is returned: 
return connection;

I attached the code of a short example i made. I always get the following output, as tcpsocket instance is not initialized:

Controller IP Address is: 172.16.1.161 , and gateway: 172.16.0.102, network mask 255.255.0.0
Listening TCP Socket 51811 bind return: 0  PORT: 51811 
Listening TCP Socket 51811 listen return: 0  PORT: 51811 
Accept function returns: 0
Connected.
++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0   : 20030000
R1   : 2000D7C8
R2   : 00000006
R3   : 00412403
R4   : 00000000
R5   : 20000D3C
R6   : 00000000
R7   : 00000000
R8   : 00000000
R9   : 00000000
R10  : 00000000
R11  : 00000000
R12  : 08013BA9
SP   : 2000D7A0
LR   : 08008131
PC   : 0800E4AA
xPSR : 21000000
PSP  : 2000D738
MSP  : 2002FFD8
CPUID: 410FC241
HFSR : 40000000
MMFSR: 00000000
BFSR : 00000082
UFSR : 00000000
DFSR : 00000008
AFSR : 00000000
BFAR : 20030010
Mode : Thread
Priv : Privileged
Stack: PSP

-- MbedOS Fault Handler --

++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
Location: 0x800E879
Error Value: 0x800E4AA
Current Thread: Id: 0x2000D7F8 Entry: 0x8011E91 StackSize: 0x1000 StackMem: 0x2000C7F8 SP: 0x2002FFA0 
-- MbedOS Error Info --

When i modyfied the accept function to the following everything started working fine:

TCPSocket *connection = NULL; is defined at the begining        
...
if (0 == ret) {
        connection = new TCPSocket();

....
return connection;

[example program.txt](https://github.com/ARMmbed/mbed-os/files/2351930/example.program.txt)

kjbracey-arm, 2018-09-05:
Ah, thank you - I see it now. @SeppoTakalo - inner-scope `connection` declaration shadowing the top-level one.

nikolak94, 2018-09-05:
Can u help me with a simple question concerned TCPSocket API. When i call accept or connect function, does my main thread gets blocked until connection is established or after sending SYNC request main thread goes on and the connection logic is done in some parallel thread? if there is a signal or something indicating that connection have been established, how can i catch it from main thread? its imortant to me that main thread doesnt block while connection is getting established and to have information when it is finished while doing other stuff?

kjbracey-arm, 2018-09-05:
It depends on whether the socket is marked blocking or non-blocking. By default sockets are blocking.

If blocking, connect will return when the connection completes or fails, and accept will return when someone connects to you.

If non-blocking you will get the "would block" or "in progress" error while the stack is waiting for something to happen.  Set up a callback with the "sigio" method, and when that call is made, trigger your main thread to retry the accept or connect call. Eventually you will get a final success or failure indication.

Your sigio handler can't call the socket directly, but you can queue something on an `EventQueue` or set an `EventFlag` or `Thread` flag to trigger work.

One proviso - if using lwIP, it currently does not actually do non-blocking connect. This is a backwards-compatibility thing - I believe this can be made to work if you just take out the `netconn_set_nonblocking` calls from `LWIP::socket_connect`.

nikolak94, 2018-09-05:
Thank you very much Kevin, i am grateful for fast response.

kjbracey-arm, 2018-09-05:
No problem. This is a tricky area, but the trickiness around non-blocking APIs is not specific to Mbed OS - the socket API provided here is pretty conventional - Google "non-blocking socket connect" and you'll see discussions for other platforms and most of that is applicable. We have all the same core concepts, just some of the spelling is a little different.

nikolak94, 2018-09-05:
I tried to use TCPSocket API, set socket blocking to true and use connect() function, without getting into netconn api, and then  start another thread which is going to handle connecting. That new thread can block until connecting is finished, and than signal the main thread when connection is established? I do not have to do all the work without additional threads, would this then be the right way to continue the main thread while the other one is handling the connection part?

nikolak94, 2018-09-05:
Thank you for your help, that have been struggling me for days now.

SeppoTakalo, 2018-09-05:
Fix provided in #7992

PR #7992 (2018-09-14): Fix TCPSocket::accept()
### Description

TCPSocket::accept() was actually always return NULL, and unsted because we only
have tests with one device.

We will add the testcase for this soon.

This fixes #7936

@kjbracey-arm please review

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Breaking change
closed 2018-09-14 by cmonr