ARMmbed/mbed-os #7942: M2351: Cannot partition flash/SRAM through mbed_app.json

ccli8, 2018-08-31:
### Description

On NUMAKER_PFM_M2351, I can partition flash/SRAM through `mbed_app.json` before. For example, in the example below:
https://github.com/OpenNuvoton/NuMaker-mbed-TZ-secure-example
I can partition secure/non-secure flash to 64 KiB/448 KiB and SRAM to 8 KiB/88 KiB by modifying `target.mbed_rom_start`, but this override fails on `mbed-os-5.9.5`/`mbed-os-5.9.6`.

<pre>
"target_overrides": {
        "*": {
            "platform.stdio-baud-rate": 9600,
            "platform.stdio-convert-newlines": true
        },
        "NUMAKER_PFM_M2351": {
            "target.core": "Cortex-M23",
            "target.inherits": ["NUMAKER_PFM_M2351"],
            "target.device_has_remove": ["TRNG", "SERIAL", "SERIAL_ASYNCH", "SERIAL_FC", "STDIO_MESSAGES"],
           <b> 
            "target.mbed_rom_start":    "0x0",
            "target.mbed_rom_size":     "0x10000",
            "target.mbed_ram_start":    "0x20000000",
            "target.mbed_ram_size":     "0x2000"
           </b>
        }
    }
</pre>

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

andrewleech, 2018-09-03:
I ran into the same issue on my nordic platform, it should be addressed in #7957

If this used to work on your platform you hopefully shouldn't need extra linker script updates, just the patch to the python config system file.

ccli8, 2018-09-03:
@andrewleech Thanks for your information, but #7957 cannot fix my issue. Currently, I roll back to old version which supports flash/SRAM override.

andrewleech, 2018-09-03:
Do you get an error on newer version, or does it just not change the ram setting?

What version are you using that does work?

ccli8, 2018-09-03:
@andrewleech I try on `mbed-os-5.9.6` with #7957 patch, but flash/SRAM start/size are not overridden through `mbed_app.json`. Now, I roll back to mbed-os 63f62165d89f5562c529cd3ecb94823ce1dc7f13.

0xc0170, 2018-09-03:
@ccli8 What has changed in `mbed-os-5.9.5/mbed-os-5.9.6.` ? Can you "git blame" the commit that broke it?

ccli8, 2018-09-03:
@0xc0170 @andrewleech Support for overriding flash/SRAM start/size is raised in #7133, which will be released in `5.10.0-rc1`, and so is not merged yet in `5.9.5`/`5.9.6`.

0xc0170, 2018-09-03:
It provides a new functionality - this issue will be resolved via 5.10 release. It won't be part for 5.9.x. What is this issue about, am I missing something?

ccli8, 2018-09-03:
@0xc0170 I misunderstand with #7133 having merged, override function will be available in `5.9.5`/`5.9.6`. I would close the issue. Thanks for your support.

closed 2018-09-03 by ccli8