ARMmbed/mbed-os #7946: Mbed OS 5.9.6 release version number has not been updated

toyowata, 2018-08-31:
### Description

The ```MBED_PATCH_VERSION``` has not been changed to 6.
https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.9/platform/mbed_version.h#L46

### Issue request type

[ ] Question  
[ ] Enhancement  
[x] Bug

0xc0170, 2018-08-31:
Thanks for reporting, we are looking at this

0xc0170, 2018-08-31:
PR #7874 failed to update the block

cc @ARMmbed/mbed-os-maintainers

0xc0170, 2018-08-31:
We added this to the known issues, please see https://github.com/ARMmbed/mbed-os/releases/tag/mbed-os-5.9.6

It can't be changed now as the release + tags were published. We will make sure this will be fixed in the upcoming releases (version block always updated, need to find a root cause that this was not updated by the release script).

Considering this as resolved.

toyowata, 2018-08-31:
@0xc0170 Thank you very much for your quick response.

>It can't be changed now as the release + tags were published.

I agree with this.

closed 2018-08-31 by 0xc0170