ARMmbed/mbed-os #7947: Expose more LoRaWAN internal state

EnsilicaEdd, 2018-08-31:
### Description
My project has a requirement to display the current uplink and downlink counters, as well as the network and application session keys (ignoring for a moment whether or not this is a good idea).

The counters are tantalizingly close to the surface, as they're made available to `LoRaWANStack` by `LoRaMAC` as part of the rx and tx metadata, but just not copied into the user-facing versions of the same.  But the keys remain hidden away deep in `LoRaMAC.`

One way or another, I'm going to get hold of the information I need, but would favour doing it via a clean interface.  Do you see any merit in exposing some or all this information with an official API?

I'm happy to make a proposal, but only if there is a wider appetite for it.

### Issue request type

[ ] Question  
[X] Enhancement  
[ ] Bug

kivaisan, 2018-09-03:
@EnsilicaEdd We already have plan to implement session info which also includes keys.

EnsilicaEdd, 2018-09-03:
Thanks, @kivaisan.  Do you have a timescale you could share?  I'd be happy to help if I can.

kivaisan, 2018-09-04:
@EnsilicaEdd At the moment target is mbed-os 5.11.

EnsilicaEdd, 2018-09-05:
Thanks for the information.
I need this right now, so am going to go ahead and make my own local changes.  

But on the off-chance I can save myself some work, does 'target 5.11' mean "it's on the list of things to look at after 5.10" or "we've already done the work, but it won't drop until 5.11" ?
If it's the latter, is there a patch you could share in advance?
Cheers.

AnttiKauppila, 2018-09-05:
It means it will be merged into mbed OS master branch when it is ready, but will be officially published in mbed OS 5.11 (around mid December).

EnsilicaEdd, 2018-09-05:
Understood.  I'll close this then.
Thanks all.

closed 2018-09-05 by EnsilicaEdd